luxonis/depthai-core #854: [Feature-Request] IR Dot Projector On/Off Mode

borongyuan, 2023-06-21:
### Start with the `why`:
IR dot projector is useful for improving depth estimation. However, for VSLAM applications, the projected dots will affect feature extraction. We don't want to completely disable the IR dot projector, which would sacrifice active stereo vision functionality. One solution for this is to add an On/Off mode to IR dot projector. This feature is also available in realsense and mynteye products.

### Move to the `what`:
We want to enable the IR dot projector for one frame, and then turn it off for the next frame, switch repeatedly. At this time, assuming that the frame rate is set to 60FPS, then we will get 30FPS images with dot projection (for depth estimation), and 30FPS clean images (for visual odometry).

### Move to the `how`:

Erol444, 2023-06-22:
Hi @borongyuan ,
I believe this could already be implemented via Script node: https://github.com/luxonis/depthai-python/blob/main/examples/MonoCamera/mono_preview_alternate_pro.py
Thoughts?

borongyuan, 2023-06-23:
Thanks,
I didn't notice this because it is not mentioned in the documentation. This could be very useful. It would be even better if there is a firmware-level implementation later.

closed 2023-06-23 by borongyuan

yyh-pc, 2024-08-19:
@borongyuan  hello, when I try to set the fps to 60, the part of the script that checks for left and right image frame time synchronization always reports an error like this:
1844301011CD3BF500] [3.2] [6.534] [Script(6)] [error] frame/event desync! Fr-Ev: 1 frames, 16.667 ms; R-L: 0.013 ms
[1844301011CD3BF500] [3.2] [6.551] [Script(6)] [error] frame/event desync! Fr-Ev: 1 frames, 16.667 ms; R-L: 0.012 ms
[1844301011CD3BF500] [3.2] [6.568] [Script(6)] [error] frame/event desync! Fr-Ev: 1 frames, 16.666 ms; R-L: 0.013 ms
[1844301011CD3BF500] [3.2] [6.584] [Script(6)] [error] frame/event desync! Fr-Ev: 1 frames, 16.667 ms; R-L: 0.011 ms
[1844301011CD3BF500] [3.2] [6.601] [Script(6)] [error] frame/event desync! Fr-Ev: 1 frames, 16.666 ms; R-L: 0.011 ms

The parameters are set as follows:
if 1: # PoE config
fps = 60
res = dai.MonoCameraProperties.SensorResolution.THE_400_P
poolSize = 25 # default 3, increased to prevent desync
else: # USB
fps = 30
res = dai.MonoCameraProperties.SensorResolution.THE_720_P
poolSize = 8 # default 3, increased to prevent desync

I tried to change the poolSize parameter but it still doesn't work. How should I fix this problem?