luxonis/depthai-core #856: [Feature-Request] DepthUnit: `TENTH_OF_MILLIMETER`

Erol444, 2023-06-22:
### Start with the `why`:
In some cases ([example here](https://discuss.luxonis.com/d/1716-change-depth-unit-to-01-mm/7)) it would be useful to have sub-mm precision depth values, especially for short-range OAK cameras.

### Move to the `what`:
Add another `DepthUnit` that's below `MILLIMETER`, perhaps `TENTH_OF_MILLIMETER`.

szabi-luxonis, 2023-06-26:
This is already achievable by using `customDepthUnitMultiplier`

Robengaard, 2023-06-26:
Hi @szabi-luxonis 
Can you provide me of an example?
`stereo.initialConfig.setDepthUnit(stereo.initialConfig.AlgorithmControl.DepthUnit.CUSTOM)
stereo.initialConfig.AlgorithmControl.customDepthUnitMultiplier(0.1) # 0.1 mm`
does not work.
Br Martin

szabi-luxonis, 2023-06-26:
@Robengaard please refer to our [documentation](https://docs.luxonis.com/projects/api/en/latest/references/cpp/?highlight=customDepthUnitMultiplier#_CPPv4N3dai20RawStereoDepthConfig16AlgorithmControl25customDepthUnitMultiplierE:~:text=Custom%20depth%20unit%20multiplier%2C%20if%20custom%20depth%20unit%20is%20enabled%2C%20relative%20to%201%20meter.%20A%20multiplier%20of%201000%20effectively%20means%20depth%20unit%20in%20millimeter.)

Custom depth unit multiplier, if custom depth unit is enabled, relative to 1 meter. A multiplier of 1000 effectively means depth unit in millimeter.

Robengaard, 2023-06-26:
Hi @szabi-luxonis 

When using you code I get the following error: 
    depth.initialConfig.AlgorithmControl.customDepthUnitMultiplier(10 * 1000)
TypeError: 'property' object is not callable
where depth = self.pipeline.create(dai.node.StereoDepth)

szabi-luxonis, 2023-06-26:
> depth.initialConfig.AlgorithmControl.customDepthUnitMultiplier(10 * 1000)

You are using the wrong syntax. @Robengaard 

`depth.initialConfig.AlgorithmControl.customDepthUnitMultiplier = 10 * 1000`

Robengaard, 2023-06-26:
@szabi-luxonis thanks for pointing it out. However I seem to get the same depth image no matter the setting of depth.initialConfig.AlgorithmControl.customDepthUnitMultiplier.
If I change stereo.initialConfig.setDepthUnit I get a different output.

szabi-luxonis, 2023-06-26:
@Robengaard still wrong API/syntax used. I see this stems from the initial support thread from our forum. CC: @Erol444 
This is the correct one:
```
stereo.initialConfig.setDepthUnit(stereo.initialConfig.AlgorithmControl.DepthUnit.CUSTOM)
config = stereo.initialConfig.get()
config.algorithmControl.customDepthUnitMultiplier = 10 * 1000
stereo.initialConfig.set(config)
```

Robengaard, 2023-06-26:
@szabi-luxonis Thanks, this was the right thing to do. I have an extra question. It seems like the depth is jumping with 0.6 mm this is some values in row 100 in the x direction.
`3839, 3839, 3839, 3839, 3832, 3832, 3839, 3839, 3839, 3832, 3839,
       3839, 3832, 3832, 3832, 3839, 3832, 3832, 3832, 3832, 3832, 3832,
       3832, 3832, 3839, 3839, 3839, 3839, 3839, 3839, 3839, 3839, 3832,
       3832, 3832, 3839, 3839, 3839, 3839, 3839, 3832, 3832, 3832, 3832,
       3832, 3832, 3832, 3832, 3832, 3832, 3832, 3832, 3832, 3832, 3832,
       3832, 3832, 3832, 3832, 3832, 3832, 3832, 3832, 3832, 3826, 3826,
       3826, 3826, 3826, 3832, 3832`
Is this a hardware limitation or can I do something to get a higher resolution?

szabi-luxonis, 2023-06-26:
@Robengaard you can increase number of subpixel levels:

```
stereo.initialConfig.setSubpixel(True)
stereo.initialConfig.setSubpixelFractionalBits(5)
```

Robengaard, 2023-07-04:
Thanks, it works. :)

closed 2023-07-05 by Erol444