luxonis/depthai-core #857: [BUG] Cannot find camd on OAK-FFC 4P with OV9782 cameras

markapron, 2023-06-23:
Similar to [647](https://github.com/luxonis/depthai-core/issues/647) and [this forum post](https://discuss.luxonis.com/d/1538-cannot-find-camd-on-oak-ffc-4p-oak-som-pro-s3) but there's no clear solution.

**Describe the bug**
I've tried using two different OAK-FFC-4P boards with several different cameras and will always get the output, 
`[error] Camera not detected on socket: 3`
 when trying to connect to cam-d with the cam_test.py script as well as other custom scripts.

The device_manager.py shows the following information about the OAK-FFC-4P:

    Name of connect device: 1.2
    Version of newest bootloader: 0.0.26
    Current Depthai version: 2.22.0.0
    Device state: BOOTLOADER
    Flashed bootloader version: 0.0.22
    Current commit: (blank)

**Minimal Reproducible Example**
Plug in two OV9782 cameras into the OAK-FFC 4P board in ports cam-a and cam-d.
Run the cam_test.py script

**Expected behavior**
Expect two camera feeds should appear but only cam-a appears.

**Screenshots**
```
$ python3 cam_test.py --cameras rgb,c camd,c
Enabled cameras:
    rgb : color
   camd : color
DepthAI version: 2.22.0.0
DepthAI path: /home/mkapron/.local/lib/python3.8/site-packages/depthai.cpython-38-x86_64-linux-gnu.so
cam_test.py:284: DeprecationWarning: Use constructor taking 'UsbSpeed' instead
  with dai.Device(*dai_device_args) as device:
[14442C108183D5D600] [1.2] [3.417] [ColorCamera(5)] [error] Camera not detected on socket: 3
[14442C108183D5D600] [1.2] [3.417] [ColorCamera(3)] [warning] Unsupported resolution set for detected camera OV9782, needs 800_P or 720_P. Defaulting to 800_P
Connected cameras:
 -socket CAM_A : OV9782 1280 x  800 focus:fixed - COLOR MONO
USB speed: SUPER
IR drivers: []
Cam:      rgb           camd     [host | capture timestamp]
FPS:  30.02| 30.00   0.00|  0.00
```

**Additional context**
I tried running this on two different OAK-FFC 4P boards with different cameras connected to each ports on both boards. All the cameras will work on cam-a, cam-b, and cam-c but nothing is working on cam-d.
I've also tried running this on two different host machines, and x64 laptop running 20.04 and on an Orin NX with the same behavior on both.

markapron, 2023-06-23:
Here is the

markapron, 2023-06-23:
Here is the output with debug logging

```
$ DEPTHAI_LEVEL=debug python3 cam_test.py --cameras rgb,c camd,c
[2023-06-23 10:41:35.656] [depthai] [debug] Python bindings - version: 2.22.0.0 from  build: 2023-06-13 01:32:28 +0000
[2023-06-23 10:41:35.656] [depthai] [debug] Library information - version: 2.22.0, commit: 82ab07d037c02f56042d1d2d55a718f379651ed9 from 2023-06-13 02:13:14 +0200, build: 2023-06-13 01:18:16 +0000
[2023-06-23 10:41:35.657] [depthai] [debug] Initialize - finished
Enabled cameras:
    rgb : color
   camd : color
DepthAI version: 2.22.0.0
DepthAI path: /home/mkapron/.local/lib/python3.8/site-packages/depthai.cpython-38-x86_64-linux-gnu.so
cam_test.py:284: DeprecationWarning: Use constructor taking 'UsbSpeed' instead
  with dai.Device(*dai_device_args) as device:
[2023-06-23 10:41:35.727] [depthai] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.26.tar.xz' open: 2ms, archive read: 69ms
[2023-06-23 10:41:36.051] [depthai] [debug] Resources - Archive 'depthai-device-fwp-f033fd9c7eb0b3578d12f90302e87759c78cfb36.tar.xz' open: 1ms, archive read: 393ms
[2023-06-23 10:41:36.067] [depthai] [debug] Searching for booted device: DeviceInfo(name=1.2, mxid=14442C108183D5D600, X_LINK_BOOTLOADER, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[2023-06-23 10:41:36.071] [depthai] [debug] Connected bootloader version 0.0.22
[2023-06-23 10:41:36.071] [depthai] [info] New bootloader version available. Device has: 0.0.22, available: 0.0.26
[2023-06-23 10:41:36.820] [depthai] [debug] DeviceBootloader about to be closed...
[2023-06-23 10:41:36.820] [depthai] [debug] XLinkResetRemote of linkId: (0)
[2023-06-23 10:41:37.571] [depthai] [debug] DeviceBootloader closed, 751
[2023-06-23 10:41:37.575] [depthai] [debug] Searching for booted device: DeviceInfo(name=1.2, mxid=14442C108183D5D600, X_LINK_BOOTED, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[14442C108183D5D600] [1.2] [3.407] [system] [info] Memory Usage - DDR: 0.12 / 337.18 MiB, CMX: 2.04 / 2.50 MiB, LeonOS Heap: 7.49 / 80.03 MiB, LeonRT Heap: 2.89 / 41.06 MiB
[14442C108183D5D600] [1.2] [3.407] [system] [info] Temperatures - Average: 39.75C, CSS: 40.51C, MSS 39.11C, UPA: 39.81C, DSS: 39.58C
[14442C108183D5D600] [1.2] [3.407] [system] [info] Cpu Usage - LeonOS 36.43%, LeonRT: 0.66%
[14442C108183D5D600] [1.2] [1687531300.556] [host] [debug] Schema dump: {"connections":[{"node1Id":0,"node1Output":"out","node1OutputGroup":"","node2Id":5,"node2Input":"inputControl","node2InputGroup":""},{"node1Id":5,"node1Output":"isp","node1OutputGroup":"","node2Id":4,"node2Input":"in","node2InputGroup":""},{"node1Id":0,"node1Output":"out","node1OutputGroup":"","node2Id":3,"node2Input":"inputControl","node2InputGroup":""},{"node1Id":3,"node1Output":"isp","node1OutputGroup":"","node2Id":2,"node2Input":"in","node2InputGroup":""}],"globalProperties":{"calibData":null,"cameraTuningBlobSize":null,"cameraTuningBlobUri":"","leonCssFrequencyHz":700000000.0,"leonMssFrequencyHz":700000000.0,"pipelineName":null,"pipelineVersion":null,"sippBufferSize":18432,"sippDmaBufferSize":16384,"xlinkChunkSize":-1},"nodes":[[0,{"id":0,"ioInfo":[[["","out"],{"blocking":false,"group":"","id":20,"name":"out","queueSize":8,"type":0,"waitForMessage":false}]],"name":"XLinkIn","properties":[185,3,189,7,99,111,110,116,114,111,108,130,0,0,80,0,8]}],[1,{"id":1,"ioInfo":[[["","out"],{"blocking":false,"group":"","id":19,"name":"out","queueSize":8,"type":0,"waitForMessage":false}]],"name":"XLinkIn","properties":[185,3,189,9,116,111,102,67,111,110,102,105,103,130,0,0,80,0,8]}],[2,{"id":2,"ioInfo":[[["","in"],{"blocking":true,"group":"","id":18,"name":"in","queueSize":8,"type":3,"waitForMessage":true}]],"name":"XLinkOut","properties":[185,3,136,0,0,128,191,189,3,114,103,98,0]}],[3,{"id":3,"ioInfo":[[["","video"],{"blocking":false,"group":"","id":17,"name":"video","queueSize":8,"type":0,"waitForMessage":false}],[["","still"],{"blocking":false,"group":"","id":15,"name":"still","queueSize":8,"type":0,"waitForMessage":false}],[["","isp"],{"blocking":false,"group":"","id":14,"name":"isp","queueSize":8,"type":0,"waitForMessage":false}],[["","preview"],{"blocking":false,"group":"","id":16,"name":"preview","queueSize":8,"type":0,"waitForMessage":false}],[["","raw"],{"blocking":false,"group":"","id":13,"name":"raw","queueSize":8,"type":0,"waitForMessage":false}],[["","frameEvent"],{"blocking":false,"group":"","id":12,"name":"frameEvent","queueSize":8,"type":0,"waitForMessage":false}],[["","inputConfig"],{"blocking":false,"group":"","id":11,"name":"inputConfig","queueSize":8,"type":3,"waitForMessage":false}],[["","inputControl"],{"blocking":true,"group":"","id":10,"name":"inputControl","queueSize":8,"type":3,"waitForMessage":false}]],"name":"ColorCamera","properties":[185,26,185,27,0,3,0,0,0,185,3,0,0,0,185,5,0,0,0,0,0,185,5,0,0,0,0,0,0,0,0,0,0,0,0,185,3,0,0,0,185,3,0,0,0,0,0,0,0,0,0,0,0,0,0,0,189,0,255,0,1,0,129,44,1,129,44,1,255,255,255,255,0,136,0,0,240,65,0,136,0,0,128,191,136,0,0,128,191,1,185,4,1,1,1,1,3,3,4,4,4,190]}],[4,{"id":4,"ioInfo":[[["","in"],{"blocking":true,"group":"","id":9,"name":"in","queueSize":8,"type":3,"waitForMessage":true}]],"name":"XLinkOut","properties":[185,3,136,0,0,128,191,189,4,99,97,109,100,0]}],[5,{"id":5,"ioInfo":[[["","video"],{"blocking":false,"group":"","id":8,"name":"video","queueSize":8,"type":0,"waitForMessage":false}],[["","still"],{"blocking":false,"group":"","id":6,"name":"still","queueSize":8,"type":0,"waitForMe [... cut ...]
[14442C108183D5D600] [1.2] [5.410] [system] [info] Cpu Usage - LeonOS 19.78%, LeonRT: 2.40%
FPS:  29.99| 29.99   0.00|  0.00 [14442C108183D5D600] [1.2] [6.411] [system] [info] Memory Usage - DDR: 104.55 / 337.18 MiB, CMX: 2.09 / 2.50 MiB, LeonOS Heap: 24.49 / 80.03 MiB, LeonRT Heap: 3.66 / 41.06 MiB
[14442C108183D5D600] [1.2] [6.411] [system] [info] Temperatures - Average: 40.04C, CSS: 40.51C, MSS 39.81C, UPA: 39.58C, DSS: 40.28C
[14442C108183D5D600] [1.2] [6.411] [system] [info] Cpu Usage - LeonOS 19.89%, LeonRT: 2.36%
FPS:  30.01| 30.00   0.00|  0.00
```

themarpe, 2023-06-24:
@markapron 
do you mind running with `DEPTHAI_DEBUG=1` and posting those results as well? And `examples/calibration/calibration_dump.py`

markapron, 2023-06-26:
Output from `DEPTHAI_DEBUG=1` 
```
$ DEPTHAI_DEBUG=1 python3 cam_test.py --cameras rgb,c camd,c
[2023-06-26 10:19:36.065] [depthai] [info] DEPTHAI_DEBUG enabled, lowered DEPTHAI_LEVEL to 'debug'
[2023-06-26 10:19:36.065] [depthai] [debug] Python bindings - version: 2.22.0.0 from  build: 2023-06-13 01:32:28 +0000
[2023-06-26 10:19:36.065] [depthai] [debug] Library information - version: 2.22.0, commit: 82ab07d037c02f56042d1d2d55a718f379651ed9 from 2023-06-13 02:13:14 +0200, build: 2023-06-13 01:18:16 +0000
[2023-06-26 10:19:36.066] [depthai] [debug] Initialize - finished
Enabled cameras:
    rgb : color
   camd : color
DepthAI version: 2.22.0.0
DepthAI path: /home/user/.local/lib/python3.8/site-packages/depthai.cpython-38-x86_64-linux-gnu.so
cam_test.py:284: DeprecationWarning: Use constructor taking 'UsbSpeed' instead
  with dai.Device(*dai_device_args) as device:
[2023-06-26 10:19:36.158] [depthai] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.26.tar.xz' open: 2ms, archive read: 87ms
[2023-06-26 10:19:36.430] [depthai] [debug] Resources - Archive 'depthai-device-fwp-f033fd9c7eb0b3578d12f90302e87759c78cfb36.tar.xz' open: 2ms, archive read: 358ms
[2023-06-26 10:19:36.444] [depthai] [debug] Searching for booted device: DeviceInfo(name=1.2, mxid=14442C108183D5D600, X_LINK_BOOTLOADER, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[2023-06-26 10:19:36.448] [depthai] [debug] Connected bootloader version 0.0.22
[2023-06-26 10:19:36.448] [depthai] [info] New bootloader version available. Device has: 0.0.22, available: 0.0.26
[2023-06-26 10:19:37.201] [depthai] [debug] DeviceBootloader about to be closed...
[2023-06-26 10:19:37.202] [depthai] [debug] XLinkResetRemote of linkId: (0)
[2023-06-26 10:19:37.948] [depthai] [debug] DeviceBootloader closed, 746
[2023-06-26 10:19:37.952] [depthai] [debug] Searching for booted device: DeviceInfo(name=1.2, mxid=14442C108183D5D600, X_LINK_BOOTED, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[14442C108183D5D600] [1.2] [3.396] [system] [warning] PRINT:LeonCss: GPIO boot mode 0x3, interface SPI_MASTER_EFF
Setting aons(0..4) back to boot from flash (offset = 0)
====ENABLE WATCHDOG====1
initial keepalive, countdown: 10
PLL0: 700000 AUX_IO0: 24000 AUX_IO1: 24000 MCFG: 24000 MECFG: 24000
Board init ret 3
DepthAI Firmware - version: f033fd9c7eb0b3578d12f90302e87759c78cfb36
eeprom configuration version: 55AA0007
Reading VERSION 7 --- ->
eeprom configuration load from user area, status: 0
Reading VERSION 7 --- ->
Found a device/board entry matching the eeprom data - Board: OAK-FFC-4P (R7M1E7) Device: OAK-FFC-4P
Board options: 00000000
--> brdInit ...
brdInitAuxDevices: Error: SC = 27: io_initialize expander_cam_gpios_1 [OK]

spi_N25Q_init: Flash JEDEC ID: 20 bb 19
inited hal ---
def: 53 46 45 34 10 06 08 34 10 53 14 45 46 08 06
val: 53 46 45 34 10 06 08 34 10 53 14 45 46 08 06
Opening bus for IR driver: 0
Failed to probe IR driver LM3644
Opening bus for IR driver: 1
Failed to probe IR driver LM3644
Opening bus for IR driver: 2
Failed to probe IR driver LM3644
Opening bus for IR driver: 3
Failed to probe IR driver LM3644
Closing EEPROm!
MyriaX board configuration
pll0 frequency: 700000, ref0 frequency: 24000
Is booted from flash by bootloader: 0
Networking not available...
=== Enumerating on socket: Cam_A / RGB / Center
  >> Registered camera CCM OV9782 as /dev/Camera_0
camera socket: 0, name: CAMA-4L
        config - w: 1280, h: 720, type: MONO
        config - w: 1280, h: 720, type: COLOR
        config - w: 1280, h: 800, type: MONO
        config - w: 1280, h: 800, type: COLOR
Adding socket 0: cam 5. Sen name: OV9782
=== Enumerating on socket: Cam_B / Left
initial keepalive, countdown: 9
=== Enumerating on socket: Cam_C / Right
initial keepalive, countdown: 8
=== Enumerating on socket: CAM_D
initial keepalive, countdown: 7
[... 213 lines cut ...]
    ],
    "rectifiedRotationRight": [
      [
        0.0,
        0.0,
        0.0
      ],
      [
        0.0,
        0.0,
        0.0
      ],
      [
        0.0,
        0.0,
        0.0
      ]
    ],
    "rightCameraSocket": 255
  },
  "version": 7
}
User calibration raw: [7, 0, 170, 85, 0, 0, 0, 0, 0, 68, 68, 50, 48, 57, 48, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 82, 55, 77, 49, 69, 55, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 255, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 2 [... cut ...] 48, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 104, 30, 57, 100, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255]
```

themarpe, 2023-06-26:
CC: @jakaskerl do you mind reproing this as well as testing with B&C sockets?

Also worth testing would be using `develop` branch

jakaskerl, 2023-06-29:
I only had rev5 board available, as rev7 (R7M1E7 - the board you have) boards are still in delivery. 
It works on my rev5 with reproduced environment (depthai versions). 

You can see in your debug log that cam-d acts like it is not even connected:
```
 >> Registered camera CCM OV9782 as /dev/Camera_0
camera socket: 0, name: CAMA-4L
        config - w: 1280, h: 720, type: MONO
        config - w: 1280, h: 720, type: COLOR
        config - w: 1280, h: 800, type: MONO
        config - w: 1280, h: 800, type: COLOR
Adding socket 0: cam 5. Sen name: OV9782
=== Enumerating on socket: Cam_B / Left
initial keepalive, countdown: 9
=== Enumerating on socket: Cam_C / Right
initial keepalive, countdown: 8
=== Enumerating on socket: CAM_D
initial keepalive, countdown: 7
initial keepalive, countdown: 6
```
Idea behind why that happens:
The camera sockets on the 4P have not been changed from revision 5 to 7. What has been changed, is the FSYNC - hardware syncing setup. 

In previous revisions (5 and lower) the cam_a and cam_d were hard-synced together and so were the cam_b and cam_c. But the pairs were not synced to each other, though you could achieve synchronization by flipping on a switch to create a connection between them [source](https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/sync_frames/#oak-ffc-4p-fsin). 
In revision 7, this was changed so a, b, c and d are now hard-synced together out of the box. This led to a conclusion that it might be this change that caused cam_d to stop working properly. 

So, could you try manually setting the GPIO6 pin to LOW:
```
config = dai.Device.Config()
config.board.gpio[6] = dai.BoardConfig.GPIO(dai.BoardConfig.GPIO.OUTPUT,
                                            dai.BoardConfig.GPIO.Level.LOW)
```

Let me know if that helps.
I will be able to debug this further when we receive the board. 

Thanks.

markapron, 2023-06-29:
Thanks for the insight. Unfortunately the change did not solve the problem. I tested with a camera connect to cam-a and cam-d and added that change to line 188 of `cam_test.py` as well as a slight modification of `rgb_preview.py` (see below) and got the following output:
```
$ ./rgb_preview.py 
[14442C10214FE7D600] [1.2] [3.389] [ColorCamera(0)] [error] Camera not detected on socket: 3
Connected cameras: [{socket: CAM_A, sensorName: OV9782, width: 1280, height: 800, orientation: AUTO, supportedTypes: [COLOR, MONO], hasAutofocus: 0, name: CAMA-4L}]
Usb speed: SUPER
Bootloader version: 0.0.22
Device name: OAK-FFC-4P
```

rgb_preview.py:
```
#!/usr/bin/env python3

import cv2
import depthai as dai

# Change config to hopefully fix camd
config = dai.Device.Config()
config.board.gpio[6] = dai.BoardConfig.GPIO(dai.BoardConfig.GPIO.OUTPUT,
                                            dai.BoardConfig.GPIO.Level.LOW)

# Create pipeline
pipeline = dai.Pipeline()

# Define source and output
camRgb = pipeline.create(dai.node.ColorCamera)
camRgb.setBoardSocket(dai.CameraBoardSocket.CAM_D)
camRgb.setResolution(dai.ColorCameraProperties.SensorResolution.THE_720_P)
xoutRgb = pipeline.create(dai.node.XLinkOut)

xoutRgb.setStreamName("rgb")

# Properties
camRgb.setPreviewSize(300, 300)
camRgb.setInterleaved(False)
camRgb.setColorOrder(dai.ColorCameraProperties.ColorOrder.RGB)

# Linking
camRgb.preview.link(xoutRgb.input)

# Connect to device and start pipeline
with dai.Device(pipeline) as device:

    print('Connected cameras:', device.getConnectedCameraFeatures())
    # Print out usb speed
    print('Usb speed:', device.getUsbSpeed().name)
    # Bootloader version
    if device.getBootloaderVersion() is not None:
        print('Bootloader version:', device.getBootloaderVersion())
    # Device name
    print('Device name:', device.getDeviceName())

    # Output queue will be used to get the rgb frames from the output defined above
    qRgb = device.getOutputQueue(name="rgb", maxSize=4, blocking=False)

    while True:
        inRgb = qRgb.get()  # blocking call, will wait until a new data has arrived

        # Retrieve 'bgr' (opencv format) frame
        cv2.imshow("rgb", inRgb.getCvFrame())

        if cv2.waitKey(1) == ord('q'):
            break
```

closed 2023-07-11 by markapron

mitrasang, 2023-08-24:
@jakaskerl  can we get this fix on RVC3 as well?