luxonis/depthai-core #859: [Feature-Request] Depth alignment on multiple sensors

hidmic, 2023-06-26:
### Start with the `why`:

I'm using Oak D Lite cameras for RGBD odometry in an environment very poor lighting conditions. This works for the most part, but struggles sometimes. The stereo pair, on the other hand, has a much better time. I suspect the automatic ambient light adjustment is kicking in. I wonder if we can improve our odometry by feeding each monochrome camera as a separate depth camera, provided a depth stream aligned with each is available.

### Move to the `what`:
Have the StereoDepth node output depth aligned with more than one sensor.

### Move to the `how`:
IIUC the StereoDepth node computes disparity from left to right and from right to left already, to cope with occlusions. It also provides the means to align depth to either camera or the central RGB camera. Perhaps splitting depth alignment into its own node and exposing more of the StereoDepth internals is the most flexible solution.

themarpe, 2023-06-26:
@hidmic 
Do you mind expanding on the idea a bit, as to which sensors do you intend to align to? So essentially getting the one depth map aligned to left and one to right camera?

Such a generic "Reprojection" node is in the works.

Otherwise you may run Stereo twice, once aligned to each side

hidmic, 2023-06-26:
>  So essentially getting the one depth map aligned to left and one to right camera?

Correct.

> Such a generic "Reprojection" node is in the works.

Wonderful. May I ask what the timeline is?

> Otherwise you may run Stereo twice, once aligned to each side

I thought about that, and it is definitely an option. It is less convenient though, as I'll have two pairs of synchronized left and right images (unless all node outputs in a pipeline are synchronized, are they?).

themarpe, 2023-06-26:
We are actively working on it - eta to develop 1 month

Those outputs will be synchronize as same timestamls will be taken together. So that approach would work with the penalty of running stereo twice.