luxonis/depthai-core #862: `CameraFeatures::hasAutofocus` is wrong or misleading on OAK-D-Lite

diablodale, 2023-06-27:
My ***fixed focus*** OAK-D-Lite sensors are returning CameraFeatures vectors from getConnectedCameraFeatures() that have unexpected `CameraFeatures::hasAutofocus == true` values
But these are fixed-focus sensors. Therefore, I expect that value to be false.

### Setup

* depthai-core 5f75ae7 plus private fixes
* Microsoft Windows [Version 10.0.22621.1848]
* vs2022 v17.6.4

### Repro

1. build a c++ app that calls device.getConnectedCameraFeatures() and inspects the result
2. attache an OAK-D-Lite fixed focus device
3. run the app and inspect

### Result

hasAutofocus = true

[image]

### Expected

hasAutofocus = false

### Notes

The SDK code has a unusual comment

```c++
/**
*  Whether an autofocus VCM IC was detected
*/
bool hasAutofocus = false;
```
> 

Perhaps there is an autofocus IC somewhere on the board.
However, I can think of no customer that would be interested in this. We want to know if auto-focus is physically and logically enabled for the camera in question. An unused/disabled IC is of no interest to us.

themarpe, 2023-07-03:
Thanks for the report - this will be addressed in next release for standard models.

The general comment will still apply for eg FFC cases, as discovery in that case will be dictated by the VCM IC, which can be present also on FF CCMs.

diablodale, 2023-07-03:
Then for clarity, I suggest one of two options:

* have only `bool hasAutofocus; // it represents the customer-facing capability of auto or fixed-focus`
* have two bools
  ```c++
  bool hasAutofocus; // it represents the customer-facing capability of auto or fixed-focus
  bool hasAutofocusIC; // it represents the electrical presence of a VCM IC
  ```

Looking at https://docs.luxonis.com/projects/hardware/en/latest/pages/ffc-cameras/, seems to me the FFC board (which has no camera) and the pluggable FFC camera(s) are independent. I would expect the firmware on the FFC board identify the attached FFC camera and its capabilities. And therefore the firmware knows if that camera is fixed or auto-focus. And therefore can correctly populate `hasAutofocus`.

The presence if a VC or any chip on a board is dubious. Why? Because it is the ***capability*** that is important. And the capability could be implemented by firmware, IC, ffpga, software, etc.

AnielAlexa, 2023-07-14:
Hello, you can test the fix for autofocus.
https://github.com/luxonis/depthai-core/tree/autofocus_fix
Let me know if problem still persists.

diablodale, 2023-07-17:
Hi. I do not have the bandwidth to test a private build of Luxonis firmware. Instead I have feedback on https://github.com/luxonis/depthai-core/commit/40df8f3aaa3b251b42a9d64f4c58a2a3d3294f6e

* Where are the test cases? Test cases are needed to validate this fix across all OAK hardware models now and to prevent regressions on this API.
* How did the API change? I see no depthai-core code changes in this commit. What is the expected behavior of the APIs? Above I discuss the ambiguity yet there is no response from Luxonis on expected behavior. Without clear expected behavior no test cases can be written and I can not validate anything.

themarpe, 2023-07-27:
WRT the actual behavior - it currently only addresses known negatives.

We still cannot properly know for sure about capability for certain types of devices (eg FFC boards /w FFC modules connected).

The addition of IC parameter wouldn't be exactly the whole story. We'd need to modify hasAutofocus to be optional, whether or not FW is confident about the capability, or it leaves it up to be usage defined. IC could still be used to override.

Perhaps we retain the below:
```
bool hasAutofocus; // it represents the customer-facing capability of auto or fixed-focus or in when not possible to detect, reflects hasAutofocusIC (custom FFC devices)
bool hasAutofocusIC; // it represents the electrical presence of a VCM IC
```

Which will hold for standard devices while not break API

---

@AnielAlexa do you mind exposing the above proposed hasAutofocusIC, will be a good verbose feature to have in some edge cases
WRT tests, perform same check on host - FF specified in eeprom & hasAutofocusIC == true THEN hasAutofocus == false.

diablodale, 2023-07-31:
I see this at two levels:

### Hardware level

* Retail units, Luxonis knows definitely the cameras and their features.
* FFC boards are custom solutions assembled by someone. If that someone wants a full featured functioning unit, then they need to fully and correctly save needed config values to firmware. One of the many values is `eeprom.lensPosition`. If someone doesn't correctly save that value to firmware, then all features/functionality related to it will probably not work...by design.
* It is possible to probe the i2c bus for devices. `i2cdetect` does this. It may then be possible to prove the found devices to identify them with registers or other methods. I think it is reasonable for this probe to only work on supported cameras.

### depthai-core and app level

* Retail units, Luxonis knows definitely the cameras and their features. depthai-core must correctly indicate FF or AF functionality for all models.
* FFC boards are custom solutions assembled by the "someone" I write above. "Someone#2" writes custom code for their custom FFC board. Therefore, their entire team knows definitively what cameras are attached. They are responsible for either: 1) saving values to firmware (so that `.hasAutofocus` works correctly, or 2) not inspecting `.hasAutofocus` since they already know if their camera is FF or AF.

### testing

The short test proposed by @themarpe has no testing from hardware to the api. Meaning, it does not test that values are always correctly saved into firmware for all models. And that those values are then correctly transformed into a FF or AF boolean. One way to achieve this is to have the mapping of specific OAK models to specific FF/AF values. Then when you test case is run, the test must identify the OAK model, use the lookup table, and then check if the FF/AF api boolean matches hte value in the lookup table. If it is an FFC boards, then this can be handles by the testing environment knowing what it attached via a environment variable or similar method. Your CI knows what device it attached because someone physically attached it. So an environment variable for that test CI can also be set.