luxonis/depthai-core #864: UART Serial read / read_until doesn't work as expected

RichardHMaxwell, 2023-07-03:
I'm trying to read data in a python ScriptNode via UART, and the read and read_line functions return immediately rather than blocking until the configured timeout is reached. By default, the timeout is None, which means that those pySerial functions [should block until the requested number of bytes are available](https://pyserial.readthedocs.io/en/latest/pyserial_api.html) / until a line feed is received. I have tested this on both an OAK-FFC 4p and OAK-PoE 1p using the default UART settings on both the transmitter and receiver (i.e. 1 stop bit, no flow control or parity bits). To reproduce, you can run the following script:

```
    import serial
     
    ser = serial.Serial("/dev/ttyS0", baudrate=115200, timeout=None)
    while True:
        result = ser.read_until()
        node.warn(f"received {len(result)} bytes: {str(result.decode(encoding='ascii',errors='ignore'))}")
```

If nothing is plugged into the UART pins you will see the script print 1000s of messages per second, and always 0 bytes read. If you actually send a line feed terminated message over UART you will see that your message is broken over several log lines rather than just the expected single line - and of course there will still be 1000s of 0 byte messages in between.

Note that if you set timeout=1 (1 second according to the pySerial docs), the behaviour is approximately the same as above, however if you set much larger values, then the results are strange: with timeout=1000 the call blocks for ~1.6 seconds, and for timeout=100 it blocks for 23 seconds 🤷‍♂️

I also noticed that on two occasions that the very first call to read_until would block indefinitely even when the TX side was sending data.

PS: the docs don't say that it is pySerial that is provided by the script node environment, or which version of the library is being used. I'm just assuming this is the case based on the module and method names.

themarpe, 2023-07-03:
@RichardHMaxwell 

Its using VTime impl, based off of 3.0.1 pyserial.

Its odd that read_until does not function properly. What does `read` return when no characters are sent? Just a None?

Given the implementation of `read_until` it might be best if it was redone on your end as such:

```
def read_until(self, terminator=LF, size=None, timeout=None):
    """\
    Read until a termination sequence is found ('\n' by default), the size
    is exceeded or until timeout occurs.
    """
    tstart = time.time()
    lenterm = len(terminator)
    line = bytearray()
    while True:
        c = self.read(1)
        if c:
            line += c
            if line[-lenterm:] == terminator:
                break
            if size is not None and len(line) >= size:
                break
        if timeout is not None and time.time() - tstart > timeout:
             break
    return bytes(line)
```

This should loop until the actual `read_until` gets fully complete.

There are 2 options left that you may test:
 - Setting timeout to `timeout=-1`
 - Setting timeout to `timeout=-1/1000`

---

Unfortunatelly actual impl. on device will not likely change

RichardHMaxwell, 2023-07-04:
@themarpe 

> What does read return when no characters are sent? Just a None?

It returns an empty bytes object.

> There are 2 options left that you may test:
> 
> Setting timeout to timeout=-1
> Setting timeout to timeout=-1/1000

These throw a ValueError because they are negative.

Thanks for providing an implementation of read_until. I suppose the main issue would be high CPU usage, but you can work around that by sleeping for a few millis on `ser.inWaiting() < 1` , although this may slightly lower throughput.