luxonis/depthai-core #866: [BUG] BMI270 Latency issue

ibaiGorordo, 2023-07-04:
Hi,

I have noticed there is some latency (~80 ms) on my IMU streams (with respect to the latest RGB frame time) when I use an OAK-D Wide that has a BMI270 IMU. But when I test the same code using an OAK-D IOT with a BNO086, the IMU times are properly ahead of the RGB times.

It is probably related to #684 since the latency issue is fixed (latency becomes negative) when using depthai==2.20.2.0 but the issue happens when using the latest version.

Will there be a way to have a stable sampling frequency and low latency using the BMI270?

Thanks!

**Code:**
```python
#!/usr/bin/env python3

import cv2
import depthai as dai

def timeDeltaToMilliS(delta) -> float:
    return delta.total_seconds() * 1000

# Create pipeline
pipeline = dai.Pipeline()

# Define sources and outputs
imu = pipeline.create(dai.node.IMU)
imu.enableIMUSensor(dai.IMUSensor.ACCELEROMETER_RAW, 400)
imu.enableIMUSensor(dai.IMUSensor.GYROSCOPE_RAW, 400)
imu.setBatchReportThreshold(10)
imu.setMaxBatchReports(10)

xlinkOut = pipeline.create(dai.node.XLinkOut)
xlinkOut.setStreamName("imu")
imu.out.link(xlinkOut.input)

camRgb = pipeline.create(dai.node.ColorCamera)
camRgb.setResolution(dai.ColorCameraProperties.SensorResolution.THE_1080_P)
camRgb.setInterleaved(False)
camRgb.setColorOrder(dai.ColorCameraProperties.ColorOrder.RGB)

xoutRgb = pipeline.create(dai.node.XLinkOut)
xoutRgb.setStreamName("rgb")
camRgb.isp.link(xoutRgb.input)

monoLeft = pipeline.create(dai.node.MonoCamera)
monoRight = pipeline.create(dai.node.MonoCamera)
depth = pipeline.create(dai.node.StereoDepth)
monoLeft.setResolution(dai.MonoCameraProperties.SensorResolution.THE_400_P)
monoLeft.setCamera("left")
monoRight.setResolution(dai.MonoCameraProperties.SensorResolution.THE_400_P)
monoRight.setCamera("right")

depth.setDefaultProfilePreset(dai.node.StereoDepth.PresetMode.HIGH_DENSITY)
depth.initialConfig.setMedianFilter(dai.MedianFilter.KERNEL_7x7)
depth.setLeftRightCheck(True)
monoLeft.out.link(depth.left)
monoRight.out.link(depth.right)

xout = pipeline.create(dai.node.XLinkOut)
xout.setStreamName("disparity")
depth.disparity.link(xout.input)

# Pipeline is defined, now we can connect to the device
with dai.Device(pipeline, dai.UsbSpeed.SUPER_PLUS) as device:

    # Output queue for imu bulk packets
    imuQueue = device.getOutputQueue(name="imu", maxSize=50, blocking=False)
    rgbQueue = device.getOutputQueue(name="rgb", maxSize=4, blocking=False)
    dispQueue = device.getOutputQueue(name="disparity", maxSize=4, blocking=False)

    imuTs = None
    rgbTs = None
    while True:

        if imuQueue.has():
            imuData = imuQueue.get()  # blocking call, will wait until a new data has arrived
            imuPackets = imuData.packets[-1]
            imuTs = imuPackets.acceleroMeter.getTimestampDevice()

        if rgbQueue.has():
            rgbData = rgbQueue.get()  # blocking call, will wait until a new data has arrived
            rgbTs = rgbData.getTimestampDevice()

        if imuTs is None or rgbTs is None:
            continue
        latency = timeDeltaToMilliS(rgbTs - imuTs)
        tsF  = "{:.03f}"
        print(f"Latency: {tsF.format(latency)} ms")

        if cv2.waitKey(1) == ord('q'):
            break
```

JanLipovsek, 2023-07-06:
Hi @ibaiGorordo we have found and fixed the latency issue on the BMI270 IMU on the latest develop branch.

ibaiGorordo, 2023-07-07:
Hi @JanLipovsek ,

I just tested it and it works. For anyone reading this in the future, with the batch threshold at 10, some of the IMU packets had still about 30 ms latency, but using `imu.setBatchReportThreshold(5)` improved a lot and only a few packets have positive latency but very small.

Thanks!

closed 2023-07-07 by ibaiGorordo