luxonis/depthai-core #868: [BUG] IMX462 anti-banding appears to cause exposure to cycle each second

atmccarthy, 2023-07-05:
Notice in the following video how the exposure cycles approximately once per second:

https://github.com/luxonis/depthai-core/assets/23569342/f02c966b-7f25-4605-800f-bf62bcdcf45a

Anti-banding also doesn't work in some scenes: in the following video, the antibanding works in all camera angles except when pointed at the ground:

https://github.com/luxonis/depthai-core/assets/23569342/eab0bb5c-e32e-4351-b101-f7937615bf67

themarpe, 2023-07-27:
Hi @atmccarthy 

To be addressed in August, as we further our IMX462 support

themarpe, 2023-10-05:
CC: @alex-luxonis on above if any modifications made it in

alex-luxonis, 2023-10-31:
@atmccarthy The antibanding flashing issue was caused by exposure and ISO not having the same effect on sensor, e.g. `setExposure(4000, 100)` resulted in a different brightness than `setExposure(1000, 400)`, with the latter being darker. 
With antibanding, the exposure is set in multiples of half of powerline period, that is:
- for 60Hz (US, etc): 8.333, 16.667, 25, ... [ms]
- for 50Hz (Europe, etc): 10, 20, 30, ... [ms]

and at the transition between exposures, it's expected that ISO should be increased to compensate for. If exposure is currently (30000, 100) and the scene gets brighter, AE would go to the equivalent (20000, 150) and further lower ISO, etc.

We found that ISO was affected, and the gain applied appeared to be 1/3 of expected. Double-checked the formula in sensor datasheet, but couldn't find what we did wrong. The fix for now was just to multiply raw gain x3, and the branch imx283 was already merged to `develop`.

If you still have issues with anti-banding, please check that the scene isn't too bright, as IMX462 is quite sensitive, and AE would force exposure to go lower than the minimum required (half of powerline period). You can check the values by running https://github.com/luxonis/depthai-python/blob/main/utilities/cam_test.py with the argument `-show`, or toggle at runtime with `/` key.

Not much can be done in this case, but setting the FPS as frame period to be a multiple of half of powerline frequency can alleviate the flickering observed (by "syncing" with the flicker rate), however banding may still be seen on rolling shutters.

atmccarthy, 2023-11-02:
@alex-luxonis I've tested out the latest on the imx283 branch, and the issue is resolved :) Thanks!

Do i still need to be on the imx283 branch in order to use the imx462? Are there plans to roll it into master/stable releases?

themarpe, 2023-11-02:
@atmccarthy 
Yes, branch is already in develop as per: https://github.com/luxonis/depthai-python/issues/725#issuecomment-1788110173

Next slated release doesn't have a date set just yet, but we are looking into adding some additional features and then releasing.

atmccarthy, 2023-11-02:
:+1: cool, i'll close this issue as fixed.

closed 2023-11-02 by atmccarthy