luxonis/depthai-core #871: Dynamic library build error

razanhuss, 2023-07-06:
on the current latest version v2.22.0

Ubuntu 20.04.6
 
I want to integrate this lib with my project so

I tried to build a dynamic version of the library as it is mentioned in the readme file

### Dynamic library

To build dynamic version of library configure with following option added
```
cmake -S. -Bbuild -D'BUILD_SHARED_LIBS=ON'
cmake --build build
```

but I got the following error:

`depthai-core$ sudo cmake --build build
-- Using existing generated toolchain
-- Using toolchain file: /home/razan/Desktop/VGD/virtual-guidedog/Localization/ORB_SLAM3/Thirdparty/depthai-core/build/generated/toolchain.cmake
-- [hunter] Calculating Toolchain-SHA1
-- [hunter] Calculating Config-SHA1
-- [hunter] HUNTER_ROOT: /root/.hunter
-- [hunter] [ Hunter-ID: cb0ea1f | Toolchain-ID: a3a48bb | Config-ID: 66f7275 ]
-- [hunter] NLOHMANN_JSON_ROOT: /root/.hunter/_Base/cb0ea1f/a3a48bb/66f7275/Install (ver.: 3.9.1)
-- [hunter] XLINK_ROOT: /root/.hunter/_Base/cb0ea1f/a3a48bb/66f7275/Install (ver.: luxonis-2021.4.2-develop)
-- [hunter] BZIP2_ROOT: /root/.hunter/_Base/cb0ea1f/a3a48bb/66f7275/Install (ver.: 1.0.8-p0)
-- [hunter] FP16_ROOT: /root/.hunter/_Base/cb0ea1f/a3a48bb/66f7275/Install (ver.: luxonis-0.0.0)
-- [hunter] LIBARCHIVE-LUXONIS_ROOT: /root/.hunter/_Base/cb0ea1f/a3a48bb/66f7275/Install (ver.: hunter-3.5.2)
-- [hunter] SPDLOG_ROOT: /root/.hunter/_Base/cb0ea1f/a3a48bb/66f7275/Install (ver.: 1.8.2)
-- [hunter] ZLIB_ROOT: /root/.hunter/_Base/cb0ea1f/a3a48bb/66f7275/Install (ver.: 1.2.11-p2)
-- [hunter] BACKWARD_ROOT: /root/.hunter/_Base/cb0ea1f/a3a48bb/66f7275/Install (ver.: 1.6)
-- [hunter] LIBNOP_ROOT: /root/.hunter/_Base/cb0ea1f/a3a48bb/66f7275/Install (ver.: 1.0-ec8f75a)
-- Checking file: /home/razan/Desktop/VGD/virtual-guidedog/Localization/ORB_SLAM3/Thirdparty/depthai-core/shared/depthai-shared/src/datatype/DatatypeEnum.cpp
-- Checking file: /home/razan/Desktop/VGD/virtual-guidedog/Localization/ORB_SLAM3/Thirdparty/depthai-core/shared/depthai-shared/src/utility/Checksum.cpp
-- Performing Test FLAG_-Wformat=2
-- Performing Test FLAG_-Wformat=2 - Success
-- Performing Test FLAG_-Werror=self-assign-field
-- Performing Test FLAG_-Werror=self-assign-field - Failed
-- Performing Test FLAG_-Werror=unused-lambda-capture
-- Performing Test FLAG_-Werror=unused-lambda-capture - Failed
-- Performing Test FLAG_-Werror=return-type
-- Performing Test FLAG_-Werror=return-type - Success
-- Performing Test FLAG_-Werror=non-virtual-dtor
-- Performing Test FLAG_-Werror=non-virtual-dtor - Success
-- Performing Test FLAG_-Werror=sign-compare
-- Performing Test FLAG_-Werror=sign-compare - Success
-- Performing Test FLAG_-Werror=reorder
-- Performing Test FLAG_-Werror=reorder - Success
-- Performing Test FLAG_-Werror=switch-enum
-- Performing Test FLAG_-Werror=switch-enum - Success
-- Compiling depthai-core resources in PATCH_ONLY mode
-- Downloading Depthai device side binaries from server...
-- Downloading depthai and patch
-- commit: aa3e0564c0de3ef66cb6b240ff2b65ed3ed70aba
-- File already downloaded (resources): depthai-shared-commit-hash-a555263a9dbff1166088bcb758ab4306dca9ae1a.txt
CMake Warning at cmake/DepthaiDownloader.cmake:201 (message):
  depthai-shared codebases differ between device and host.  Enforce (CI):
  (device: 52fb1b4c0ada7d57c9999822a6b8f4c81bf6932b, host:
  aa3e0564c0de3ef66cb6b240ff2b65ed3ed70aba
Call Stack (most recent call first):
  CMakeLists.txt:338 (DepthaiDownload)

-- Downloading and checking depthai-device-fwp.tar.xz
-- File already downloaded (resources): depthai-device-fwp-a555263a9dbff1166088bcb758ab4306dca9ae1a.tar.xz
-- Downloading depthai bootloader
-- folder: /home/razan/Desktop/VGD/virtual-guidedog/Localization/ORB_SLAM3/Thirdparty/depthai-core/build/resources
-- maturity: release
-- commit_version_arg: 0.0.21
[... 77 lines cut ...]
[ 78%] Building CXX object CMakeFiles/depthai-core.dir/src/pipeline/datatype/StereoDepthConfig.cpp.o
[ 79%] Building CXX object CMakeFiles/depthai-core.dir/src/pipeline/datatype/EdgeDetectorConfig.cpp.o
[ 81%] Building CXX object CMakeFiles/depthai-core.dir/src/pipeline/datatype/TrackedFeatures.cpp.o
[ 82%] Building CXX object CMakeFiles/depthai-core.dir/src/pipeline/datatype/FeatureTrackerConfig.cpp.o
[ 83%] Building CXX object CMakeFiles/depthai-core.dir/src/utility/Initialization.cpp.o
[ 85%] Building CXX object CMakeFiles/depthai-core.dir/src/utility/Resources.cpp.o
[ 86%] Building CXX object CMakeFiles/depthai-core.dir/src/utility/Path.cpp.o
[ 87%] Building CXX object CMakeFiles/depthai-core.dir/src/utility/Platform.cpp.o
[ 89%] Building CXX object CMakeFiles/depthai-core.dir/src/utility/Environment.cpp.o
[ 90%] Building CXX object CMakeFiles/depthai-core.dir/src/xlink/XLinkConnection.cpp.o
[ 91%] Building CXX object CMakeFiles/depthai-core.dir/src/xlink/XLinkStream.cpp.o
[ 93%] Building CXX object CMakeFiles/depthai-core.dir/src/openvino/OpenVINO.cpp.o
[ 94%] Building CXX object CMakeFiles/depthai-core.dir/src/openvino/BlobReader.cpp.o
[ 95%] Building C object CMakeFiles/depthai-core.dir/src/bspatch/bspatch.c.o
[ 97%] Linking CXX shared library libdepthai-core.so
/usr/bin/ld: /root/.hunter/_Base/cb0ea1f/a3a48bb/66f7275/Install/lib/libspdlog.a(spdlog.cpp.o): relocation R_X86_64_TPOFF32 against `_ZGVZN6spdlog7details2os9thread_idEvE3tid' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /root/.hunter/_Base/cb0ea1f/a3a48bb/66f7275/Install/lib/libXLink.a(PlatformDeviceFd.cpp.o): relocation R_X86_64_PC32 against symbol `_ZNSt13unordered_mapImPvSt4hashImESt8equal_toImESaISt4pairIKmS0_EEED1Ev' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/depthai-core.dir/build.make:1037: libdepthai-core.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:513: CMakeFiles/depthai-core.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
`

Can anyone help with this, please? Thanks :)

themarpe, 2023-07-06:
@razanhuss remove the Hunter cache as per readme and retry. That should address this issue

razanhuss, 2023-07-07:
@themarpe thanks for your reply, I have already run this command `rm -rf ~/.hunter` to remove the hunter cache and deleted the build folder then tried to build it again and had the same error.
any other suggestions?

themarpe, 2023-07-07:
@razanhuss what is you platform?

Try `-D HUNTER_ROOT=[path to build dir]/.hunter` along the CMake args, and before remove the build directory

themarpe, 2023-07-07:
```/root/.hunter/```

Perhaps you've build with sudo?

remove the exact above dir:
```
rm -rf /root/.hunter/
```

And try again.

Then try the above suggestion, but it will probably not be needed

razanhuss, 2023-07-12:
I successfully resolved the issue by updating the hunter config file. Should I proceed with pushing the updated file to the repository, or would it be preferable for someone from the DepthAI development team to handle this task?

razanhuss, 2023-07-12:
@themarpe  [dynamic_build_fix_patch.zip](https://github.com/luxonis/depthai-core/files/12026127/dynamic_build_fix_patch.zip)
please find my fix in the patch file, it is to apply on the top of the main branch commit [82ab07d037]

themarpe, 2023-07-13:
@razanhuss please create a fork and open a PR against `develop` branch

Thought the patch as is, likely won't be accepted - setting `CMAKE_POSITION_INDEPENDENT_CODE=ON` already happens with `pic.cmake` toolchain that is applied when building a shared library. 

Reason why that isn't working, has to be addressed instead. Unless we forego the pic toolchain altogether and just keep code as -fPIC always (would address some hunter issues as well, which might be preffered)

razanhuss, 2023-08-07:
If you are suggesting to add the following lines at the beginning of the CMake file:
```
set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -fPIC")
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fPIC")
```
Unfortunately, this approach didn't resolve the build error. I have tried various methods, but the only change that effectively fixed the build error was updating the config.cmake for zlib as shown below:

```
hunter_config(
    ZLIB
    VERSION "1.2.11-p2"
    URL "https://github.com/luxonis/zlib/archive/refs/tags/v1.2.11-p2.tar.gz"
    SHA1 "fb8b6486183b13a86040f793a939b128f6d27095"
    CMAKE_ARGS
        CMAKE_POSITION_INDEPENDENT_CODE=ON  # this line to enable -fPIC flag
)
```

By including CMAKE_POSITION_INDEPENDENT_CODE=ON in the hunter_config for zlib, the necessary -fPIC flag is applied to all relevant code in the project and its dependencies, effectively resolving the build error and allowing successful creation of shared libraries.

themarpe, 2023-08-07:
@razanhuss - this is what happens using the pic.cmake toolchain. Are you using your own toolchain file? If so, make sure that that flag (as in pic.cmake) is applied as well

razanhuss, 2023-08-07:
no, I am not using any other than the toolchain file in the repo whose content is
```
# Simple PIC toolchain, which enables shared library building
# This option is propagated to 3rdparty dependecies as well
set(CMAKE_POSITION_INDEPENDENT_CODE ON)
```

razanhuss, 2023-08-07:
I tried it now and this seems to be working, but do you believe this is a solution?

themarpe, 2023-08-07:
@razanhuss when `BUILD_SHARED_LIBS=ON` this CMake toolchain should be auto applied. I think there might be some caches at play that weren't removed and it caused this. You may retry by deleting both: build folder, as well as hunter (`~/.hunter`), and then set just build shared libs paramter instead of explicitly setting the cmake toolchain as well.

If the issue remains, then we'll need to check (also, this must be on either latest `main` or `develop` - there were some changes compared to previous versions)

razanhuss, 2023-08-09:
I am attempting to integrate the depthai-core library into my main project by following the instructions provided in the README file. The method I'm using involves the add_subdirectory approach. However, even after following the steps outlined in the instructions, I am encountering build errors.

```
According to the README:

#### Using add_subdirectory

This method is more intrusive but simpler as it doesn't require building the library separately.

Add `add_subdirectory` which points to the `depthai-core` folder **before** the `project` command. Then link to any required targets.

add_subdirectory(${CMAKE_CURRENT_LIST_DIR}/depthai-core EXCLUDE_FROM_ALL)
...
project(my-app)
...
target_link_libraries([my-app] PRIVATE depthai::opencv)
```
I have made sure to place the add_subdirectory command before the project command in my main project's CMakeLists.txt file, as instructed. However, despite these efforts, the build error persists.

I have successfully built the depthai-core library separately using the commands:

```
cmake -S. -Bbuild -D'DEPTHAI_BUILD_EXAMPLES=ON'
cmake --build build
```

However, the build error only occurs when attempting to integrate it into my main project as per the README instructions.

Could you kindly assist me in identifying the cause of this build error and suggest any potential solutions?

themarpe, 2023-08-09:
@razanhuss are the errors the same as the ones posted above?

Solution is adding the CMAKE_TOOLCHAIN_FILE=[pic.cmake] above add_subdirectory or through parameters. Or linking statically (AKA not using BUILD_SHARED_LIBS=ON)