luxonis/depthai-core #875: [Feature-Request] Depth image compression by colorization

borongyuan, 2023-07-16:
### Start with the `why`:
VideoEncoder only supports NV12 or GRAY8 format. It cannot be used to encode depth images, only disparity images with subpixel disabled. This creates a big headache for us with PoE models. We hope to be able to transmit depth images with high resolution, high precision and high frame rate. The approach presented in [this RealSense white paper](https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras) seems promising. Hope it works for OAK cameras.

### Move to the `what`:
Add depth image compression by colorization feature.

### Move to the `how`:
The basic idea is to colorize the 16-bit depth image or disparity image according to the method mentioned in the white paper. Then use VideoEncoder for compressed transport. The recovery part on the Host side should be relatively easy to implement. The main feature that needs to be added is colorization on the camera. Currently ImageManip does not support color conversion of 16-bit images. Perhaps this particular color conversion method should be added to ImageManip, or implemented elsewhere.

borongyuan, 2024-07-26:
Any plan for this? It should be possible to create a custom blob and use the new Cast node. But if you can add this to ImageManip node, I'll now waste time on this.

themarpe, 2024-07-26:
CC: @moratom on whats the applicability on RVC2

borongyuan, 2024-07-27:
If we use disparityWidth of 96 and subpixelFractionalBits of 4, the max disparity should be 1520. This colorization method provides 1529 discrete levels. So there should be no quantization error, and normalization step might be skipped.

borongyuan, 2024-10-30:
Another solution is to use the [RVL](https://www.microsoft.com/en-us/research/publication/fast-lossless-depth-image-compression/) depth image compression algorithm. [RVL codec](https://github.com/ros-perception/image_transport_plugins/blob/jazzy/compressed_depth_image_transport/src/rvl_codec.cpp) has been added to ROS since Noetic/Jazzy. We are also considering adding it to RTAB-Map.