luxonis/depthai-core #876: [BUG] {cmake blocked}

CraneShen-0928, 2023-07-17:
E:\depthai-core>cmake -S. -Bbuild
-- Building for: Visual Studio 17 2022
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Using existing generated toolchain
-- Using toolchain file: E:/depthai-core/build/generated/toolchain.cmake
CMake Deprecation Warning at C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_apply_gate_settings.cmake:4 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
Call Stack (most recent call first):
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_finalize.cmake:5 (include)
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_add_package.cmake:7 (include)
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/Hunter:34 (include)
  cmake/HunterGate.cmake:536 (include)
  CMakeLists.txt:34 (HunterGate)

CMake Deprecation Warning at C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_calculate_config_sha1.cmake:4 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
Call Stack (most recent call first):
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_apply_gate_settings.cmake:8 (include)
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_finalize.cmake:5 (include)
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_add_package.cmake:7 (include)
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/Hunter:34 (include)
  cmake/HunterGate.cmake:536 (include)
  CMakeLists.txt:34 (HunterGate)

CMake Deprecation Warning at C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_lock_directory.cmake:4 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
Call Stack (most recent call first):
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_calculate_config_sha1.cmake:9 (include)
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_apply_gate_settings.cmake:8 (include)
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_finalize.cmake:5 (include)
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_add_package.cmake:7 (include)
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/Hunter:34 (include)
  cmake/HunterGate.cmake:536 (include)
  CMakeLists.txt:34 (HunterGate)

CMake Deprecation Warning at C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_make_directory.cmake:4 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
Call Stack (most recent call first):
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_calculate_config_sha1.cmake:10 (include)
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_apply_gate_settings.cmake:8 (include)
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_finalize.cmake:5 (include)
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_add_package.cmake:7 (include)
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/Hunter:34 (include)
  cmake/HunterGate.cmake:536 (include)
  CMakeLists.txt:34 (HunterGate)

CMake Deprecation Warning at C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_lock_directory.cmake:4 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
[... 640 lines cut ...]
  CMake that the project does not need compatibility with older versions.
Call Stack (most recent call first):
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_save_to_cache.cmake:8 (include)
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_download.cmake:26 (include)
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/projects/nlohmann_json/hunter.cmake:6 (include)
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_add_package.cmake:62 (include)
  cmake/depthaiDependencies.cmake:11 (hunter_add_package)
  CMakeLists.txt:167 (include)

CMake Deprecation Warning at C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_lock_directory.cmake:4 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
Call Stack (most recent call first):
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_make_directory.cmake:7 (include)
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_save_to_cache.cmake:8 (include)
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_download.cmake:26 (include)
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/projects/nlohmann_json/hunter.cmake:6 (include)
  C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_add_package.cmake:62 (include)
  cmake/depthaiDependencies.cmake:11 (hunter_add_package)
  CMakeLists.txt:167 (include)

-- [hunter] NLOHMANN_JSON_ROOT: C:/.hunter/_Base/cb0ea1f/e291960/900ca43/Install (ver.: 3.9.1)  --> **blocked here**

themarpe, 2023-07-17:
@CraneShen-0928 have you tried CTRL+C and retrying? 

We have observed this, but it went through next time. 
Also, it seems you have an old CMake version, I'd suggest you update to latest

Jason-xy, 2023-08-21:
when I run cmake with `-DHUNTER_STATUS_DEBUG=ON`, it seems there are something wrong with nlohmann_json's DONE metafile. There are some terminal outpus below:
```
-- [hunter *** DEBUG *** 2023-08-21T16:13:40] Downloading DONE metafile (try #0 of 10):
-- [hunter *** DEBUG *** 2023-08-21T16:13:40]   https://raw.githubusercontent.com/cpp-pm/hunter-cache/master/a3a48bb/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2023-08-21T16:13:40]   -> /home/jason/.hunter/_Base/Cache/meta/a3a48bb/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2023-08-21T16:13:40] File not found
-- [hunter *** DEBUG *** 2023-08-21T16:13:40] Downloading DONE metafile (try #1 of 10):
-- [hunter *** DEBUG *** 2023-08-21T16:13:40]   https://raw.githubusercontent.com/cpp-pm/hunter-cache/master/a3a48bb/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2023-08-21T16:13:40]   -> /home/jason/.hunter/_Base/Cache/meta/a3a48bb/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
################################
# the cmake process was blocked here!!! 
################################
```

themarpe, 2023-08-21:
@Jason-xy I'd first suggest retrying (CTRL+C & retry), if that doesn't work, perhaps a clean of hunter cache (`rm -rf ~/.hunter`) and retry