luxonis/depthai-core #881: [BUG] Read access violation on pipeline creation

GarethWoodgrove, 2023-07-24:
I am trying to run the rgb_video sample code from the C++ rgb_video example in a standalone project to make sure I have the project set up properly for my own development. The project builds successfully, however once I try to run it I get a read access violation exception on the first attempt to instantiate the pipeline class, indicated below.
`#include <iostream>

// Includes common necessary includes for development using depthai library
#include "depthai/depthai.hpp"

int main() {
    // Create pipeline
    dai::Pipeline pipeline;
    auto dev_info = dai::DeviceInfo("10.10.80.25");
    // Define source and output
    auto camRgb = pipeline.create<dai::node::ColorCamera>(); // <------------- Here
    auto xoutVideo = pipeline.create<dai::node::XLinkOut>();

    //

    xoutVideo->setStreamName("video");

    // Properties
    camRgb->setBoardSocket(dai::CameraBoardSocket::CAM_A);
    camRgb->setResolution(dai::ColorCameraProperties::SensorResolution::THE_1080_P);
    camRgb->setVideoSize(1920, 1080);

    xoutVideo->input.setBlocking(false);
    xoutVideo->input.setQueueSize(1);

    // Linking
    camRgb->video.link(xoutVideo->input);

    // Connect to device and start pipeline
    dai::Device device(pipeline, dev_info);

    auto video = device.getOutputQueue("video");

    while (true) {
        auto videoIn = video->get<dai::ImgFrame>();

        // Get BGR frame from NV12 encoded video frame to show with opencv
        // Visualizing the frame on slower hosts might have overhead
        cv::imshow("video", videoIn->getCvFrame());

        int key = cv::waitKey(1);
        if (key == 'q' || key == 'Q') {
            return 0;
        }
    }
    return 0;
}`
The specific error message is displayed on line 1567 of the xhash file in the include directory of my visual studio installation of MSVC 14.36.32532 and is as follows:
Exception thrown: read access violation.
this->Vec.Mypair.Myval2.Myfirst was 0x11101110111016A.
`template <class _Keyty> //line 1563
    _NODISCARD _Hash_find_last_result<_Nodeptr> _Find_last(const _Keyty& _Keyval, const size_t _Hashval) const {
        // find the insertion point for _Keyval and whether an element identical to _Keyval is already in the container
        const size_type _Bucket = _Hashval & _Mask;
        _Nodeptr _Where         = _Vec._Mypair._Myval2._Myfirst[(_Bucket << 1) + 1]._Ptr; //<----Here
        const _Nodeptr _End     = _List._Mypair._Myval2._Myhead;
        if (_Where == _End) {
            return {_End, _Nodeptr{}};
        }

        const _Nodeptr _Bucket_lo = _Vec._Mypair._Myval2._Myfirst[_Bucket << 1]._Ptr;
        for (;;) {
            // Search backwards to maintain sorted [_Bucket_lo, _Bucket_hi] when !_Standard
            if (!_Traitsobj(_Keyval, _Traits::_Kfn(_Where->_Myval))) {
                if constexpr (!_Traits::_Standard) {
                    if (_Traitsobj(_Traits::_Kfn(_Where->_Myval), _Keyval)) {
                        return {_Where->_Next, _Nodeptr{}};
                    }
                }

                return {_Where->_Next, _Where};
            }

            if (_Where == _Bucket_lo) {
                return {_Where, _Nodeptr{}};
            }

            _Where = _Where->_Prev;
        }
    }`
I am using the prebuilt depthai v2.22.0 library found [here](https://github.com/luxonis/depthai-core/releases) and opencv 4.7.0 if that helps narrow things down. I can run the example *.exe just fine when building the entire depthai-core repository per the instructions with cmake in vscode which has me confused as it is the exact same code in both cases.

themarpe, 2023-07-27:
@GarethWoodgrove perhaps there are some changes between library build upfront vs your local compilation setup.

If I'm reading this correctly, if you build examples as part of depthai-core and compile everything from source, it works - right?

GarethWoodgrove, 2023-07-27:
@themarpe Yes, when I build everything together per the instructions in this repo using cmake, the examples work with some minor modification to point to the camera's IP address. However, now that it has all built, I am attempting to use the .lib files and include directories to build a standalone console application in visual studio using its project properties to link everything as opposed to cmake. While that project also builds successfully, when I try to run the exact same code as the example that just ran properly in the cmake version, I am getting the above error.

themarpe, 2023-07-30:
I see - @GarethWoodgrove  in this case you should cmopare the build flags that you use between each project. 
Make sure same std's are used, etc... there might be quite some parts to it, so a bit of research on it should be required to make sure things are binary compatible to be used together

lakshin-flight, 2024-08-13:
I had the same issue but updating the includes in my custom project for the latest depth ai core includes and it resolved the issue. The below project is a good starting point which has a VS solution with the required includes.

 https://github.com/luxonis/depthai-core-example