luxonis/depthai-core #885: [BUG] install_depthai.sh - 'libtbb2' replaced

negative-video, 2023-07-26:
Running the installation script on Ubuntu 23.04 using `sudo wget -qO- https://docs.luxonis.com/install_depthai.sh | bash` returns the following output before terminating:

> Package libtbb2 is not available, but is referred to by another package.
> This may mean that the package is missing, has been obsoleted, or
> is only available from another source
> However the following packages replace it:
>   libtbbmalloc2
> 
>E: Package 'libtbb2' has no installation candidate
> 
> Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 100 at line 116 
>
>
> Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 100 at line 192 

As the output implies, it sounds like the script at https://docs.luxonis.com/install_dependencies.sh should be modified within `readonly ubuntu_pkgs=(` on line 25 such that `libtbb2` is replaced with `libtbbmalloc2`.

wasauce, 2024-02-04:
@Erol444 what is the solution here? I am trying to run

`sudo curl -fL https://docs.luxonis.com/install_dependencies.sh | bash` on a Raspberry pi 5 - running bookworm - https://en.wikipedia.org/wiki/Raspberry_Pi_OS

and I get:

$ sudo curl -fL https://docs.luxonis.com/install_dependencies.sh | bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  5671  100  5671    0     0  30467      0 --:--:-- --:--:-- --:--:-- 30467
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package libtbb2 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libtbbmalloc2:armhf libtbbmalloc2

E: Package 'libtbb2' has no installation candidate
E: Unable to locate package libdc1394-22-dev

Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 100 at line 118

themarpe, 2024-02-04:
@jakaskerl do you mind addressing this one?

jakaskerl, 2024-02-05:
https://discuss.luxonis.com/d/3406-oak-d-s2-connect-with-raspberry-pi-4/4 should fix it for now. We'll update the script.