luxonis/depthai-core #886: Firmware Configuration of BMI270

borongyuan, 2023-07-30:
Hi,
I'm doing Allan variance analysis on OAK's IMU. I recorded 10 hours of rosbag for each OAK camera. Now I read ACCELEROMETER_RAW and GYROSCOPE_RAW at 100Hz. For BMI270, these seem to be the only two available. But since BMI270 has functions such as **_programmable filter_** and **_CRT_**, I'd like to confirm whether ACCELEROMETER_RAW and GYROSCOPE_RAW are **_real RAW data_**. And you have defined other items in depthhai.IMUSensor. I'd like to understand their purpose more clearly. Are these post-processing intended to be implemented by the IC of IMU or OAK's firmware?
Here is the data I measured from one device.
[image]
[image]
From the graph it looks like the filter is not enabled, or at least the filter has a high bandwidth. This is what we want! We don't want any latency brought by the built-in filter, our own algorithm will handle the noise propagation. But the random walk in the graph of gyroscope looks a bit weird. Not sure if this is related to the CRT.
```
#Accelerometer
accelerometer_noise_density: 0.0014050369537968298 # m/s^2/√Hz
accelerometer_random_walk: 7.268205240359128e-05 # m/s^3/√Hz

#Gyroscope
gyroscope_noise_density: 0.0001116082909576279 # rad/s/√Hz
gyroscope_random_walk: 1.0601803124823304e-05 # rad/s^2/√Hz
```
The noise density from datasheet is 160 ug/√Hz and 0.007 deg/s/√Hz (BW=100Hz), and is close to our measurement. Bias instability is not available from datasheet.

szabi-luxonis, 2023-07-30:
@borongyuan the following configuration is applied for BMI:
Accelerometer:
```
                /* Gravity range of the sensor (+/- 2G, 4G, 8G, 16G). */
                accelCfg.cfg.acc.range = BMI2_ACC_RANGE_8G;
                /* The bandwidth parameter is used to configure the number of sensor samples that are averaged
                * if it is set to 2, then 2^(bandwidth parameter) samples
                * are averaged, resulting in 4 averaged samples.
                * Note1 : For more information, refer the datasheet.
                * Note2 : A higher number of averaged samples will result in a lower noise level of the signal, but
                * this has an adverse effect on the power consumed.
                */
                accelCfg.cfg.acc.bwp = BMI2_ACC_NORMAL_AVG4;
```
Gyro:
```
                /* Gyroscope Angular Rate Measurement Range.By default the range is 2000dps. */
                gyroCfg.cfg.gyr.range = BMI2_GYR_RANGE_2000;
                /* Gyroscope bandwidth parameters. By default the gyro bandwidth is in normal mode. */
                gyroCfg.cfg.gyr.bwp = BMI2_GYR_NORMAL_MODE;
```

[This](https://github.com/luxonis/BMI270-Sensor-API/tree/fix_interrupt_mapping) is the driver we use in FW

borongyuan, 2023-07-31:
Thanks, this is helpfu.

closed 2023-07-31 by borongyuan