luxonis/depthai-core #887: non functioning camera port on RVC3 6P-FFC module

PeterQFR, 2023-08-08:
I'm having an issue where one of my cameras on a 6P-FFC module, particularly the Left camera socket is not getting data from the attached camera. The module seems to recognise there is a camera there, but when I add the camera using the depthai-python/utilities/cam_test.py it fails to produce images and blocks other cameras from retrieving their images. 

This is the output with DEPTHAI_DEBUG=1

```
$ python cam_test.py -cams left,m right,m rgb,m camd,m came,m -cres 800 -fps 4
[2023-07-13 17:37:44.206] [info] DEPTHAI_DEBUG enabled, lowered DEPTHAI_LEVEL to 'debug'
[2023-07-13 17:37:44.206] [debug] Python bindings - version: 2.19.1.0.dev+11a40ed426e44610d40f1a328eb6a054d9af313f from  build: 2023-07-10 16:43:48 +0000
[2023-07-13 17:37:44.206] [debug] Library information - version: 2.19.1, commit: 0ee1ebf76bfd8181b701dfcf781fb6402b26f0ff from 2023-07-10 16:08:00 +0000, build: 2023-07-10 16:28:42 +0000
[2023-07-13 17:37:44.208] [debug] Initialize - finished
Enabled cameras:
   left : mono
  right : mono
    rgb : mono
   camd : mono
   came : mono
DepthAI version: 2.19.1.0.dev+11a40ed426e44610d40f1a328eb6a054d9af313f
DepthAI path: /home/peter-laptop/depthai-python/venv/lib/python3.8/site-packages/depthai.cpython-38-x86_64-linux-gnu.so
left
right
rgb
camd
came
[2023-07-13 17:37:44.300] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.22.tar.xz' open: 2ms, archive read: 90ms
[2023-07-13 17:37:44.675] [debug] Resources - Archive 'depthai-device-fwp-b511dc1a23aa9016007f7d6e9979a5a322bcefbe.tar.xz' open: 2ms, archive read: 465ms
[2023-07-13 17:37:44.724] [debug] Device - OpenVINO version: 2022.1
[2023-07-13 17:37:44.724] [debug] Device - BoardConfig: {"camera":[],"emmc":null,"gpio":[],"logDevicePrints":true,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
0000: b9 0f b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be be bb 00 bb 00 be be be
0020: be be be 01 bb 00
[2023-07-13 17:37:44.735] [debug] OS version: , Gate version: 
[2023-07-13 17:37:44.735] [debug] DeviceGate createSession: {"fwp_version":"0.0.1+57109fe11bf03b4cb130b9afcc6a13246692b825","library_version":"2.19.1","name":"depthai_session","protected":true}
[2023-07-13 17:37:44.746] [debug] DeviceGate createSession response: {"connection_information":"","create_datetime":"2022-07-29T17:36:34.664436","destroy_datetime":null,"fwp_checksum":null,"fwp_exists":false,"fwp_version":"0.0.1+57109fe11bf03b4cb130b9afcc6a13246692b825","id":"13","library_version":"2.19.1","name":"depthai_session","owner":"","protected":true,"start_datetime":null,"state":"CREATED","stop_datetime":null}
[2023-07-13 17:37:45.665] [debug] DeviceGate upload fwp successful
[2023-07-13 17:37:46.178] [debug] DeviceGate start fwp successful
Connected cameras: [{socket: RGB/CENTER/CAM_A, sensorName: OV9282, width: -1, height: -1, orientation: AUTO, supportedTypes: [], hasAutofocus: 0, name: keembay_cam_a2}, {socket: LEFT/CAM_B, sensorName: OV9282, width: -1, height: -1, orientation: AUTO, supportedTypes: [], hasAutofocus: 0, name: keembay_camex_0_stereo}, {socket: RIGHT/CAM_C, sensorName: OV9282, width: -1, height: -1, orientation: AUTO, supportedTypes: [], hasAutofocus: 0, name: keembay_camex_1_stereo}, {socket: VERTICAL/CAM_D, sensorName: OV9282, width: -1, height: -1, orientation: AUTO, supportedTypes: [], hasAutofocus: 0, name: keembay_cam_c1}, {socket: CAM_E, sensorName: OV9282, width: -1, height: -1, orientation: AUTO, supportedTypes: [], hasAutofocus: 0, name: keembay_cam_c2}]
 -socket RGB   : OV9282   -1 x   -1 focus:fixed - 
 -socket LEFT  : OV9282   -1 x   -1 focus:fixed - 
 -socket RIGHT : OV9282   -1 x   -1 focus:fixed - 
 -socket CAM_D : OV9282   -1 x   -1 focus:fixed - 
 -socket CAM_E : OV9282   -1 x   -1 focus:fixed - 
USB speed: UNKNOWN
IR drivers: []
[2023-07-13 17:37:48.173] [debug] Schema dump: {"connections":[{"node1Id":10,"node1Output":"out","node1OutputGroup":"","node2Id":9,"node2Input":"in","node2InputGroup":""},{"node1Id":8,"node1Output":"out","node1OutputGroup":"","node2Id":7,"node2Input":"in","node2InputGroup":""},{"node1Id":6,"node1Output":"out","node1OutputGroup":"","node2Id":5,"node2Input":"in","node2InputGroup":""},{"node1Id":4,"node1Output":"out","node1OutputGroup":"","node2Id":3,"node2Input":"in","node2InputGroup":""},{"node1Id":2,"node1Output":"out","node1OutputGroup":"","node2Id":1,"node2Input":"in","node2InputGroup":""},{"node1Id":0,"node1Output":"out","node1OutputGroup":"","node2Id":10,"node2Input":"inputControl","node2InputGroup":""},{"node1Id":0,"node1Output":"out","node1OutputGroup":"","node2Id":8,"node2Input":"inputControl","node2InputGroup":""},{"node1Id":0,"node1Output":"out","node1OutputGroup":"","node2Id":6,"node2Input":"inputControl","node2InputGroup":""},{"node1Id":0,"node1Output":"out","node1OutputGroup":"","node2Id":4,"node2Input":"inputControl","node2InputGroup":""},{"node1Id":0,"node1Output":"out","node1OutputGroup":"","node2Id":2,"node2Input":"inputControl","node2InputGroup":""}],"globalProperties":{"calibData":null,"cameraTuningBlobSize":null,"cameraTuningBlobUri":"","leonCssFrequencyHz":700000000.0,"leonMssFrequencyHz":700000000.0,"pipelineName":null,"pipelineVersion":null,"xlinkChunkSize":-1},"nodes":[[10,{"alias":"","id":10,"ioInfo":[[["","out"],{"blocking":false,"group":"","name":"out","queueSize":8,"type":0,"waitForMessage":false}],[["","raw"],{"blocking":false,"group":"","name":"raw","queueSize":8,"type":0,"waitForMessage":false}],[["","frameEvent"],{"blocking":false,"group":"","name":"frameEvent","queueSize":8,"type":0,"waitForMessage":false}],[["","inputControl"],{"blocking":true,"group":"","name":"inputControl","queueSize":8,"type":3,"waitForMessage":false}]],"name":"MonoCamera","parentId":-1,"properties":{"boardSocket":4,"cameraName":"","fps":4.0,"imageOrientation":-1,"initialControl":{"aeLockMode":false,"aeRegion":{"height":0,"priority":0,"width":0,"x":0,"y":0},"afRegion":{"height":0,"priority":0,"width":0,"x":0,"y":0},"antiBandingMode":0,"autoFocusMode":3,"awbLockMode":false,"awbMode":0,"brightness":0,"chromaDenoise":0,"cmdMask":0,"contrast":0,"effectMode":0,"expCompensation":0,"expManual":{"exposureTimeUs":0,"frameDurationUs":0,"sensitivityIso":0},"frameSyncMode":0,"lensPosAutoInfinity":0,"lensPosAutoMacro":0,"lensPosition":0,"lowPowerNumFramesBurst":0,"lowPowerNumFramesDiscard":0,"lumaDenoise":0,"miscControls":[],"saturation":0,"sceneMode":0,"sharpness":0,"strobeConfig":{"activeLevel":0,"enable":0,"gpioNumber":0},"strobeTimings":{"durationUs":0,"exposureBeginOffsetUs":0,"exposureEndOffsetUs":0},"wbColorTemp":0},"numFramesPool":3,"numFramesPoolRaw":3,"resolution":1}}],[9,{"alias":"","id":9,"ioInfo":[[["","in"],{"blocking":true,"group":"","name":"in","queueSize":8,"type":3,"waitForMessage":true}]],"name":"XLinkOut","parentId":-1,"properties":{"maxFpsLimit":-1.0,"metadataOnly":false,"streamName":"came"}}],[8,{"alias":"","id":8,"ioInfo":[[["","out"],{"blocking":false,"group":"","name":"out","queueSize":8,"type":0,"waitForMessage":false}],[["","raw"],{"blocking":false,"group":"","name":"raw","queueSize":8,"type":0,"waitForMessage":false}],[["","frameEvent"],{"blocking":false,"group":"","name":"frameEvent","queueSize":8,"type":0,"waitForMessage":false}],[["","inputControl"],{"blocking":true,"group":"","name":"inputControl","queueSize":8,"type":3,"waitForMessage":false}]],"name":"MonoCamera","parentId":-1,"properties":{"boardSocket":3,"cameraName":"","fps":4.0,"imageOrientation":-1,"initialControl":{"aeLockMode":false,"aeRegion":{"height":0,"priority":0,"width":0,"x":0,"y":0},"afRegion":{"height":0,"priority":0,"width":0,"x":0,"y":0},"antiBandingMode":0,"autoFocusMode":3,"awbLockMode":false,"awbMode":0,"brightness":0,"chromaDenoise":0,"cmdMask":0,"contrast":0,"effectMode":0,"expCompensation":0,"expManual":{"exposureTimeUs":0,"frameDurationUs":0,"sensitivityIso":0},"frameSyncMode":0,"lensPosAutoInfinity":0,"lensPosAutoMacro":0,"lensPosition":0,"lowPowerNumFramesBurst":0,"lowPowerNumFramesDiscard":0,"lumaDenoise":0,"miscControls":[],"saturation":0,"sceneMode":0,"sharpness":0,"strobeConfig":{"activeLevel":0,"enable":0,"gpioNumber":0},"strobeTimings":{"durationUs":0,"exposureBeginOffsetUs":0,"exposureEndOffsetUs":0},"wbColorTemp":0},"numFramesPool":3,"numFramesPoolRaw":3,"resolution" [... cut ...]
When I exclude this port the system works reliably up to 4 cameras. I did have issues when attaching to port camf that would prevent other cameras from working. 

Even when connecting a single camera to the left socket it does not work.

I've rotated the cameras around and have confirmed that all the cameras do work and I've reattached a camera to the left socket a couple of times and am confident that that connection is a correct as I can make it. 

The previous discussion is [here](https://discuss.luxonis.com/d/1841-oak-ffc-6p-left-socket-not-working/3)

I have 5 cameras so let me know if there is some test you would like me to do.

Erol444, 2023-08-08:
(from support email)

@PeterQFR upgraded FIP to 1.5.0, which resulted in such image:
[image]

PeterQFR, 2023-08-08:
Not sure if upgrading to 1.5.0 resulted in that image. I don't really know what FIP version the FFC6P shipped with. That is from the OV9782 which was appearing programmatically as a OV9282. Other than the pattern in the image, I couldn't tell the difference between the OV9782 or the OV9282 programmatically or physically (ie there is no silkscreen label).. just wanted to clarify the above comment. The identification of the cameras is probably a separate issue.

PeterQFR, 2023-08-08:
So I've replugged the cameras back in. I can get all of them to work now, if I restrict the mres to 400. so for example 
```
$cam_test.py --cameras left,m -mres 400
```

The colour camera is still not producing a colour image. The above pattern is only visible when capturing full scale mono images (whatever the default resolution is)

I usually have issues getting all the mono cameras to produce images even though they individually do. I think this is a thing in polling the xlink queues. It works better if the images are smaller, you're probably aware of this.

alex-luxonis, 2023-08-08:
Currently OV9782 (handled as color instead of monochrome) is only possible by flashing a different FIP, as below (taken from https://discuss.luxonis.com/d/1550-oak-ffc-ov9782-w-not-giving-rgb-image)
> Flashing this FIP should make OV9282 or OV9782 on any sockets handled as color:
https://artifacts.luxonis.com/artifactory/tmp/fip-oak-ffc-6p-kb-1.5.0_OV9782.bin
`movisoc-fwu -a fip-oak-ffc-6p-kb-1.5.0_OV9782.bin`
>
> A note: MonoCamera won't work afterwards, as color/mono is set hardcoded in DTS. Can revert by flashing: https://luxonisos.fra1.digitaloceanspaces.com/release/recovery/fip-oak-ffc-6p-rvc3-1.2.0.bin
>
> We're also looking to make the color/mono selection possible with depthai, like on RVC2. It would probably require a Luxonis OS update.
>
> Another thing is frame syncing, for multiple ColorCamera is needs to be manually set for now, can configure one OV9782 to generate the sync signal, and the other(s) to receive it:
`colorLeft.initialControl.setFrameSyncMode(dai.CameraControl.FrameSyncMode.OUTPUT)`
`colorRight.initialControl.setFrameSyncMode(dai.CameraControl.FrameSyncMode.INPUT)`

PeterQFR, 2023-08-09:
Okay, so I shouldn't expect the colour sensor to work as colour for now? In general  if I am looking for reliable behaviour, would using the rvc2 be a better option if I am looking to move beyond bench testing and commence prototype integration? I've been using OAK-D Pro for a while now and it suits my needs. I read that the RVC2 is used on those devices but can I also just connect it to the 6P-FFC board?

Erol444, 2023-08-14:
Hi @PeterQFR ,
RVC2 is definitely more battle-tested, and RVC3 is still in early release, so not everything will work as expected (such as the issue above). Note that RVC2 isn't compatible with the 6P-FFC, and it supports max 4 cameras (hence FFC-4P). Thoughts?