luxonis/depthai-core #889: [BUG] Issues connecting to the PoE OAK S-2 Camera under Windows only (same program works under Linux)

mlockett42, 2023-08-15:
**Check if issue already exists**

I have been working on this issue for some time on the community forums see https://discuss.luxonis.com/d/1842-issues-shutting-down-and-restarting-connection-to-poe-s-2-camera/4 best to start from the comment "I have done this It upgraded from version 0.0.21 to version 0.0.26 but that appears to have made things worse"

That includes looking for other solutions and going through all the community level support.

I have been asked to escalate this issue from the community forum to the Github Issues

**Describe the bug**
My program using the OAK D S-2 camera, cannot connect to it. This program runs under Windows and will be deployed to a client site running on a Windows tablet.
To more clearly reproduce this (as it is just the connection) I wrote a minimal program to demostrate the issue. (See Minimal Reproducible Example below)
This program reliably fails under Windows and reliably works under Linux.
I am used a Dell Vostro 7590 running Dual Boot between Windows 10 and Linux Ubuntu 22.04 so the same hardware works on Linux not on Windows

The error message give under Windows is
```
[2023-07-26 14:29:02.667] [depthai] [warning] Monitor thread (device: 19443010B1937F1300 [169.254.1.222]) - ping was missed, closing the device connection

Traceback (most recent call last):

File "C:\code\cadds-visual-tyre-handler\source\recorder\main.py", line 118, in <module>

with dai.Device(pipeline, device_info, maxUsbSpeed=dai.UsbSpeed.HIGH) as device:

depthai.XLinkWriteError: Couldn't write data to stream: '__bootloader' (X_LINK_ERROR)
```

This error message was taken with the network in Link-Local mode, the same thing happens if we have DHCP set up.

I can ping the camera before starting the program to verify it is there.

This error only occurs with the PoE Camera, we also have a USB Camera which works fine and which we can use to develop the functionality of our application but the USB Camera is not suitable for production use for us.

**Minimal Reproducible Example**
[DepthAI Windows Issue Demo](https://github.com/mlockett42/depthai-windows-issue-demo)

**Expected behavior**
The program should start, open a window and begin streaming output from the RGB Camera to the window. Under both Windows and Linux

themarpe, 2023-08-15:
@mlockett42 
At the time being, you cannot access the device using `Device` through the LL address (`169.254.1.222` due to a bug).

We'll take a look at this and get back to you

mlockett42, 2023-08-31:
@asahtik Any news on this?

Anything I can do to help?

Erol444, 2023-08-31:
Hi @mlockett42 ,
I tried your code and it seems to work (for several minutes):
- OS: Windows 10 Pro (22H2)
- Python 3.9
- depthai  library version 2.22.0.0.dev0+ce2627c5aea3c23009da1d1f9b0546be6f8956e5
- Bootloader: Tried both 0.24 and 0.26 (latest)

I have connected Oak-D-S2 PoE directly to my computer, and powered it via M8 connector (Programming board), as I don't have an injector (only POE switch, but then it wouldn't be fallback static IP). Could you try to replicate these variables @mlockett42 ?

mlockett42, 2023-09-07:
@Erol444 
I tried those settings so now I have
OS Windows 10 Pro (22H2)
Python 3.9.13
Depthai-python installed directly from the file system as of commit ce2627c5aea3c23009da1d1f9b0546be6f8956e5. This is now implemented as a git submodule in the repo
I only have Bootloader 0.0.26 - previous attempts to downgrade failed.
It is powered by my PoE Switch and gets it's IP address via DHCP from my home router. My laptop is connected to the same PoE switch via an ethernet cable and USB to Ethernet Adaptor.

Install via a regular virtual environment instead of pipenv

I have new branch for this attempt-1-to-get-working

After a few attempts to get working and attempts to check I cleared out old build artifacts and the .hunter directory, I got it partially working.

Over 5 attempts
3 failed with the same error as before.
1 worked for a few seconds then gave a different error (RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Bad packet, couldn't parse (invalid message type)')
1 worked perfectly, but when I stopped it by shutting the program down normally I could not restart it.
I would power cycle the camera each time it failed to work but this seems to not have much impact.

mlockett42, 2023-10-19:
This issue has been solved by upgrading all the network equipment to gigabit. Gigabit PoE switch plus gigabit ethernet adaptor. I am guessing the ping packet is not received because the network link is overwhelmed with traffic.

Leaving comment here for any future users searching for this issue on google

closed 2023-10-19 by mlockett42