luxonis/depthai-core #893: I2C Nodes

RaymondCM, 2023-08-17:
Given the support of I2C on the 6P boards, we have multiple sensors integrated on the device which are now fully functional. We would like to ideally integrate these with depthai-core, so we can configure custom I2C nodes via depthai pipelines.

This allows a user to add I2C devices directly to the boards and have access to the data via the host system without the need to dump locally/send over usb etc.

I imagine the functionality would be similar to the SPI node/IMU node.

Is there a roadmap for this functionality? If we start development of this feature, what we should be aware of.

We're worried that if we send the data over the USB connection then on initialisation of a camera pipeline, this will take precedence and result in corrupt USB packets.

On another note - any suggestions on how best to achieve this outside of the code changes?

RaymondCM, 2023-08-17:
Given the other nodes utilise I2C, this should be a fairly quick addition! On the 6P boards exposing pins 30 and 31 in a default node with a hardware address flag should enable most of this functionality.

themarpe, 2023-08-17:
@RaymondCM on RVC3, I2C can be accessed through Linux operating system and the related tooling. Either bypassing DepthAI or through Script node.

Such node is not on roadmap for DepthAI

closed 2023-08-17 by RaymondCM