luxonis/depthai-core #895: Snapshot Mode for Arducam IMX296 on OAK-FFC-4P

rajsinghmn, 2023-08-23:
Hi, 

I'm trying to get snapshot triggered acquisition working for an IMX296 camera connected to an OAK-FFC-4P baseboard. My goal is to be able to send pulses at arbitrary times (and exposures) to control image acquisition. My approach was to use the example script below (modified from the example [here](https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/sync_frames/#synchronizing-frames-externally)) while supplying an external 5us 1.8v pulse to the headers directly on the camera boards. This seems to work fine when testing on OV9782 and AR0234 cameras, i.e. the cameras wait indefinitely for the pulse before capturing. With the IMX296 however, I get essentially continuous free-run despite applying a similar configuration. I'm using the XVS and GND headers on the IMX296 camera board as recommended by the [Arducam documentation](https://docs.arducam.com/Raspberry-Pi-Camera/Global-Shutter-Camera/external-trigger/). 

I see that IMX296 is listed as "fully integrated" on the list of [supported sensors](https://docs.luxonis.com/projects/hardware/en/latest/pages/articles/supported_sensors/). Is it possible that snapshot feature is not yet supported?

Example Script:
```
import cv2
import depthai as dai
import numpy as np
import time

# Create pipeline
pipeline = dai.Pipeline()

# Define sources and outputs
camRgb = pipeline.create(dai.node.ColorCamera)
camRgb.setBoardSocket(dai.CameraBoardSocket.CAM_A)
xoutRgb = pipeline.create(dai.node.XLinkOut)
xoutRgb.setStreamName("rgb")

# Properties
camRgb.setResolution(dai.ColorCameraProperties.SensorResolution.THE_1440X1080)
camRgb.setPreviewSize(1440, 1080)
camRgb.setInterleaved(False)
camRgb.setFps(40)
camRgb.initialControl.setFrameSyncMode(dai.CameraControl.FrameSyncMode.INPUT)
camRgb.initialControl.setExternalTrigger(2,1)      
camRgb.preview.link(xoutRgb.input)

# Connect to device and start pipeline
with dai.Device(pipeline) as device:
    
    qRgb = device.getOutputQueue(name="rgb", maxSize=4, blocking=False)
    frame = None   
    counter = 0 
    
    while True:                
        
        if qRgb.has():
            counter+=1
            inRgb = qRgb.get()                    
            print(f'{counter} frame(s) detected.')
            if inRgb is not None:
                frame = inRgb.getCvFrame()

        if frame is not None:
            cv2.imshow(name, frame)

        if cv2.waitKey(1) == ord('q'):
            break

```

l4es, 2023-09-09:
Hi @rajsinghmn,

For snapshot mode, you will need to put the IMX296 to slave mode explicitly by hardware configuring the relevant pin. 

Also, I think that IMX296 could work similarly to OV9782 in continuous mode for the stereo pair : one as master, other as slave and XVS is driven by the master. Once again, you will need to configure the relevant hardware pin to put the sensor into the slave mode explicitly.

Best Regards,
Khang

themarpe, 2023-09-09:
Support (with a note) ready in 47055d5270b13adbb740cb1a897b7f9fe7a98aa5

closed 2023-09-09 by themarpe