luxonis/depthai-core #903: [Feature-Request] Support IMX378 and 2x AR0234 Sensors

kneave, 2023-09-18:
### Start with the `why`:
I've designed a robot that uses a FFC-3P board with two AR0234 cameras with wide angle lenses to generate a depth map, and an IMX378 as a centre camera with a narrow FOV for more detailed imagery for object recognition. This means I can generate a coloured 3d point cloud of the wider area rather than being limited to the narrow field from the centre camera. Currently this is flagged as an unsupported combination as there is no model defined for it.

As per [this post](https://discuss.luxonis.com/d/2160-image-quality-in-mixing-imx477-with-ar0234-stereo-pairs)

This is a product prototype that I'm intending to make commercially available and hope to use the FFC-3P, or possible RVC4 powered successor, at it's core.

### Move to the `what`:
I would like a model generated from this camera combination, as I understand it this requires access to proprietary tools which aren't public.

### Move to the `how`:
As above, the tools are proprietary so I've not knowledge of how they work. If there was an option to share them, I'd be happy to help with this however.

themarpe, 2023-09-22:
Hi @kneave - please reachout to our support directly for this enabling this custom tuning combination

kneave, 2023-09-22:
Ah, ok, I was told to open a feature request on the forums. Want me to leave this open or close it?

themarpe, 2023-09-22:
Feel free to leave open for the time being

kneave, 2023-10-11:
Thanks, fingers crossed as it'd be handy having the IMX378 in the centre spot rather than needed to get a third AR0234!

closed 2023-11-03 by kneave