luxonis/depthai-core #907: Oak-ffc-4p-poe Having Configuration Issues

justin-larking-pk, 2023-09-22:
I've got a Oak-ff-c4p-poe board that I'm attempting to set a static IP on, this has brought up some issues the board doesn't seem to have any existing configuration, attempting to view the config produces:
[image]
Attempting to flash a new config produces:
[image]

I was informed that this could be fixed via a factory reset thus, I connected via usb-c, setting the device to usb boot mode and attempted to factory reset. However:
[image]

I then reconnected and attempted to just update the bootloader and it worked. Bootloader when I originally got the device was 0.0.22, device_manager updated it to 0.0.26.
[image]
I then switched back to poe boot and could see the bootloader has updated. 
[image]

**However even after the bootloader update viewing or flashing a config produces the exact same error as before.** Furthermore factory resetting also doesn't work producing the same error "Error Flashing". Is the device faulty? Whats going on here?

I cant flash an IP dynamic or static, factory resetting doesn't work. My depthai is the latest from pip.

themarpe, 2023-09-22:
@justin-larking-pk 

Do you mind updating IMU first and trying again? We've seen that sometimes these two might clash

justin-larking-pk, 2023-09-25:
This was it, flash the IMU firmware and then flashed a config and everything is working fine.

closed 2023-09-25 by justin-larking-pk