luxonis/depthai-core #915: [BUG] OAK-FFC-4P-POE Firmware Crash with Custom Camera Tuning Blob

justin-larking-pk, 2023-10-29:
**Describe the bug**
Firmware crash on Oak-ffc-4p-poe and crash dump after enabling custom camera tuning blob.

Specific error: "Error creating output stream: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'detections_l' (X_LINK_ERROR)'"

A oak-ffc-4p-poe was setup with two ar0234 m12 cameras. Previously the setup was functioning fine and operating a pipeline. However we require low exposure times thus attempted to use the 500us camera tuning blob (https://docs.luxonis.com/projects/api/en/latest/tutorials/image_quality/#:~:text=Camera%20exposure%20limit)

Simply adding this line: pipeline.setCameraTuningBlobPath(camera_blob_path); to our pipeline setup caused a firmware crash and a crash dump to be produced. 

[crashDump_0_18443010111CCC0F00_f033fd9c7eb0b3578d12f90302e87759c78cfb36.json](https://github.com/luxonis/depthai-core/files/13199275/crashDump_0_18443010111CCC0F00_f033fd9c7eb0b3578d12f90302e87759c78cfb36.json)

**Expected behavior**
The addition of the camera tuning blob would limit max exposure time to 500us on the oak-ffc-4p-poe with two aro234 cameras connected.

**Pipeline Graph**
Pieline details as json here: https://drive.google.com/file/d/1n3JYvl-YDbfGb_6GX9v8m6vP6wpP6CnH/view?usp=sharing 
[image]

justin-larking-pk, 2023-11-13:
Any update on this?

alex-luxonis, 2023-11-15:
Those tuning blobs had some issues with certain sensors. Recommended now to use the branch `ae_exp_limit` (PR https://github.com/luxonis/depthai-python/pull/921 to be merged soon) where a configurable exposure limit can be set (and updated at runtime as well if desired). It will also use the proper image quality tuning for the specific sensor (AR0234 in this case).