luxonis/depthai-core #938: [BUG] bug with log "Please report to developers."

LuoQuestionmark, 2023-12-11:
**Describe the bug**

While trying to connect with two PoE cameras, the following error occurs, and the connection is broken.

```log
[184430105145A20F00] [169.254.1.11] [6079.212] [system] [critical] Fatal error. Please report to developers. Log: 'pool' '180'
```

Another suspicious error is

```log
[1844301091838E0F00] [169.254.1.12] [1701967503.847] [host] [warning] Monitor thread (device: 1844301091838E0F00 [169.254.1.12]) - ping was missed, closing the device connection
```

which is yet another problem.

**Minimal Reproducible Example**

Have not written one yet.

**Expected behavior**

Lost connection with the reporting camera.

**Additional context**

I have reported the same error [here](https://github.com/luxonis/depthai/issues/1138) with no response. The bug seems to happen less frequently if:

- connection via PoE rather than via USB ;
- less image processing on each iteration ;
- connection with only one, not two cameras ;
- program running with 2.23 version of library instead of 2.22.

**Log**

The following log is from my program. Each record with a timestamp ahead is generated by my program; others are from the library.

```log
2023-12-07 17:21:15:321	[VIDEO]	camera try to connect

2023-12-07 17:21:15:322	[VIDEO]	camera try to connect

[184430105145A20F00] [169.254.1.11] [1701966075.323] [host] [warning] Using a custom watchdog value of 4500ms

[1844301091838E0F00] [169.254.1.12] [1701966075.325] [host] [warning] Using a custom watchdog value of 4500ms

[1844301091838E0F00] [169.254.1.12] [695.097] [system] [critical] Fatal error. Please report to developers. Log: 'pool' '180'

2023-12-07 17:33:01:279	[VIDEO]	Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'RGB' (X_LINK_ERROR)'

2023-12-07 17:33:01:780	[VIDEO]	camera try to connect

[1844301091838E0F00] [169.254.1.12] [1701966781.782] [host] [warning] Using a custom watchdog value of 4500ms

[1844301091838E0F00] [169.254.1.12] [1701967503.847] [host] [warning] Monitor thread (device: 1844301091838E0F00 [169.254.1.12]) - ping was missed, closing the device connection

2023-12-07 17:45:08:098	[VIDEO]	Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'DEPTH' (X_LINK_ERROR)'

2023-12-07 17:45:08:598	[VIDEO]	camera try to connect

[1844301091838E0F00] [169.254.1.12] [1701967508.600] [host] [warning] Using a custom watchdog value of 4500ms

[184430105145A20F00] [169.254.1.11] [1701972945.643] [host] [warning] Monitor thread (device: 184430105145A20F00 [169.254.1.11]) - ping was missed, closing the device connection

2023-12-07 19:15:48:099	[VIDEO]	Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'RGB' (X_LINK_ERROR)'

2023-12-07 19:15:48:599	[VIDEO]	camera try to connect

[184430105145A20F00] [169.254.1.11] [1701972948.601] [host] [warning] Using a custom watchdog value of 4500ms

[184430105145A20F00] [169.254.1.11] [12186.805] [system] [critical] Fatal error. Please report to developers. Log: 'pool' '180'

2023-12-07 22:39:10:348	[VIDEO]	Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'RGB' (X_LINK_ERROR)'

2023-12-07 22:39:10:849	[VIDEO]	camera try to connect

[184430105145A20F00] [169.254.1.11] [1701985150.851] [host] [warning] Using a custom watchdog value of 4500ms

[184430105145A20F00] [169.254.1.11] [3758.599] [system] [critical] Fatal error. Please report to developers. Log: 'pool' '180'

2023-12-07 23:42:03:987	[VIDEO]	Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'DEPTH' (X_LINK_ERROR)'

2023-12-07 23:42:04:488	[VIDEO]	camera try to connect

[184430105145A20F00] [169.254.1.11] [1701988924.490] [host] [warning] Using a custom watchdog value of 4500ms

[1844301091838E0F00] [169.254.1.12] [1701990470.066] [host] [warning] Monitor thread (device: 1844301091838E0F00 [169.254.1.12]) - ping was missed, closing the device connection

F: [global] [    472066] [Scheduler00Thr] tcpipPlatformWrite:300	Cannot find file descriptor by key: 5c

F: [global] [    472066] [EventRead00Thr] tcpipPlatformRead:272	Cannot find file descriptor by key: 5c

2023-12-08 00:07:54:317	[VIDEO]	Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'MONO' (X_LINK_ERROR)'

2023-12-08 00:07:54:817	[VIDEO]	camera try to connect

[1844301091838E0F00] [169.254.1.12] [1701990474.819] [host] [warning] Using a custom watchdog value of 4500ms

[184430105145A20F00] [169.254.1.11] [1701993678.627] [host] [warning] Monitor thread (device: 184430105145A20F00 [169.254.1.11]) - ping was missed, closing the device connection

2023-12-08 01:01:22:878	[VIDEO]	Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'MONO' (X_LINK_ERROR)'

2023-12-08 01:01:23:378	[VIDEO]	camera try to connect

[184430105145A20F00] [169.254.1.11] [1701993683.380] [host] [warning] Using a custom watchdog value of 4500ms

[184430105145A20F00] [169.254.1.11] [6079.212] [system] [critical] Fatal error. Please report to developers. Log: 'pool' '180'

2023-12-08 02:42:58:546	[VIDEO]	Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'RGB' (X_LINK_ERROR)'

2023-12-08 02:42:59:047	[VIDEO]	camera try to connect

[184430105145A20F00] [169.254.1.11] [1701999779.049] [host] [warning] Using a custom watchdog value of 4500ms

[184430105145A20F00] [169.254.1.11] [6748.546] [system] [critical] Fatal error. Please report to developers. Log: 'pool' '180'

2023-12-08 04:35:40:795	[VIDEO]	Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'DEPTH' (X_LINK_ERROR)'

2023-12-08 04:35:41:296	[VIDEO]	camera try to connect

[184430105145A20F00] [169.254.1.11] [1702006541.298] [host] [warning] Using a custom watchdog value of 4500ms

[184430105145A20F00] [169.254.1.11] [1702006561.337] [host] [warning] Monitor thread (device: 184430105145A20F00 [169.254.1.11]) - ping was missed, closing the device connection

2023-12-08 04:36:05:542	[VIDEO]	Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'RGB' (X_LINK_ERROR)'

2023-12-08 04:36:06:043	[VIDEO]	camera try to connect

[184430105145A20F00] [169.254.1.11] [1702006566.045] [host] [warning] Using a custom watchdog value of 4500ms

[184430105145A20F00] [169.254.1.11] [1702006739.136] [host] [warning] Monitor thread (device: 184430105145A20F00 [169.254.1.11]) - ping was missed, closing the device connection

2023-12-08 04:39:03:388	[VIDEO]	Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'RGB' (X_LINK_ERROR)'

2023-12-08 04:39:03:888	[VIDEO]	camera try to connect

[184430105145A20F00] [169.254.1.11] [1702006743.890] [host] [warning] Using a custom watchdog value of 4500ms
```

asahtik, 2023-12-12:
Hi @LuoQuestionmark, sorry for the issue. Please provide an MRE so that I can reproduce and fix the bug.

LuoQuestionmark, 2023-12-12:
@asahtik 
One such example is currently being written; since I need to remove a lot from my existing code while making sure that the bug can be reproduced, it can take a several day before it is finished.

LuoQuestionmark, 2023-12-19:
@asahtik 

Sorry for being late, I have been trying to reproduce the bug with a simpler program instead of the project I am working on, with multiple header files, classes, etc.. To this point I have written a program even more complex than the original program I had making the bug; and the program still cannot reproduce the bug. Therefore I decide to stop for now, if I happen to make it happen again I will return with updates.

closed 2023-12-19 by LuoQuestionmark