luxonis/depthai-core #960: [BUG] Discussion on frame drifting issue

LuoQuestionmark, 2024-01-26:
## Check if issue already exists

Yes. This is previously known as:
- [[BUG] Color camera out of synch](https://github.com/luxonis/depthai/issues/390)
- [Sync NN meta with depth(x,y,z), sync previewout & metaout](https://github.com/luxonis/depthai/pull/157)

## Describe the bug

The RGB and MONO frames are not *perfectly* synchronized, as the previous threads pointed out.

> As RGB and Mono cameras are not synchronized, another problem we saw was a slow drift, 1 frame every 80 seconds at 30fps.

I am aware of the existence of solutions mentioned in these threads, as well as the article on [Message syncing](https://docs.luxonis.com/projects/api/en/latest/tutorials/message_syncing/#message-syncing). However, there are things that do not make senses to me. At the same time, I have been trying to fix the issue of lost of synchronization for months, as the camera delays can up to one second after 24 hours of execution - a little drifted from the *perfect* state. Therefore I open this thread looking for some explications.

### The frame "drifted"

Let us start from the so-called "drifted frame". As far as I can understand, frames between two types of camera are not hardware level synchronized. If we assume that by setting both cameras at 24 fps, one will be running at 24.001 fps, another will be at 24.002 fps. Without any correction, I can see that after a certain amount of time, the two frames will not arrive at the save time. At 24 fps, one frame comes after another at roughly 0.04s. Therefore the max possible delay between frames from different type can up to 0.02s, at that is how I interpret the word "drift".

At the same time, we can see from the test of [[BUG] Color camera out of synch](https://github.com/luxonis/depthai/issues/390), the two cameras can reach a difference of 0.12s. So how can the "drifted frame" explicate the phenomenon?

### Frame synchronization with sequence number

As the article on message syncing mentioned,

>If sensors are set to the same FPS (default is 30), the above two approaches are already integrated into depthai and enabled by default, which allows us to achieve sub-ms delay between all frames + messages generated by these frames!

The example I am going to give in the next section have all sensors running at 24 fps. Whereas after a few moment, we still get a message like this:

```log
rgb seq: 23575, depth seq: 23574,
rgb ts: 985484420711, depth ts: 985442805972
rgb seq: 23576, depth seq: 23575,
rgb ts: 985526089330, depth ts: 985484472372
```

(`ts` as timestamp from `getTimestampDevice().time_since_epoch().count()`)

So the frames are nicely synchronized with the sequence number, whereas message sharing the same sequence number are not necessarily received at the same time. What's more, once such difference between cameras is found, it does not disappear. At the same time, I am setting the pipeline as the follows:

```c++
rgb_xout1->input.setQueueSize(1);
rgb_xout1->input.setBlocking(false);

mono_xout1->input.setQueueSize(1);
mono_xout1->input.setBlocking(false);

depth_xout1->input.setQueueSize(1);
depth_xout1->input.setBlocking(false);
```

I am expected always to get the same, latest frames. So I am wondering if there is a "hidden buffer" somewhere within the program that save the outdated frame? Otherwise I would not get the frame from 0.04s ago instead of the newer frame (and with the same sequence number).

## Minimal Reproducible Example

```c++
#include <condition_variable>
#include <iostream>
#include <mutex>
#include <opencv2/ximgproc/disparity_filter.hpp>
#include <thread>

#include "depthai/depthai.hpp"

namespace {
std::mutex mutex_cam1;

std::condition_variable condition_var1;
std::condition_variable condition_var2;

cv::Mat frame1;
[... 131 lines cut ...]
rgb seq: 48971, depth seq: 48970,
rgb ts: 2043650985892, depth ts: 2043609367041
rgb seq: 48972, depth seq: 48971,
rgb ts: 2043692654212, depth ts: 2043651033172
rgb seq: 48973, depth seq: 48972,
rgb ts: 2043734315012, depth ts: 2043692700431
rgb seq: 48974, depth seq: 48973,
rgb ts: 2043775983680, depth ts: 2043734366605
rgb seq: 48975, depth seq: 48974,
rgb ts: 2043817644330, depth ts: 2043776033085
rgb seq: 48976, depth seq: 48975,
rgb ts: 2043859320068, depth ts: 2043817698975
rgb seq: 48977, depth seq: 48976,
rgb ts: 2043900980910, depth ts: 2043859366517
rgb seq: 48978, depth seq: 48977,
rgb ts: 2043942656614, depth ts: 2043901032705
rgb seq: 48979, depth seq: 48978,
rgb ts: 2043984317321, depth ts: 2043942699508
rgb seq: 48980, depth seq: 48979,
rgb ts: 2044025993308, depth ts: 2043984366138
rgb seq: 48981, depth seq: 48980,
rgb ts: 2044067654054, depth ts: 2044026034432
rgb seq: 48982, depth seq: 48981,
rgb ts: 2044109329892, depth ts: 2044067699957
```

moratom, 2024-02-08:
@LuoQuestionmark as the sensors are not HW synced, the better approach is to use the newly added Sync node which syncs them based on timestamps.

See https://github.com/luxonis/depthai-python/blob/main/examples/Sync/depth_video_synced.py for an example.

diablodale, 2024-03-20:
> So I am wondering if there is a "hidden buffer" somewhere within the program that save the outdated frame?

https://github.com/luxonis/depthai-core/issues/366 and its PRs allow the host-side queue to be zero. Then nothing is stored or buffered...incoming data only goes to callbacks.

LuoQuestionmark, 2024-03-21:
@moratom 
@diablodale 

Thanks for your comments, I actually forgot that the issue was still open. So here is my update on the subject.

The answer to the question "if there is a 'hidden buffer'" is *no*. That said, there are actually two co-existing buffers, which I consider as sort of a bad design of api.

The following code

```c++
mono_xout1->input.setQueueSize(1);
mono_xout1->input.setBlocking(false);
```

was working as expected, to make the `xout` work as a non-blocking queue.

Meanwhile, there is a second buffer configured while using the following code:

```c++
auto depth_queue = device.getOutputQueue("DEPTH");
```
.

The source code (version 2.24)

```c++
///////////////// Device.hpp
/**
 * Gets an output queue corresponding to stream name. If it doesn't exist it throws
 *
 * @param name Queue/stream name, created by XLinkOut node
 * @returns Smart pointer to DataOutputQueue
 */
std::shared_ptr<DataOutputQueue> getOutputQueue(const std::string& name);

/**
 * Gets a queue corresponding to stream name, if it exists, otherwise it throws. Also sets queue options
 *
 * @param name Queue/stream name, set in XLinkOut node
 * @param maxSize Maximum number of messages in queue
 * @param blocking Queue behavior once full. True specifies blocking and false overwriting of oldest messages. Default: true
 * @returns Smart pointer to DataOutputQueue
 */
std::shared_ptr<DataOutputQueue> getOutputQueue(const std::string& name, unsigned int maxSize, bool blocking = true);

//////////////// Device.cpp

std::shared_ptr<DataOutputQueue> Device::getOutputQueue(const std::string& name) {
    // Throw if queue not created
    // all queues for xlink streams are created upfront
    if(outputQueueMap.count(name) == 0) {
        throw std::runtime_error(fmt::format("Queue for stream name '{}' doesn't exist", name));
    }
    // Return pointer to this DataQueue
    return outputQueueMap.at(name);
}

std::shared_ptr<DataOutputQueue> Device::getOutputQueue(const std::string& name, unsigned int maxSize, bool blocking) {
    // Throw if queue not created
    // all queues for xlink streams are created upfront
    if(outputQueueMap.count(name) == 0) {
        throw std::runtime_error(fmt::format("Queue for stream name '{}' doesn't exist", name));
    }

    // Modify max size and blocking
    outputQueueMap.at(name)->setMaxSize(maxSize);
    outputQueueMap.at(name)->setBlocking(blocking);

    // Return pointer to this DataQueue
    return outputQueueMap.at(name);
}
```
Now, the thing is, if this function is called without specifying `maxSize` and `blocking`, it will use the default setting from `DataQueue.hpp`, which is the following:

```c++
DataOutputQueue(const std::shared_ptr<XLinkConnection> conn, const std::string& streamName, unsigned int maxSize = 16, bool blocking = true);
```

which means a second buffer is created as a `blocking=True` and `maxSize=16`.

So at the end of day we got a `blocking=false` `maxSize=1` buffer, and a `blocking=true` `maxSize=16` buffer. The combination is, alas, `blocking=true` `maxSize=16`. Along side with the issue mentioned as [frame drifting](https://github.com/luxonis/depthai/issues/390), the frame drifting is slowly but steady.

So after specifying the configuration of the second buffer, the problem has been solved. We have had our program running for 17 days without any observable delay after the changes.

```c++
aDevice.getOutputQueue("RGB", 1, false); // after the first config for xlink
```

closed 2024-03-21 by LuoQuestionmark