luxonis/depthai-core #969: [Feature-Request] Update  “sdcard_no_eth” branch to current version

KarolPiotrRoesner, 2024-02-23:
We are attempting to use an FFC 4P board in standalone mode, attaching it to a drone with four cameras. Our goal is to directly save images to a micro SD card without relying on additional hardware like a Raspberry Pi Zero.

We’re aware of the existence of the “sdcard_no_eth” branch, which has been functional for us. We’ve successfully used a script to write a “Hello World” file to the SD card. However, we’re currently facing an issue with the OV9281 global shutter camera. These cameras only support 800P resolution, but unfortunately, the “sdcard_no_eth” branch does not accommodate this resolution.

Is it possible to update the “sdcard_no_eth” branch to the current DepthAI core version (2.24)?”

moratom, 2024-03-13:
@zrezke would you mind taking a look if this is doable?

zrezke, 2024-03-13:
Yes, will take a look @moratom

zrezke, 2024-03-13:
@KarolPiotrRoesner are you trying to create a `ColorCamera` node for the OV9281? The resolution `THE_800_P` is available for the `MonoCamera` node on the `sdcard_no_eth`, tho it is missing for the `ColorCamera`..

KarolPiotrRoesner, 2024-03-14:
Hey @zrezke, 
its correct, we initially used the ColorCamera node until our issue here. Today, we tested the MonoCamera node on the "sdcard_no_eth" branch. While it did display a camera image, we encountered several issues that are not present in the latest core version.

When attempting to retrieve four camera images, our code terminates with the following error:

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  MonoCamera(2) - CameraBoardSocket AUTO: Neither LEFT nor RIGHT camera sockets are available
```

Additionally, it appears that the sdcard_no_eth branch does not support Hardware Sync. This presents an issue for us since Hardware Sync is essential for our intended image stitching process in later stages.
The function we mean is:
`CameraControl& dai::CameraControl::setFrameSyncMode(FrameSyncMode  mode)`

Furthermore, it would greatly benefit our project if we could retrieve the images as ColorImages. This is crucial for us because the data will be visualized to assist firefighters during their missions. The grayscale images also show some kind of artifacts when we zoom in, a picture is included.
[image]

We would appreciate any further help.

zrezke, 2024-03-14:
Hi @KarolPiotrRoesner  
Alright, thanks for expanding on your initial post! I'll look into getting the SD support running on the latest depthai (2.25.0.0)

zrezke, 2024-03-20:
Hey @KarolPiotrRoesner 
There is now an updated branch: [sdcard_no_eth_updated](https://github.com/luxonis/depthai-core/tree/sdcard_no_eth_updated). It is based on latest develop, pretty much v2.25.0.0. Let me know if it works for you, thanks.

moratom, 2024-03-21:
Thansk @zrezke!

@KarolPiotrRoesner do you mind sharing what goes wrong if you don't explicitly disable the PCL during the libraries build process?

KarolPiotrRoesner, 2024-03-22:
I should stop switching between my account :D 

@moratom @zrezke 
The output was:
```
-- Configuring done
CMake Error in CMakeLists.txt:
  Target "depthai-pcl" INTERFACE_INCLUDE_DIRECTORIES property contains path:

    "/tmp/depthai-core/build/hunter/_Base/cb0ea1f/21498a3/69d66b5/Install/include"

  which is prefixed in the build directory.

CMake Error in CMakeLists.txt:
  Target "depthai-pcl" INTERFACE_INCLUDE_DIRECTORIES property contains path:

    "/tmp/depthai-core/build/hunter/_Base/cb0ea1f/21498a3/69d66b5/Install/include"

  which is prefixed in the build directory.Target "depthai-pcl"
  INTERFACE_INCLUDE_DIRECTORIES property contains path:

    "/tmp/depthai-core/build/hunter/_Base/cb0ea1f/21498a3/69d66b5/Install/include"

  which is prefixed in the source directory.

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
```