luxonis/depthai-core #978: outdated Hunter causes 3000+ lines of errors, making builds difficult

diablodale, 2024-03-13:
depthai-core uses an outdated Hunter. Cmake now issues warnings of deprecation which causes 3000+ lines of warnings for a build of depthai-core

https://github.com/luxonis/depthai-core/blob/eba7400699e69c81f04dc0dc1e2c2b6aafc3b962/CMakeLists.txt#L38

for example
```
...
cmake] CMake Deprecation Warning at C:/Users/dale/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_make_directory.cmake:4 (cmake_minimum_required):
[cmake]   Compatibility with CMake < 3.5 will be removed from a future version of
[cmake]   CMake.
[cmake] 
[cmake]   Update the VERSION argument <min> value or use a ...<max> suffix to tell
[cmake]   CMake that the project does not need compatibility with older versions.
[cmake] Call Stack (most recent call first):
[cmake]   C:/Users/dale/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_save_to_cache.cmake:8 (include)
[cmake]   C:/Users/dale/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_download.cmake:26 (include)
[cmake]   C:/Users/dale/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_add_package.cmake:49 (include)
[cmake]   cmake/depthaiDependencies.cmake:5 (hunter_add_package)
[cmake]   cmake/depthaiDependencies.cmake:32 (maybe_hunter_add_package)
[cmake]   CMakeLists.txt:171 (include)
[cmake] 
[cmake] 
[cmake] CMake Deprecation Warning at C:/Users/dale/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_lock_directory.cmake:4 (cmake_minimum_required):
[cmake]   Compatibility with CMake < 3.5 will be removed from a future version of
[cmake]   CMake.
[cmake] 
[cmake]   Update the VERSION argument <min> value or use a ...<max> suffix to tell
[cmake]   CMake that the project does not need compatibility with older versions.
[cmake] Call Stack (most recent call first):
[cmake]   C:/Users/dale/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_make_directory.cmake:7 (include)
[cmake]   C:/Users/dale/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_save_to_cache.cmake:8 (include)
[cmake]   C:/Users/dale/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_download.cmake:26 (include)
[cmake]   C:/Users/dale/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_add_package.cmake:49 (include)
[cmake]   cmake/depthaiDependencies.cmake:5 (hunter_add_package)
[cmake]   cmake/depthaiDependencies.cmake:32 (maybe_hunter_add_package)
[cmake]   CMakeLists.txt:171 (include)
[cmake]
...
```

moratom, 2024-03-13:
Thanks for the report @diablodale - we'll address the warnings before the next release.

As far as I understand, the warnings are not only related to the hunter version, but also to the dependancies that hunter downloads which have the minimum `cmake` version set to less than `3.5` so we'll need to bump those as well to avoid the warning.

diablodale, 2024-03-13:
You will probably also see warnings for cmake policy CMP0144 due to hunter dependencies with miXeD cAsE like `BZip2`.
Adding `cmake_policy(SET CMP0144 NEW)` or correcting the mixed case is needed. I also think the `cmake_policy(SET` already in cmakelist.txt should be moved to the top after the cmake minimum so that hunter and other toolchains are aware of the policy changes. You'll want to test this yourself because my depthai-core fork diverges.