luxonis/depthai-core #981: [BUG] cannot compile: SSL Connect error in Hunter

artificiel, 2024-03-14:
The compilation hangs on downloads within the hunter gathering (specifically, at nlohmann_json).

One macOS14.4, Apple Silicon:

```
git clone git@github.com:luxonis/depthai-core.git
cd depthai-core
git submodule update --init --recursive
cmake -S. -Bbuild -DHUNTER_STATUS_DEBUG=1
cmake --build build
```
after a fair amount of hunter/cmake warnings:
```bash
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] nlohmann_json versions available: [2.1.1-p0;2.1.1-p1;3.0.0;3.0.1;3.1.0;3.1.1;3.1.2;3.2.0;3.3.0;3.4.0;3.5.0;3.6.0;3.6.1;3.7.0;3.7.1;3.7.2;3.7.3;3.8.0]
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Package 'nlohmann_json' CONFIGURATION_TYPES: Release
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Package 'nlohmann_json' is cacheable: YES
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Locking directory: /Users/user/.hunter/_Base/Download/nlohmann_json/3.9.1/f8a20a7
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Lock done
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Install to: /Users/user/.hunter/_Base/cb0ea1f/3155fcd/7f91c96/Install
-- [hunter] NLOHMANN_JSON_ROOT: /Users/user/.hunter/_Base/cb0ea1f/3155fcd/7f91c96/Install (ver.: 3.9.1)
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Locking directory: /Users/user/.hunter/_Base/cb0ea1f/3155fcd/7f91c96
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Lock done
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Already locked: /Users/user/.hunter/_Base/cb0ea1f/3155fcd/7f91c96
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Locking directory: /Users/user/.hunter/_Base/Download/nlohmann_json/3.9.1/f8a20a7
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Lock done
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Package 'nlohmann_json' default arguments: 'JSON_BuildTests=OFF'
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Add extra CMake args: 'JSON_BuildTests' = 'OFF'
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Locking directory: /Users/user/.hunter/_Base/Cache/meta/3155fcd
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Lock done
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Locking directory: /Users/user/.hunter/_Base/Cache/meta/3155fcd/nlohmann_json/3.9.1/f8a20a7
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Lock done
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Locking directory: /Users/user/.hunter/_Base/Cache/meta/3155fcd/nlohmann_json/3.9.1/f8a20a7/151cbba
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Lock done
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Locking directory: /Users/user/.hunter/_Base/Cache/meta/3155fcd/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Lock done
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Locking directory: /Users/user/.hunter/_Base/Cache/meta/3155fcd/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Lock done
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Locking directory: /Users/user/.hunter/_Base/Cache
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Lock done
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Using environment variable HOME
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Downloading DONE metafile (try #0 of 10):
-- [hunter *** DEBUG *** 2024-03-14T17:10:46]   https://raw.githubusercontent.com/cpp-pm/hunter-cache/master/3155fcd/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-03-14T17:10:46]   -> /Users/user/.hunter/_Base/Cache/meta/3155fcd/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] File not found
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Downloading DONE metafile (try #1 of 10):
-- [hunter *** DEBUG *** 2024-03-14T17:10:46]   https://raw.githubusercontent.com/cpp-pm/hunter-cache/master/3155fcd/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-03-14T17:10:46]   -> /Users/user/.hunter/_Base/Cache/meta/3155fcd/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Download error ("SSL connect error")
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Downloading DONE metafile (try #2 of 10):
-- [hunter *** DEBUG *** 2024-03-14T17:10:46]   https://raw.githubusercontent.com/cpp-pm/hunter-cache/master/3155fcd/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-03-14T17:10:46]   -> /Users/user/.hunter/_Base/Cache/meta/3155fcd/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-03-14T17:10:46] Retry after 5 seconds (attempt #2) ...
-- [hunter *** DEBUG *** 2024-03-14T17:10:51] Download error ("SSL connect error")
-- [hunter *** DEBUG *** 2024-03-14T17:10:51] Downloading DONE metafile (try #3 of 10):
-- [hunter *** DEBUG *** 2024-03-14T17:10:51]   https://raw.githubusercontent.com/cpp-pm/hunter-cache/master/3155fcd/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-03-14T17:10:51]   -> /Users/user/.hunter/_Base/Cache/meta/3155fcd/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-03-14T17:10:51] Retry after 5 seconds (attempt #3) ...
-- [hunter *** DEBUG *** 2024-03-14T17:10:56] Download error ("SSL connect error")
-- [hunter *** DEBUG *** 2024-03-14T17:10:56] Downloading DONE metafile (try #4 of 10):
-- [hunter *** DEBUG *** 2024-03-14T17:10:56]   https://raw.githubusercontent.com/cpp-pm/hunter-cache/master/3155fcd/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-03-14T17:10:56]   -> /Users/user/.hunter/_Base/Cache/meta/3155fcd/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-03-14T17:10:56] Retry after 15 seconds (attempt #4) ...
-- [hunter *** DEBUG *** 2024-03-14T17:11:11] Download error ("SSL connect error")
-- [hunter *** DEBUG *** 2024-03-14T17:11:11] Downloading DONE metafile (try #5 of 10):
-- [hunter *** DEBUG *** 2024-03-14T17:11:11]   https://raw.githubusercontent.com/cpp-pm/hunter-cache/master/3155fcd/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-03-14T17:11:11]   -> /Users/user/.hunter/_Base/Cache/meta/3155fcd/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-03-14T17:11:11] Retry after 60 seconds (attempt #5) ...
-- [hunter *** DEBUG *** 2024-03-14T17:12:11] Download error ("SSL connect error")
-- [hunter *** DEBUG *** 2024-03-14T17:12:11] Downloading DONE metafile (try #6 of 10):
-- [hunter *** DEBUG *** 2024-03-14T17:12:11]   https://raw.githubusercontent.com/cpp-pm/hunter-cache/master/3155fcd/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-03-14T17:12:11]   -> /Users/user/.hunter/_Base/Cache/meta/3155fcd/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-03-14T17:12:11] Retry after 90 seconds (attempt #6) ...
```

moratom, 2024-03-14:
Sorry for the issue @artificiel,
there's a bug with the `cmake` that comes with `brew`.

See https://github.com/luxonis/depthai-core/issues/953#issuecomment-1905696609 for a workaround:
`cmake -S. -Bbuild -DHUNTER_TLS_VERIFY=OFF`

artificiel, 2024-03-15:
OK thanks!

closed 2024-03-15 by artificiel