luxonis/depthai-core #982: [BUG] problem generating CMake files when PCL is present (macOS14.4)

artificiel, 2024-03-15:
When using PCL (on macOS14.4, installed via `brew install pcl`) there is a problem with some names related to jsoncpp (did not find reference to this in the existing issues):

```
CMake Error at /opt/homebrew/lib/cmake/vtk-9.3/VTK-targets.cmake:1013 (set_target_properties):
  The link interface of target "VTK::jsoncpp" contains:

    JsonCpp::JsonCpp

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /opt/homebrew/lib/cmake/vtk-9.3/vtk-config.cmake:145 (include)
  /opt/homebrew/share/pcl-1.14/PCLConfig.cmake:275 (find_package)
  /opt/homebrew/share/pcl-1.14/PCLConfig.cmake:324 (find_VTK)
  /opt/homebrew/share/pcl-1.14/PCLConfig.cmake:569 (find_external_library)
  cmake/depthaiDependencies.cmake:85 (find_package)
  CMakeLists.txt:172 (include)
```
(without PCL everything compiles fine (dynamic libs and examples))

moratom, 2024-03-15:
@artificiel thanks for the report, we'll look into it.

moratom, 2024-03-21:
https://github.com/luxonis/depthai-core/pull/980 should address the issue - do you mind trying it out @artificiel ?

dtorre38, 2024-03-28:
This seems to be the only place that this issue is being discussed. I tried #980 but it didn't work. Adding the following did work:

```
# Check if JsonCpp target does not exist and then create it manually
if(NOT TARGET JsonCpp::JsonCpp)
  # Attempt to find JsonCpp library - modify paths as needed
  find_library(JSONCPP_LIBRARY NAMES jsoncpp PATHS /opt/homebrew/Cellar/jsoncpp/1.9.5/lib /usr/local/lib)
  find_path(JSONCPP_INCLUDE_DIR NAMES json/json.h PATHS /opt/homebrew/Cellar/jsoncpp/1.9.5/include /usr/local/include)

  # Create an imported target
  add_library(JsonCpp::JsonCpp SHARED IMPORTED)
  set_target_properties(JsonCpp::JsonCpp PROPERTIES
    IMPORTED_LOCATION "${JSONCPP_LIBRARY}"
    INTERFACE_INCLUDE_DIRECTORIES "${JSONCPP_INCLUDE_DIR}")
endif()
```