luxonis/depthai-core #988: `DeviceBootloader::getVersion()` returns wrong version when `DeviceBootloader(allowFlashingBootloader=true)`

diablodale, 2024-03-20:
When `allowFlashingBootloader=true`  --> getVersion() returns the embedded blob version (WRONG)
When `allowFlashingBootloader=false`  --> getVersion() returns the version on the device (CORRECT)

depthai-core v2.25.0 (reminder that this embeds bootloader v0.0.26)
OAK-D-PRO-POE with bootloader firmware v0.0.17

```
auto deviceBootloader = std::make_unique<dai::DeviceBootloader>(targetDeviceInfo, false);
const auto test_good = deviceBootloader->getVersion();
assert(test_good == v0.0.17)

auto deviceBootloader = std::make_unique<dai::DeviceBootloader>(targetDeviceInfo, true);
const auto test_bad = deviceBootloader->getVersion();
assert(test_bad == v0.0.26)  //  💣 this assert fails
```

diablodale, 2024-03-20:
Don't use `DeviceBootloader::getVersion()`. That's what I have learned.

After collect bit-and-pieces of information and some reverse engineering, this is an *incomplete* function I've written to get the actual version of a bootloader saved in the device's flash.

```c++
// DeviceBootloader::getVersion() is not reliable, use this instead
// BUGBUG https://github.com/luxonis/depthai-core/issues/988 errant versions when flash=true
std::optional<dai::DeviceBootloader::Version> getBootloaderVersionStoredOnDevice(const dai::DeviceBootloader& deviceBootloader) {
    // allowFlashingBootloader=true always dynamically boots the embedded blob bootloader which overwrites the information of the actual stored bootloader
    if (deviceBootloader.isAllowedFlashingBootloader()) {
        if ([[maybe_unused]] const auto memInfo = deviceBootloader.getMemoryInfo(dai::DeviceBootloader::Memory::FLASH); memInfo.available) {
            [[maybe_unused]] const auto appInfo = deviceBootloader.readApplicationInfo(dai::DeviceBootloader::Memory::FLASH);
            DPOST(std::format("Memory FLASH, size {}, info {}, app flashed {}, firmware version {}, app name {}\n", memInfo.size, memInfo.info, appInfo.hasApplication, appInfo.firmwareVersion, appInfo.applicationName));
        }
        else {
            DPOST("Memory FLASH not available\n");
        }
        assert(deviceBootloader.isEmbeddedVersion());
        throw std::runtime_error("internal error: impossible to get device stored bootloader version while also allowing flashing");
    }

    // when allowFlashingBootloader=false and isEmbeddedVersion()=true, the device has no stored bootloader (e.g. OAK-D) so it was dynamically booted from the embedded blob
    if (deviceBootloader.isEmbeddedVersion())
        return std::nullopt;

    // when allowFlashingBootloader=false and isEmbeddedVersion()=false, the device has a stored bootloader and it was booted from it
    // it could be the factory bootloader or the user bootloader: isUserBootloader()=true
    return deviceBootloader.getVersion();
}
```

Unfortunately, I can find no technique to get the bootloader version on the device when `DeviceBootloader` was constructed with `allowFlashingBootloader=true`. I even tried `getMemoryInfo()` and `readApplicationInfo()`. Nothing worked.

Is there any API or RPC that can get the bootloader version on a device ***always***? Or can do it when `allowFlashingBootloader=true`?

moratom, 2024-04-02:
Thanks for the issue report @diablodale, we have added this to our internal task list and will update the issue when we have the solution.