luxonis/depthai-core #993: `getConnectedCameraFeatures()` returns invalid `CameraSensorConfig`

diablodale, 2024-03-25:
With OAK-D-LITE (one of them without an imu), a call to `getConnectedCameraFeatures()` returns invalid `CameraSensorConfig` entries.
There are two config entries for both mono cameras on this device that are impossible. Below is the `struct CameraFeatures` for one of these mono cameras. We can see this is a camera named "left" (which is a non-color camera since it is always named "color"). We can see its `supportedTypes` contains only `dai::CameraSensorType::MONO`. Yet, when we look at `dai::CameraFeatures::configs` we can see 4 entries...and 2 of them are impossible. These two configs have `dai::CameraSensorConfig::type = dai::CameraSensorType::COLOR`. That is impossible. We have already proven this "left" `dai::CameraSensorType::MONO` camera. It is not possible for a MONO camera to have COLOR configs.

I do see the depthai-shared comment
```c++
/* For some sensors it's not possible to determine if they are color or mono
* (e.g. OV9782 and OV9282), so this could return more than one entry
*/
std::vector<CameraSensorType> supportedTypes;
```

However `supportedTypes` contains only one (1) entry of MONO and therefore that comment is moot.

```json
{
    "additionalNames": [],
    "calibrationResolution": {
        "fov": {
            "height": 815.0,
            "width": 1295.0,
            "x": 0.0,
            "y": 0.0
        },
        "height": 800,
        "maxFps": 129.60000610351563,
        "minFps": 1.687000036239624,
        "type": 1,
        "width": 1280
    },
    "configs": [
        {
            "fov": {
                "height": 720.0,
                "width": 1280.0,
                "x": 0.0,
                "y": 40.0
            },
            "height": 720,
            "maxFps": 143.10000610351563,
            "minFps": 1.687000036239624,
            "type": 1,
            "width": 1280
        },
        {
            "fov": {
                "height": 720.0,
                "width": 1280.0,
                "x": 0.0,
                "y": 40.0
            },
            "height": 720,
            "maxFps": 143.10000610351563,
            "minFps": 1.687000036239624,
            "type": 0,
            "width": 1280
        },
        {
            "fov": {
                "height": 800.0,
                "width": 1280.0,
                "x": 0.0,
                "y": 0.0
            },
            "height": 800,
            "maxFps": 129.60000610351563,
            "minFps": 1.687000036239624,
            "type": 1,
            "width": 1280
        },
        {
            "fov": {
                "height": 800.0,
                "width": 1280.0,
                "x": 0.0,
                "y": 0.0
            },
            "height": 800,
            "maxFps": 129.60000610351563,
            "minFps": 1.687000036239624,
            "type": 0,
            "width": 1280
        },
        {
            "fov": {
                "height": 800.0,
                "width": 1280.0,
                "x": 0.0,
                "y": 0.0
            },
            "height": 400,
            "maxFps": 255.6999969482422,
            "minFps": 1.687000036239624,
            "type": 1,
            "width": 640
        },
        {
            "fov": {
                "height": 800.0,
                "width": 1280.0,
                "x": 0.0,
                "y": 0.0
            },
            "height": 400,
            "maxFps": 255.6999969482422,
            "minFps": 1.687000036239624,
            "type": 0,
            "width": 640
        }
    ],
    "hasAutofocus": false,
    "hasAutofocusIC": false,
    "height": 800,
    "name": "left",
    "orientation": -1,
    "sensorName": "OV9282",
    "socket": 1,
    "supportedTypes": [
        1
    ],
    "width": 1280
},
```

moratom, 2024-04-02:
@zrezke would you mind taking a look if the recent changes to properly recognize OV9782 on the SR and wide models could be the reason for this?

zrezke, 2024-04-04:
Not introduced by those changes. But this errant config reporting is fixable for non FFC devices. I will fix it in FW,