luxonis/depthai-core #994: [BUG] Unusable ToF depth quality with OAK-D SR PoE

dmn-seasony, 2024-03-26:
**Describe the bug**
I finally managed to get some images out of this camera (I had to modify and rebuild depthai-viewer because it has not been updated with the latest version of depthai-core). But the depth image is super noisy and most of the depth measurements are completely off (as in several meters off). I assume this is just a configuration issue, but I have not found any way to access the camera settings. I tried reducing the FPS in depthai-viewer, but this did not affect the quality.

**Screenshots**
Here you can see a large part of the ceiling is measured as only 6cm away when it is actually around 2m away. The other parts of the depth image are also wrong, for example the red part of the ceiling is measured as 60cm away.
[image]

themarpe, 2024-03-26:
CC: @Erol444

CenekAlbl, 2024-04-02:
Hi @dmn-seasony , apologies for the limited ToF experience at the moment. There are several reason causing your troubles:

dmn-seasony, 2024-04-03:
@CenekAlbl Thanks for the information. Do you have an ETA for that firmware update?

tkj-seasony, 2024-04-15:
For others reading this who have already received their cameras, please ensure to check the sensors in your camera. We were shipped 4 cameras OAK-D SR PoE cameras as Daniel mentioned above which did not seem to have a working calibration and showed other lens and aliasing artifacts.
But more interestingly the cameras were equipped with the monochrome OV9282 sensor rather than the OV9782 color sensor listed in the specs.
To check which sensors your camera has, you can run:
```
python3 -c 'import depthai; device=depthai.Device(); [print(f"Sensor: {sensor.name} ({sensor.sensorName}), socket: {sensor.socket.name}, type: {sensor.supportedTypes[0].name}") for sensor in device.getConnectedCameraFeatures()]'
```
For a correct OAK-D SR PoE camera the printout should look like:
```
Sensor: tof (S5K33D), socket: CAM_A, type: TOF
Sensor: left (OV9782), socket: CAM_B, type: COLOR
Sensor: right (OV9782), socket: CAM_C, type: COLOR
```

dmn-seasony, 2024-04-23:
@CenekAlbl @Erol444 We have now received the new cameras with new front glass and supposedly updated firmware. However it seems like the "phase-wrapping" issue is still there:
[image]

The overall noise level is also rather high and the output doesn't seem to be calibrated. See for example this image, where the camera is perpendicular to a large flat surface at approximately 0.5m distance, but the point cloud is completely skewed:
[image]

DavidZah, 2024-04-23:
Hi unfortunately deptaiViewer still doesn't have support for tof node. It should be added in the next 14 days. In the meantime, you can use the following code for your testing.
`#!/usr/bin/env python3
import os
import shutil
import time

import cv2
import depthai as dai
import numpy as np
from scipy.ndimage import median_filter

print(dai.__version__)

import zlib
import scipy
from matplotlib import pyplot as plt
import matplotlib
from tqdm import tqdm

FPPN_ENABLE = True
WIGGLE_ENABLE = True
TEMPERATURE_ENABLE = False
OPTICAL_ENABLE = True

def create_pipeline():
    pipeline = dai.Pipeline()

    tof = pipeline.create(dai.node.ToF)
    tof.setNumShaves(3)

    # Configure the ToF node
    tofConfig = tof.initialConfig.get()
    # tofConfig.depthParams.freqModUsed = dai.RawToFConfig.DepthParams.TypeFMod.MIN
    #tofConfig.depthParams.freqModUsed = dai.RawToFConfig.DepthParams.TypeFMod.ALL
    tofConfig.enableFPPNCorrection = True
    tofConfig.enableOpticalCorrection = True
    tofConfig.enableWiggleCorrection = True
    tofConfig.enableTemperatureCorrection = False
    tofConfig.phaseUnwrappingLevel = 4
    xinTofConfig = pipeline.create(dai.node.XLinkIn)
    xinTofConfig.setStreamName("tofConfig")
    xinTofConfig.out.link(tof.inputConfig)

    tof.initialConfig.set(tofConfig)

    cam_tof = pipeline.create(dai.node.Camera)
    cam_tof.properties.numFramesPoolRaw = 5
    cam_tof.setFps(60)
    cam_tof.setImageOrientation(dai.CameraImageOrientation.ROTATE_180_DEG)
    cam_tof.setBoardSocket(dai.CameraBoardSocket.CAM_A)
    cam_tof.raw.link(tof.input)

    xout = pipeline.create(dai.node.XLinkOut)
    xout.setStreamName("depth")
    tof.depth.link(xout.input)

    tofConfig = tof.initialConfig.get()

    return pipeline, tofConfig

if __name__ == '__main__':
    pipeline, tofConfig = create_pipeline()
    with dai.Device(pipeline) as device:
        print('Connected cameras:', device.getConnectedCameraFeatures())
        qDepth = device.getOutputQueue(name="depth")

        tofConfigInQueue = device.getInputQueue("tofConfig")

        #tofConfigInQueue = device.getInputQueue("tofConfig")

        counter = 0
        while True:
            start = time.time()
            key = cv2.waitKey(1)
            if key == ord('f'):
                tofConfig.enableFPPNCorrection = not tofConfig.enableFPPNCorrection
                tofConfigInQueue.send(tofConfig)
            elif key == ord('o'):
                tofConfig.enableOpticalCorrection = not tofConfig.enableOpticalCorrection
                tofConfigInQueue.send(tofConfig)
            elif key == ord('w'):
                tofConfig.enableWiggleCorrection = not tofConfig.enableWiggleCorrection
                tofConfigInQueue.send(tofConfig)
            elif key == ord('t'):
                tofConfig.enableTemperatureCorrection = not tofConfig.enableTemperatureCorrection
                tofConfigInQueue.send(tofConfig)
            elif key == ord('q'):
                break
            elif key == ord('0'):
                tofConfig.enablePhaseUnwrapping = False
                tofConfig.phaseUnwrappingLevel = 4
                tofConfigInQueue.send(tofConfig)
            elif key == ord('1'):
                tofConfig.enablePhaseUnwrapping = True
                tofConfig.phaseUnwrappingLevel = 1
                tofConfigInQueue.send(tofConfig)
            elif key == ord('2'):
                tofConfig.enablePhaseUnwrapping = True
                tofConfig.phaseUnwrappingLevel = 2
                tofConfigInQueue.send(tofConfig)
            elif key == ord('3'):
                tofConfig.enablePhaseUnwrapping = True
                tofConfig.phaseUnwrappingLevel = 3
                tofConfigInQueue.send(tofConfig)
            elif key == ord('4'):
                tofConfig.enablePhaseUnwrapping = True
                tofConfig.phaseUnwrappingLevel = 4
                tofConfigInQueue.send(tofConfig)
            elif key == ord('m'):
                medianSettings = [dai.MedianFilter.MEDIAN_OFF, dai.MedianFilter.KERNEL_3x3, dai.MedianFilter.KERNEL_5x5,
                                  dai.MedianFilter.KERNEL_7x7]
                currentMedian = tofConfig.median
                nextMedian = medianSettings[(medianSettings.index(currentMedian) + 1) % len(medianSettings)]
                print(f"Changing median to {nextMedian.name} from {currentMedian.name}")
                tofConfig.median = nextMedian
                tofConfigInQueue.send(tofConfig)
            elif key == ord('r'):
                gt = input("Enter ground truth: ")
                gt = float(gt)
                plot_depth_error_analysis(depth_map, gt)
            elif key == ord('l'):
                treshold = input("Enter treshold: ")
                treshold = int(treshold)
                tofConfigInQueue.send(tofConfig)

            imgFrame = qDepth.get()  # blocking call, will wait until a new data has arrived
            depth_map = imgFrame.getFrame()

            depth_downscaled = depth_map[::4]
            non_zero_depth = depth_downscaled[depth_downscaled != 0]  # Remove invalid depth values
            if len(non_zero_depth) == 0:
                min_depth, max_depth = 0, 0
            else:
                min_depth = 200
                max_depth = 7000
            depth_colorized = np.interp(depth_map, (min_depth, max_depth), (0, 255)).astype(np.uint8)
            depth_colorized = cv2.applyColorMap(depth_colorized, cv2.COLORMAP_JET)

            # Display the modified image
            cv2.imshow("Colorized depth", depth_colorized)
            #cv2.createTrackbar('Filtration', 'Colorized depth', 0, 255, nothing)
            counter += 1

    device.close()
    print('Done')`
    
    Also you need lates depthai-python tof_decoding branch https://github.com/luxonis/depthai-python/tree/tof_decoding

DavidZah, 2024-04-23:
Hi unfortunately deptaiViewer still doesn't have support for tof node. It should be added in the next 14 days. In the meantime, you can use the following code for your testing. Also you need latest depthai-python tof_decoding branch https://github.com/luxonis/depthai-python/tree/tof_decoding  
```
#!/usr/bin/env python3
import os
import shutil
import time

import cv2
import depthai as dai
import numpy as np
from scipy.ndimage import median_filter

print(dai.__version__)

import zlib
import scipy
from matplotlib import pyplot as plt
import matplotlib
from tqdm import tqdm

FPPN_ENABLE = True
WIGGLE_ENABLE = True
TEMPERATURE_ENABLE = False
OPTICAL_ENABLE = True

def create_pipeline():
    pipeline = dai.Pipeline()

    tof = pipeline.create(dai.node.ToF)
    tof.setNumShaves(3)

    # Configure the ToF node
    tofConfig = tof.initialConfig.get()
    # tofConfig.depthParams.freqModUsed = dai.RawToFConfig.DepthParams.TypeFMod.MIN
    #tofConfig.depthParams.freqModUsed = dai.RawToFConfig.DepthParams.TypeFMod.ALL
    tofConfig.enableFPPNCorrection = True
    tofConfig.enableOpticalCorrection = True
    tofConfig.enableWiggleCorrection = True
    tofConfig.enableTemperatureCorrection = False
    tofConfig.phaseUnwrappingLevel = 4
    xinTofConfig = pipeline.create(dai.node.XLinkIn)
    xinTofConfig.setStreamName("tofConfig")
    xinTofConfig.out.link(tof.inputConfig)

    tof.initialConfig.set(tofConfig)

    cam_tof = pipeline.create(dai.node.Camera)
    cam_tof.properties.numFramesPoolRaw = 5
    cam_tof.setFps(60)
    cam_tof.setImageOrientation(dai.CameraImageOrientation.ROTATE_180_DEG)
    cam_tof.setBoardSocket(dai.CameraBoardSocket.CAM_A)
    cam_tof.raw.link(tof.input)

    xout = pipeline.create(dai.node.XLinkOut)
    xout.setStreamName("depth")
    tof.depth.link(xout.input)

    tofConfig = tof.initialConfig.get()

    return pipeline, tofConfig

if __name__ == '__main__':
    pipeline, tofConfig = create_pipeline()
    with dai.Device(pipeline) as device:
        print('Connected cameras:', device.getConnectedCameraFeatures())
        qDepth = device.getOutputQueue(name="depth")

        tofConfigInQueue = device.getInputQueue("tofConfig")

        #tofConfigInQueue = device.getInputQueue("tofConfig")

        counter = 0
        while True:
            start = time.time()
            key = cv2.waitKey(1)
            if key == ord('f'):
                tofConfig.enableFPPNCorrection = not tofConfig.enableFPPNCorrection
                tofConfigInQueue.send(tofConfig)
            elif key == ord('o'):
                tofConfig.enableOpticalCorrection = not tofConfig.enableOpticalCorrection
                tofConfigInQueue.send(tofConfig)
            elif key == ord('w'):
                tofConfig.enableWiggleCorrection = not tofConfig.enableWiggleCorrection
                tofConfigInQueue.send(tofConfig)
            elif key == ord('t'):
                tofConfig.enableTemperatureCorrection = not tofConfig.enableTemperatureCorrection
                tofConfigInQueue.send(tofConfig)
            elif key == ord('q'):
                break
            elif key == ord('0'):
                tofConfig.enablePhaseUnwrapping = False
                tofConfig.phaseUnwrappingLevel = 4
                tofConfigInQueue.send(tofConfig)
            elif key == ord('1'):
                tofConfig.enablePhaseUnwrapping = True
                tofConfig.phaseUnwrappingLevel = 1
                tofConfigInQueue.send(tofConfig)
            elif key == ord('2'):
                tofConfig.enablePhaseUnwrapping = True
                tofConfig.phaseUnwrappingLevel = 2
                tofConfigInQueue.send(tofConfig)
            elif key == ord('3'):
                tofConfig.enablePhaseUnwrapping = True
                tofConfig.phaseUnwrappingLevel = 3
                tofConfigInQueue.send(tofConfig)
            elif key == ord('4'):
                tofConfig.enablePhaseUnwrapping = True
                tofConfig.phaseUnwrappingLevel = 4
                tofConfigInQueue.send(tofConfig)
            elif key == ord('m'):
                medianSettings = [dai.MedianFilter.MEDIAN_OFF, dai.MedianFilter.KERNEL_3x3, dai.MedianFilter.KERNEL_5x5,
                                  dai.MedianFilter.KERNEL_7x7]
                currentMedian = tofConfig.median
                nextMedian = medianSettings[(medianSettings.index(currentMedian) + 1) % len(medianSettings)]
                print(f"Changing median to {nextMedian.name} from {currentMedian.name}")
                tofConfig.median = nextMedian
                tofConfigInQueue.send(tofConfig)
            elif key == ord('r'):
                gt = input("Enter ground truth: ")
                gt = float(gt)
                plot_depth_error_analysis(depth_map, gt)
            elif key == ord('l'):
                treshold = input("Enter treshold: ")
                treshold = int(treshold)
                tofConfigInQueue.send(tofConfig)

            imgFrame = qDepth.get()  # blocking call, will wait until a new data has arrived
            depth_map = imgFrame.getFrame()

            depth_downscaled = depth_map[::4]
            non_zero_depth = depth_downscaled[depth_downscaled != 0]  # Remove invalid depth values
            if len(non_zero_depth) == 0:
                min_depth, max_depth = 0, 0
            else:
                min_depth = 200
                max_depth = 7000
            depth_colorized = np.interp(depth_map, (min_depth, max_depth), (0, 255)).astype(np.uint8)
            depth_colorized = cv2.applyColorMap(depth_colorized, cv2.COLORMAP_JET)

            # Display the modified image
            cv2.imshow("Colorized depth", depth_colorized)
            #cv2.createTrackbar('Filtration', 'Colorized depth', 0, 255, nothing)
            counter += 1

    device.close()
    print('Done')```

dmn-seasony, 2024-04-30:
The issues seem to be fixed, and with the script here we get good quality colored point clouds: https://github.com/luxonis/depthai-python/blob/reprojection_depth_alignment/examples/Reprojection/reprojection_tof.py

closed 2024-04-30 by dmn-seasony