luxonis/depthai-core #996: [BUG] {nlohmann_json Hunter Download error ("Failure when receiving data from the peer")}

onthegrid007, 2024-04-02:
**Describe the bug**
```
[hunter *** DEBUG *** 2024-04-02T14:42:25] Package 'nlohmann_json' default arguments: 'JSON_BuildTests=OFF'
[hunter *** DEBUG *** 2024-04-02T14:42:25] Add extra CMake args: 'JSON_BuildTests' = 'OFF'
[hunter *** DEBUG *** 2024-04-02T14:42:25] Locking directory: /home/localadmin/.hunter/_Base/Cache/meta/87f71cb/nlohmann_json/3.11.3/2074caa
[hunter *** DEBUG *** 2024-04-02T14:42:25] Lock done
[hunter *** DEBUG *** 2024-04-02T14:42:25] Locking directory: /home/localadmin/.hunter/_Base/Cache/meta/87f71cb/nlohmann_json/3.11.3/2074caa/151cbba
[hunter *** DEBUG *** 2024-04-02T14:42:25] Lock done
[hunter *** DEBUG *** 2024-04-02T14:42:25] Locking directory: /home/localadmin/.hunter/_Base/Cache/meta/87f71cb/nlohmann_json/3.11.3/2074caa/151cbba/a23c16c
[hunter *** DEBUG *** 2024-04-02T14:42:25] Lock done
[hunter *** DEBUG *** 2024-04-02T14:42:25] Locking directory: /home/localadmin/.hunter/_Base/Cache/meta/87f71cb/nlohmann_json/3.11.3/2074caa/151cbba/a23c16c/da39a3e
[hunter *** DEBUG *** 2024-04-02T14:42:25] Lock done
[hunter *** DEBUG *** 2024-04-02T14:42:25] Locking directory: /home/localadmin/.hunter/_Base/Cache
[hunter *** DEBUG *** 2024-04-02T14:42:25] Lock done
[hunter *** DEBUG *** 2024-04-02T14:42:25] Using environment variable HOME
[hunter *** DEBUG *** 2024-04-02T14:42:25] Downloading DONE metafile (try #0 of 10):
[hunter *** DEBUG *** 2024-04-02T14:42:25]   https://raw.githubusercontent.com/cpp-pm/hunter-cache/master/87f71cb/nlohmann_json/3.11.3/2074caa/151cbba/a23c16c/da39a3e/basic-deps.DONE
[hunter *** DEBUG *** 2024-04-02T14:42:25]   -> /home/localadmin/.hunter/_Base/Cache/meta/87f71cb/nlohmann_json/3.11.3/2074caa/151cbba/a23c16c/da39a3e/basic-deps.DONE
[hunter *** DEBUG *** 2024-04-02T14:42:26] Download error ("Failure when receiving data from the peer")
[hunter *** DEBUG *** 2024-04-02T14:42:26] Downloading DONE metafile (try #1 of 10):
[hunter *** DEBUG *** 2024-04-02T14:42:26]   https://raw.githubusercontent.com/cpp-pm/hunter-cache/master/87f71cb/nlohmann_json/3.11.3/2074caa/151cbba/a23c16c/da39a3e/basic-deps.DONE
[hunter *** DEBUG *** 2024-04-02T14:42:26]   -> /home/localadmin/.hunter/_Base/Cache/meta/87f71cb/nlohmann_json/3.11.3/2074caa/151cbba/a23c16c/da39a3e/basic-deps.DONE
[hunter *** DEBUG *** 2024-04-02T14:42:26] Download error ("Failure when receiving data from the peer")
[hunter *** DEBUG *** 2024-04-02T14:42:26] Downloading DONE metafile (try #2 of 10):
[hunter *** DEBUG *** 2024-04-02T14:42:26]   https://raw.githubusercontent.com/cpp-pm/hunter-cache/master/87f71cb/nlohmann_json/3.11.3/2074caa/151cbba/a23c16c/da39a3e/basic-deps.DONE
[hunter *** DEBUG *** 2024-04-02T14:42:26]   -> /home/localadmin/.hunter/_Base/Cache/meta/87f71cb/nlohmann_json/3.11.3/2074caa/151cbba/a23c16c/da39a3e/basic-deps.DONE
[hunter *** DEBUG *** 2024-04-02T14:42:26] Retry after 5 seconds (attempt #2) ...
[hunter *** DEBUG *** 2024-04-02T14:42:31] Download error ("Failure when receiving data from the peer")
[hunter *** DEBUG *** 2024-04-02T14:42:31] Downloading DONE metafile (try #3 of 10):
[hunter *** DEBUG *** 2024-04-02T14:42:31]   https://raw.githubusercontent.com/cpp-pm/hunter-cache/master/87f71cb/nlohmann_json/3.11.3/2074caa/151cbba/a23c16c/da39a3e/basic-deps.DONE
[hunter *** DEBUG *** 2024-04-02T14:42:31]   -> /home/localadmin/.hunter/_Base/Cache/meta/87f71cb/nlohmann_json/3.11.3/2074caa/151cbba/a23c16c/da39a3e/basic-deps.DONE
[hunter *** DEBUG *** 2024-04-02T14:42:31] Retry after 5 seconds (attempt #3) ...
[hunter *** DEBUG *** 2024-04-02T14:42:36] Download error ("Failure when receiving data from the peer")
```

**Minimal Reproducible Example**
Clone depthai-core
mkdir depthai-core/build
cd depthai-core/build
cmake -G Ninja .. -DHUNTER_STATUS_DEBUG=1

**Expected behavior**
It builds correctly. Hunter shouldn't have any issues.

I tried as you can see in my logs to update nlohmann_json to 3.11.3 rather than the default to see if it was a version issue but it seems to happen with all versions, checksum not withstanding

onthegrid007, 2024-04-03:
It is also now happening with the XLink hunter repo....

moratom, 2024-04-04:
@onthegrid007 would you mind sharing a bit more about your setup? (OS, ...)

I can't reproduce on Ubuntu 20.04 with stock packages.

onthegrid007, 2024-04-04:
Arch Linux
6.6.23-1-lts
Cmake: 3.29.1
Ninja: 1.11.1

onthegrid007, 2024-04-04:
From what I can research it looks like a semi-common libcurl bug I'm not sure what hunter/cmake gets repos with but I don't know exactly how to fix it

moratom, 2024-04-16:
@jakgra do you wind giving it a look when you get a chance?

Zimamazim, 2024-04-22:
Workaround is to downgrade `curl`. Downgrading to ` curl-8.6.0-3` worked for me. See https://wiki.archlinux.org/title/Downgrading_packages if you're unsure how to do that.

onthegrid007, 2024-04-22:
Curl bug, thanks. And yes I know how to downgrade

closed 2024-04-22 by themarpe

onthegrid007, 2024-05-22:
I attempted to downgrade but still no luck for curl