luxonis/depthai-core #999: [BUG] Catching errors

gabrielmercado90, 2024-04-08:
**Describe the bug**
I am trying to catch the error or to reconnect if the camera is disconnected due to the cable or any other error. but this method is not returning the values. is there any other way to catch this errors? and also is there a Generic error handler? o where can see all the possible errors.

I am using thet c++ API (V2.24.0)

**Additional context**
The error handling documentation is not really clear, I would appreciate your help :3

moratom, 2024-04-16:
Hi @gabrielmercado90

At the moment there is no way of getting the actual error that caused the crash/disconnect.

We plan to add it in on the next major release.

gabrielmercado90, 2024-04-16:
Ok, thank you @moratom !!

closed 2024-04-16 by gabrielmercado90