luxonis/depthai-core #1001: [Documentation] Use depthai-core library in Android

christiangda, 2024-04-14:
### Start with the `why`:

Because I want to use my `OAK-1` Camera on an `Android Device` and apart from the __non-updated project__ [luxonis/depthai-android-jni-example](https://github.com/luxonis/depthai-android-jni-example) there is not detailed documentation about it. So I want to leave here my `howto notes`

### Move to the `what`:
__Clear instructions__ to `compile` and use the `depthai-core` library in an `Android Device` using `Android Studio`.

### Move to the `how`:

I achieved this after spending some time in my `Macbook Pro M1` with the following instructions, some of them coming from the [https://github.com/luxonis/depthai-core/blob/main/README.md](https://github.com/luxonis/depthai-core/blob/main/README.md).  

These are the minimal `howto` instructions for somebody who wants to achieve the same 

#### Requirements

+ [Android Studio](https://developer.android.com/studio)
+ [Android NDK](https://developer.android.com/ndk) -> I installed this using Android Studio
+ Some Environment Variables after installing both of them

#### Compile detphai-core

Add these environment variables to my shell configuration

```bash
cat >> ~/.zprofile<< _EOF

# Android SDK
export ANDROID_HOME=\$HOME/Library/Android/sdk
export PATH=\$PATH:\$ANDROID_HOME/emulator:$ANDROID_HOME/platform-tools
export NDK=\$ANDROID_HOME/ndk/25.1.8937393/

export ANDROID_SDK_HOME=\$ANDROID_HOME
export ANDROID_NDK_HOME=\$ANDROID_HOME/ndk/25.1.8937393/

export ANDROID_NDK=\$ANDROID_HOME/ndk/25.1.8937393/
export ANDROID_SDK_ROOT=\$ANDROID_SDK_HOME
export ANDROID_NATIVE_API_LEVEL=24
export STRIP=\$ANDROID_NDK/toolchains/llvm/prebuilt/darwin-x86_64/bin/llvm-strip
```

or ensure you have at least the env var `NDK` in the shell terminal when you compile `detphai-core` lib

```bash
# where
cd ~
mkdir -p git/github.com/luxonis
cd git/github.com/luxonis/

# source code
git clone git@github.com:luxonis/depthai-core.git
cd depthai-core/
git submodule update --init --recursive

# This is because MacOS has a different SSL backend and there is a problem with it
# curl when cmake is trying to download the dependencies
export CURL_SSL_BACKEND=SecureTransport

# where I want to put the library, headers and cmake files
mkdir -p ~/Projects/ExternalLibs/

cmake -S. -Bbuild \
 -D CMAKE_INSTALL_PREFIX=~/Projects/ExternalLibs/depthai-core \
 -D CMAKE_TOOLCHAIN_FILE=$NDK/build/cmake/android.toolchain.cmake \
 -D CMAKE_CONFIGURATION_TYPES=Release \
 -D BUILD_SHARED_LIBS=ON \
 -D ANDROID_ABI=arm64-v8a \
 -D ANDROID_PLATFORM=android-33

cmake --build build --config Release --parallel 8
cmake --build build --target install

unset CURL_SSL_BACKEND
```

#### Configure your Android c++ Project

Lines added to the default `CMakeLists.txt` file

```cmake
# I used this to understand how to tell cmake where to find the depthai-core lib, very useful
#set(CMAKE_FIND_DEBUG_MODE on)

# the way I configured to tell cmake where to find the detphai-core lib, the name must be depthai in lowercase
# depthai-core library
set(depthai_DIR "/Users/christian/Projects/ExternalLibs/depthai-core/lib/cmake/depthai" CACHE PATH "depthai-core dir compiled for Android")
set(depthai_ROOT "/Users/christian/Projects/ExternalLibs/depthai-core/lib/cmake/depthai" CACHE PATH "depthai-core root dir compiled for Android")
set(depthai_STATIC on)
find_package(depthai CONFIGS REQUIRED)

# ...

target_link_libraries(${CMAKE_PROJECT_NAME}                                     
        # List libraries link to the target library                             
        android                                                                 
        ${depthai}      # <- remember to link the library                                                       
        log)                                                                    
```

This is the full `CMakeLists.txt` file 

```cmake
# For more information about using CMake with Android Studio, read the
# documentation: https://d.android.com/studio/projects/add-native-code.html.
# For more examples on how to use CMake, see https://github.com/android/ndk-samples.

# Sets the minimum CMake version required for this project.
cmake_minimum_required(VERSION 3.22.1)

# Declares the project name. The project name can be accessed via ${ PROJECT_NAME},
# Since this is the top level CMakeLists.txt, the project name is also accessible
# with ${CMAKE_PROJECT_NAME} (both CMake variables are in-sync within the top level
# build script scope).
project("aicounter")
#set(CMAKE_FIND_DEBUG_MODE on)

# depthai-core library
set(depthai_DIR "/Users/christian/Projects/ExternalLibs/depthai-core/lib/cmake/depthai" CACHE PATH "depthai-core dir compiled for Android")
set(depthai_ROOT "/Users/christian/Projects/ExternalLibs/depthai-core/lib/cmake/depthai" CACHE PATH "depthai-core root dir compiled for Android")
set(depthai_STATIC on)
find_package(depthai CONFIGS REQUIRED)

# Creates and names a library, sets it as either STATIC
# or SHARED, and provides the relative paths to its source code.
# You can define multiple libraries, and CMake builds them for you.
# Gradle automatically packages shared libraries with your APK.
#
# In this top level CMakeLists.txt, ${CMAKE_PROJECT_NAME} is used to define
# the target library name; in the sub-module's CMakeLists.txt, ${PROJECT_NAME}
# is preferred for the same purpose.
#
# In order to load a library into your app from Java/Kotlin, you must call
# System.loadLibrary() and pass the name of the library defined here;
# for GameActivity/NativeActivity derived applications, the same library name must be
# used in the AndroidManifest.xml file.
add_library(${CMAKE_PROJECT_NAME}

        # This library is shared
        SHARED

        # List C/C++ source files with relative paths to this CMakeLists.txt.
        native-lib.cpp)

# Specifies libraries CMake should link to your target library. You
# can link libraries from various origins, such as libraries defined in this
# build script, prebuilt third-party libraries, or Android system libraries.
target_link_libraries(${CMAKE_PROJECT_NAME}
        # List libraries link to the target library
        android
        ${depthai}
        log)
```

Build you `Android Project` and check the build console

moratom, 2024-04-16:
@christiangda thanks a lot!

CC: @Erol444 to possibly port/include this to our documentation.

GerHelWan, 2025-02-18:
I tried to build the tag (v.2.29.0) for Android arm64 following this documentation.
@christiangda Thanks a lot!

I had some trouble with libsusb, curl and openSSL on my Linux Ubuntu 22.04 machine.

By adding the following options I succeeded the configuration and build.

```shell 
-D DEPTHAI_ENABLE_CURL=OFF \
-D DEPTHAI_ENABLE_LIBUSB=ON \
```