luxonis/depthai-core #1006: [BUG] Unable to access to 4 TOFs from 4 S5K33D through OAK-FFC-4P

Thinkminator, 2024-04-19:
I have purchased a Luxonis OAK-FFC 4P and 4 S5K33D recently, and the goal is to view TOFs from 4 cameras at the same time. However, if I run the command $python3 cam_test.py --cameras cama,t camb,t camc,t camd,t, I am only getting tof from cama, and the thing crashed in 2s.
[image]

Then, I try with something lighter which is to access only the color views from all 4 cameras concurrently by running $ python3 cam_test.py --cameras cama,c camb,c camc,c camd,c. This time, no error is received, but only 2 views from cama and camd are shown (which are the MIPI 4 lanes).
[image]

However, if I unplug 1 camera  from MIPI 2 lane, then 3 cameras will be supported for color view. Therefore, I am wondering, is it possible for us to view 4 tofs through OAK-FFC 4P concurrently, if so are there any additional steps that need to be taken, also is it also feasible for OAK-FFC 4P PoE?

Thank you very much.

MartinMotycka, 2024-05-09:
@themarpe ^^

alex-luxonis, 2024-05-13:
I think that the same issue was further discussed here:
https://github.com/luxonis/depthai-python/issues/1007