luxonis/depthai-core #1023: Feature Request: Add `maxSensitivityIso` to `setAutoExposureLimit()`

maxsitt, 2024-05-15:
With [`setAutoExposureLimit()`](https://docs.luxonis.com/projects/api/en/latest/references/python/#depthai.CameraControl.setAutoExposureLimit) a limit for the maximum exposure time can be set. This is very helpful, e.g. to avoid overexposed frames/images.

With the current implementation, only the maximum exposure time can be set. Depending on the configuration (e.g. using [`setAutoExposureRegion()`](https://docs.luxonis.com/projects/api/en/latest/references/python/#depthai.CameraControl.setAutoExposureRegion)), the produced frames can still be too bright, as ISO is increased even with the auto exposure limit.

Would it be possible to add a second parameter for `maxSensitivityIso` as in [`setManualExposure()`](https://docs.luxonis.com/projects/api/en/latest/references/python/#depthai.CameraControl.setManualExposure) also for `setAutoExposureLimit()`? This would give us more control over the camera behaviour under challenging lighting conditions and could avoid frames getting too bright due to high ISO values.

Thanks!

themarpe, 2024-05-15:
@alex-luxonis is this possible?

Erol444, 2024-06-03:
ping @alex-luxonis

maxsitt, 2024-06-15:
@alex-luxonis Thanks for looking into this!

Could you explain what you mean with "tweak the tuning"? Sorry that I don't understand what exactly you mean by this.

The possibility to set a range or limit for not only auto exposure time, but also ISO would definitely benefit various scenarios. In our use case, we would like to avoid frames getting too bright when using [setAutoExposureRegion()](https://docs.luxonis.com/projects/api/en/latest/references/python/#depthai.CameraControl.setAutoExposureRegion) in combination with detections. In our tests, we could often observe very bright frames due to small, dark (sometimes false) detections, which in turn influences the detection model.

themarpe, 2024-06-17:
@maxsitt aprt form modiyfing the imager tuning files (which requires Intel tools), there is no means of achieving this on RVC2 platform.

We'll have to close as `wontfix` as its not easily doable

closed 2024-06-17 by themarpe