luxonis/depthai-core #1027: Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 80290064' '0'

younkun, 2024-05-24:
[1844301011125D0F00] [169.254.1.222] [63.971] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 80290064' '0'
[1844301011125D0F00] [169.254.1.222] [1716539506.050] [host] [warning] Device crashed. Crash dump saved to /tmp/depthai_T9MXQF/1844301011125D0F00-depthai_crash_dump.json
Traceback (most recent call last):
  File "main_api.py", line 572, in <module>
    main()
  File "main_api.py", line 260, in main
    image_queue_data = image_queue.tryGet()
RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'rgb' (X_LINK_ERROR)'

I just run yolov8n_coco_640x352  on ubuntu20.0（depthai==2.25.1.0），and the code crashes after a few minutes of execution. The error message is as follows. How can I deal with this bug?"[https://github.com/richard-xx/DepthAI_Yolo](url)

moratom, 2024-05-24:
Hi @younkun !

Thanks a lot for the bug report and sorry for the issues.

Would you happen to have the crash dump available and could share it here?
This one - ```/tmp/depthai_T9MXQF/1844301011125D0F00-depthai_crash_dump.json```.

younkun, 2024-05-27:
[1844301011125D0F00-depthai_crash_dump(1).json](https://github.com/luxonis/depthai-core/files/15451357/1844301011125D0F00-depthai_crash_dump.1.json)
Thank you so much！this is the bug report file。

moratom, 2024-05-31:
@younkun would you mind trying the latest `v2.26.0.0` release?

It has quite a few stability improvements.

Shivam7Sharma, 2024-06-03:
I am using v2.26.0.0 and still getting this error. 
Note: I have connected only one color camera but when I run the cam_test file it shows two on different sockets.

```
shivam157@ubuntu:~/.local/lib/python3.10/site-packages/depthai_cli$ python cam_test.py -cam left,c right,c 
DepthAI version: 2.26.0.0
DepthAI path: /home/shivam157/.local/lib/python3.10/site-packages/depthai.cpython-310-aarch64-linux-gnu.so
Enabled cameras:
   left : color
  right : color
CAM:  left
CAM:  right
Connected cameras:
 -socket CAM_B : IMX378 4056 x 3040 focus:auto  - MONO
 -socket CAM_C : IMX378 4056 x 3040 focus:auto  - MONO
USB speed: SUPER
IR drivers: []
Cam:      left          right    [host | capture timestamp]
FPS:   0.00|  0.00   0.00|  0.00 [14442C10E14FCBD600] [1.2.4] [5.310] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 00, address: 00000000' '0'
FPS:   0.00|  0.00   0.00|  0.00 Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'left' (X_LINK_ERROR)'
Exiting cleanly
[14442C10E14FCBD600] [1.2.4] [1717444874.024] [host] [warning] Device crashed. Crash dump saved to /tmp/depthai_kcaFxI/14442C10E14FCBD600-depthai_crash_dump.json

```

The log file is as follows:
[14442C10E14FCBD600-depthai_crash_dump.json](https://github.com/user-attachments/files/15539103/14442C10E14FCBD600-depthai_crash_dump.json)

What is this error? Please help solve this.

moratom, 2024-06-03:
@SzabolcsGergely could you have a look at the crash?

SzabolcsGergely, 2024-06-04:
The crash dump suggest camera driver crash/hang.
@Shivam7Sharma please try socket A for IMX378

moratom, 2024-08-16:
@Shivam7Sharma were you able to test it out?