luxonis/depthai-core #1040: [BUG] error: no member named 'numShaves' in 'dai::ToFProperties'

christiangda, 2024-06-16:
**Describe the bug**
I had implemented on a project depthai-core (using git submodule) v1.25.0 and today I tried to update to the last version v1.26.0

The commands I used on my project to implement the `depthai-core` library is

```bash
git submodule add --force https://github.com/luxonis/depthai-core.git 3rdparty/depthai-core
git submodule update --init --recursive --remote
git -C 3rdparty/depthai-core checkout tags/v2.26.0
git add 3rdparty/depthai-core
git commit -m "update depthai-core to v2.26.0"
git submodule status
```

**Minimal Reproducible Example**

```bash
rm -rf build
❯ cmake -S. -Bbuild -DCMAKE_BUILD_TYPE=Release -DCMAKE_EXPORT_COMPILE_COMMANDS=YES # <-- this OK
❯ cmake --build build  # <-- this KO, see the log below
```

Ouptut of `cmake -S. -Bbuild -DCMAKE_BUILD_TYPE=Release -DCMAKE_EXPORT_COMPILE_COMMANDS=YES`

```bash
...
-- Check for working incbin assembly directive
-- Check for working incbin assembly directive - works
-- Performing Test FLAG_-Wno-switch-enum
-- Performing Test FLAG_-Wno-switch-enum - Success
-- OpenCV and required libraries (opencv_core;opencv_imgproc) found. OpenCV Support enabled
-- BUILD_DATETIME: 2024-06-16 16:10:44 +0000, BUILD_COMMIT: 90487456cb92d15bcf0129a2f54c332250f1df31, BUILD_COMMIT_DATETIME: 2024-05-25 08:30:10 +0200
-- Found OpenCV: /opt/homebrew/Cellar/opencv/4.9.0_9 (found version "4.9.0")
-- Found Boost: /opt/homebrew/lib/cmake/Boost-1.85.0/BoostConfig.cmake (found suitable version "1.85.0", minimum required is "1.74.0") found components: program_options log system log_setup filesystem
-- Found OpenSSL: /opt/homebrew/lib/libcrypto.dylib (found version "3.3.1")
-- Configuring done (9.1s)
-- Generating done (0.1s)
-- Build files have been written to: /Users/christian/git/github.com/christiangda/FaceDetector/build
  ~/git/github.com/christiangda/FaceDetector on   feat-depthi-1.26.0 !1     
```

Output of `cmake --build build`

```bash
...
❯ cmake --build build
[  1%] Generating depthai-resources resource loader
[  2%] Generating intermediate file for /Users/christian/git/github.com/christiangda/FaceDetector/build/3rdparty/depthai-core/resources/depthai-bootloader-fwp-0.0.28.tar.xz
[  3%] Generating intermediate file for /Users/christian/git/github.com/christiangda/FaceDetector/build/3rdparty/depthai-core/resources/depthai-device-fwp-24a3b465b979de3f69410cd225914d8bd029f3ba.tar.xz
[  4%] Building CXX object 3rdparty/depthai-core/CMakeFiles/depthai-resources.dir/__cmrc_depthai-resources/lib.cpp.o
[  5%] Building CXX object 3rdparty/depthai-core/CMakeFiles/depthai-resources.dir/__cmrc_depthai-resources/intermediate/depthai-device-fwp-24a3b465b979de3f69410cd225914d8bd029f3ba.tar.xz.cpp.o
[  6%] Building CXX object 3rdparty/depthai-core/CMakeFiles/depthai-resources.dir/__cmrc_depthai-resources/intermediate/depthai-bootloader-fwp-0.0.28.tar.xz.cpp.o
[  7%] Linking CXX static library libdepthai-resources.a
[  7%] Built target depthai-resources
[  8%] Building CXX object 3rdparty/depthai-core/CMakeFiles/depthai-core.dir/shared/depthai-shared/src/datatype/DatatypeEnum.cpp.o
[  9%] Building CXX object 3rdparty/depthai-core/CMakeFiles/depthai-core.dir/shared/depthai-shared/src/utility/Checksum.cpp.o
[ 10%] Building C object 3rdparty/depthai-core/CMakeFiles/depthai-core.dir/shared/depthai-bootloader-shared/src/SBR.c.o
[ 11%] Building CXX object 3rdparty/depthai-core/CMakeFiles/depthai-core.dir/shared/depthai-bootloader-shared/src/Bootloader.cpp.o
[ 12%] Building CXX object 3rdparty/depthai-core/CMakeFiles/depthai-core.dir/src/device/Device.cpp.o
[ 13%] Building CXX object 3rdparty/depthai-core/CMakeFiles/depthai-core.dir/src/device/DeviceBase.cpp.o
[ 14%] Building CXX object 3rdparty/depthai-core/CMakeFiles/depthai-core.dir/src/device/DeviceBootloader.cpp.o
[ 15%] Building CXX object 3rdparty/depthai-core/CMakeFiles/depthai-core.dir/src/device/DataQueue.cpp.o
/Users/christian/git/github.com/christiangda/FaceDetector/3rdparty/depthai-core/src/device/DataQueue.cpp:37:23: warning: variable 'numPacketsRead' set but not used [-Wunused-but-set-variable]
        std::uint64_t numPacketsRead = 0;
                      ^
/Users/christian/git/github.com/christiangda/FaceDetector/3rdparty/depthai-core/src/device/DataQueue.cpp:203:23: warning: variable 'numPacketsSent' set but not used [-Wunused-but-set-variable]
        std::uint64_t numPacketsSent = 0;
                      ^
2 warnings generated.
[ 16%] Building CXX object 3rdparty/depthai-core/CMakeFiles/depthai-core.dir/src/device/CallbackHandler.cpp.o
[... 135 lines cut ...]
[ 81%] Building CXX object 3rdparty/depthai-core/CMakeFiles/depthai-core.dir/src/utility/XLinkGlobalProfilingLogger.cpp.o
[ 82%] Building CXX object 3rdparty/depthai-core/CMakeFiles/depthai-core.dir/src/utility/Logging.cpp.o
[ 83%] Building CXX object 3rdparty/depthai-core/CMakeFiles/depthai-core.dir/src/utility/EepromDataParser.cpp.o
[ 84%] Building CXX object 3rdparty/depthai-core/CMakeFiles/depthai-core.dir/src/xlink/XLinkConnection.cpp.o
[ 85%] Building CXX object 3rdparty/depthai-core/CMakeFiles/depthai-core.dir/src/xlink/XLinkStream.cpp.o
[ 86%] Building CXX object 3rdparty/depthai-core/CMakeFiles/depthai-core.dir/src/openvino/OpenVINO.cpp.o
[ 87%] Building CXX object 3rdparty/depthai-core/CMakeFiles/depthai-core.dir/src/openvino/BlobReader.cpp.o
[ 88%] Building C object 3rdparty/depthai-core/CMakeFiles/depthai-core.dir/src/bspatch/bspatch.c.o
[ 89%] Linking CXX static library libdepthai-core.a
[ 89%] Built target depthai-core
[ 90%] Building CXX object 3rdparty/depthai-core/CMakeFiles/depthai-opencv.dir/src/opencv/ImgFrame.cpp.o
[ 91%] Linking CXX static library libdepthai-opencv.a
[ 91%] Built target depthai-opencv
[ 91%] Building CXX object CMakeFiles/FaceDetector.dir/src/main.cpp.o
[ 92%] Building CXX object CMakeFiles/FaceDetector.dir/src/FaceDetector.cpp.o
[ 93%] Building CXX object CMakeFiles/FaceDetector.dir/src/FaceAgeGenderDetector.cpp.o
[ 94%] Building CXX object CMakeFiles/FaceDetector.dir/src/MetricSender.cpp.o
[ 95%] Building CXX object CMakeFiles/FaceDetector.dir/src/CmdOptions.cpp.o
[ 96%] Building CXX object CMakeFiles/FaceDetector.dir/src/Utils.cpp.o
[ 97%] Building CXX object CMakeFiles/FaceDetector.dir/src/License.cpp.o
[ 98%] Building CXX object CMakeFiles/FaceDetector.dir/src/Camera.cpp.o
[100%] Linking CXX executable FaceDetector
[100%] Built target FaceDetector

```

SzabolcsGergely, 2024-06-18:
You should update submodules, specifically depthai-shared in this case.

christiangda, 2024-06-19:
@SzabolcsGergely thank you very much, I will try to see if this solve my problem

christiangda, 2024-06-19:
I closed this, as you mentioned @SzabolcsGergely the problem was the git submodule update.

Now this is working on my project

closed 2024-06-19 by christiangda