luxonis/depthai-core #1044: [BUG] depthai-core.dll will not link

onthegrid007, 2024-06-24:
Is there something I am doing wrong?

```
[1/3] Linking CXX shared library vendor\depthai-core\depthai-core.dll
FAILED: vendor/depthai-core/depthai-core.dll vendor/depthai-core/depthai-core.lib
C:\Windows\system32\cmd.exe /C "C:\Windows\system32\cmd.exe /C ""C:\Program Files\CMake\bin\cmake.exe" -E __create_def C:\gitlibs\FRCDepth\build\vendor\depthai-core\CMakeFiles\depthai-core.dir\.\exports.def C:\gitlibs\FRCDepth\build\vendor\depthai-core\CMakeFiles\depthai-core.dir\.\exports.def.objs --nm="C:\Program Files\LLVM\bin\llvm-nm.exe" && cd C:\gitlibs\FRCDepth\build" && C:\PROGRA~1\LLVM\bin\CLANG_~1.EXE -nostartfiles -nostdlib -O3 -DNDEBUG -D_DLL -D_MT -Xclang --dependent-lib=msvcrt  /DELAYLOAD:libusb-1.0.dll -fuse-ld=lld-link -Xlinker /DEF:vendor\depthai-core\CMakeFiles\depthai-core.dir\.\exports.def -shared -o vendor\depthai-core\depthai-core.dll  -Xlinker /MANIFEST:EMBED -Xlinker /implib:vendor\depthai-core\depthai-core.lib -Xlinker /pdb:vendor\depthai-core\depthai-core.pdb -Xlinker /version:0.0 @CMakeFiles\depthai-core.rsp  && C:\Windows\system32\cmd.exe /C "cd /D C:\gitlibs\FRCDepth\build\vendor\depthai-core && "C:\Program Files\CMake\bin\cmake.exe" -E copy_if_different C:/Users/localadmin/.hunter/_Base/cb0ea1f/6f97ead/7f91c96/Install/bin/libusb-1.0.dll C:/Users/localadmin/.hunter/_Base/cb0ea1f/6f97ead/7f91c96/Install/bin/libusb-1.0d.dll C:/gitlibs/FRCDepth/build/vendor/depthai-core""
clang++: error: no such file or directory: '/DELAYLOAD:libusb-1.0.dll'
ninja: build stopped: subcommand failed.
```

themarpe, 2024-06-24:
@onthegrid007 seems like Clang on Windows doesn't support DELAYLOAD functionality

Fastest would be to use MSVC for now. Or tweaks done to XLink dependency

onthegrid007, 2024-06-26:
@themarpe
I have fooled around looking for where the flag is inserted but can't find it out... any tricks?

themarpe, 2024-06-26:
The flag comes from XLink - its the win32 path for loading libusb

onthegrid007, 2024-06-28:
Added a fix to a forked repo, it works for the first error but a new error has occurred where it doesn't seem to link with opencv itself, I assume a missing lib link flag or something...

```
[3/4] Linking CXX shared library vendor\depthai-core\depthai-opencvd.dll
FAILED: vendor/depthai-core/depthai-opencvd.dll vendor/depthai-core/depthai-opencvd.lib
C:\Windows\system32\cmd.exe /C "C:\Windows\system32\cmd.exe /C ""C:\Program Files\CMake\bin\cmake.exe" -E __create_def C:\gitlibs\FRCDepth\build\vendor\depthai-core\CMakeFiles\depthai-opencv.dir\.\exports.def C:\gitlibs\FRCDepth\build\vendor\depthai-core\CMakeFiles\depthai-opencv.dir\.\exports.def.objs --nm="C:\Program Files\LLVM\bin\llvm-nm.exe" && cd C:\gitlibs\FRCDepth\build" && C:\PROGRA~1\LLVM\bin\CLANG_~1.EXE -nostartfiles -nostdlib -O0 -g -Xclang -gcodeview -D_DEBUG -D_DLL -D_MT -Xclang --dependent-lib=msvcrtd  -fuse-ld=lld-link -Xlinker /DEF:vendor\depthai-core\CMakeFiles\depthai-opencv.dir\.\exports.def -shared -o vendor\depthai-core\depthai-opencvd.dll  -Xlinker /MANIFEST:EMBED -Xlinker /implib:vendor\depthai-core\depthai-opencvd.lib -Xlinker /pdb:vendor\depthai-core\depthai-opencvd.pdb -Xlinker /version:0.0 vendor/depthai-core/CMakeFiles/depthai-opencv.dir/src/opencv/ImgFrame.cpp.obj  C:/gitlibs/opencv/build-clang/lib/opencv_imgproc4100.lib  vendor/depthai-core/depthai-cored.lib  C:/gitlibs/opencv/build-clang/lib/opencv_core4100.lib  -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 -loldnames  && cd ."
lld-link: error: undefined symbol: void __cdecl cv::cvtColor(class cv::debug_build_guard::_InputArray const &, class cv::debug_build_guard::_OutputArray const &, int, int)
>>> referenced by C:\gitlibs\FRCDepth\vendor\depthai-core\src\opencv\ImgFrame.cpp:104
>>>               vendor/depthai-core/CMakeFiles/depthai-opencv.dir/src/opencv/ImgFrame.cpp.obj:(public: class cv::Mat __cdecl dai::ImgFrame::getCvFrame(void))
>>> referenced by C:\gitlibs\FRCDepth\vendor\depthai-core\src\opencv\ImgFrame.cpp:132
>>>               vendor/depthai-core/CMakeFiles/depthai-opencv.dir/src/opencv/ImgFrame.cpp.obj:(public: class cv::Mat __cdecl dai::ImgFrame::getCvFrame(void))
>>> referenced by C:\gitlibs\FRCDepth\vendor\depthai-core\src\opencv\ImgFrame.cpp:136
>>>               vendor/depthai-core/CMakeFiles/depthai-opencv.dir/src/opencv/ImgFrame.cpp.obj:(public: class cv::Mat __cdecl dai::ImgFrame::getCvFrame(void))
>>> referenced 1 more times

lld-link: error: undefined symbol: void __cdecl cv::merge(class cv::debug_build_guard::_InputArray const &, class cv::debug_build_guard::_OutputArray const &)
>>> referenced by C:\gitlibs\FRCDepth\vendor\depthai-core\src\opencv\ImgFrame.cpp:118
>>>               vendor/depthai-core/CMakeFiles/depthai-opencv.dir/src/opencv/ImgFrame.cpp.obj:(public: class cv::Mat __cdecl dai::ImgFrame::getCvFrame(void))
>>> referenced by C:\gitlibs\FRCDepth\vendor\depthai-core\src\opencv\ImgFrame.cpp:128
>>>               vendor/depthai-core/CMakeFiles/depthai-opencv.dir/src/opencv/ImgFrame.cpp.obj:(public: class cv::Mat __cdecl dai::ImgFrame::getCvFrame(void))
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
```

onthegrid007, 2024-06-29:
@themarpe 
It seems to be linking with imgproc and core properly but the actual function calls aren't being recognized, any thoughts?

themarpe, 2024-06-30:
@onthegrid007 Hmm, first thing to check is what OpenCV was being compiled against (eg which `-I/include/dirs` are part of the compilation line that compiles `src/opencv/ImgFrame.cpp`), and then which was it linked against.

onthegrid007, 2024-06-30:
I got it to compile in Release build mode bud had to make some dirty fixed to my cmake file and some abs paths to library files... Everything is included on the $PATH so the fact it is finding the dlls but not the lib files is interesting

onthegrid007, 2024-06-30:
Could it be because I build OpenCV in release vs debug?

themarpe, 2024-06-30:
It could yup - try building the core for same as OpenCV is

closed 2024-07-31 by onthegrid007