luxonis/depthai-core #1056: [BUG] Pipeline crash when undistorting color image using Camera Node

borongyuan, 2024-07-04:
**Check if issue already exists**

- I mentioned it in #948 but now it becomes more serious

**Describe the bug**
When doing various computer vision tasks, I undistort all camera images, no matter what their FOV is. StereoDepth Node performs undistortion and rectification on stereo images. For color image, I have been using Generic Camera Node, as suggested by @saching13 in https://github.com/introlab/rtabmap_ros/issues/1045. When using depthai-core v2.24, the Camera Node will automatically undistort image with a wide FOV, but for image with a standard FOV, I need to setMeshSource(dai::CameraProperties::WarpMeshSource::CALIBRATION) manually. The bug is that when the StereoDepth alignment is set to `dai::CameraBoardSocket::CAM_A` and the output of the Camera Node is connected to any neural network, the pipeline will crash.
`[19443010A106A22E00] [1.5] [1.444] [system] [critical] Fatal error. Please report to developers. Log: 'ImgPreproc' '562'`
This has previously only affected one of the features I was working on, and I've been waiting for it to be fixed. But when I upgraded to depthai-core v2.26, it was not only unfixed, but became more serious! Now even without using NeuralNetwork Node, the pipeline will crash as long as undistortion is enabled in Camera Node and the alignment of StereoNode is set to `dai::CameraBoardSocket::CAM_A`.
`[1844301091AD870E00] [1.3] [1.248] [system] [critical] Fatal error. Please report to developers. Log: 'ImgPreproc' '582'`
I really don't understand how you guys create more and more bugs and make them more and more serious.

SzabolcsGergely, 2024-07-08:
@borongyuan the issue has been fixed, please check out latest develop. It will be part of next release.

borongyuan, 2024-08-03:
I've tested v2.27, the issue still exist. I can't setMeshSource() to CALIBRATION or URI. The pipeline always crashes. What do you mean by next release? v3.0? We have trained an experimental stereo dispatity estimation model https://github.com/luxonis/depthai/issues/173#issuecomment-2172081778. In this part, we also want to use Camera Node to perform stereo rectification. If this issue cannot be resolved, the new neural-assisted disparity estimation method will be shelved. If you confirm this issue is solved, I would like you to provide a C++ example of setting a custom mesh in the Camera Node.

borongyuan, 2024-08-12:
I found a way to circumvent this issue. Just create an extra ImageManip node. It does not need to perform any actual function and does not need to connect to other nodes. Just create one to avoid pipeline crashes. So this is most likely caused by incorrect resource allocation.

moratom, 2024-08-16:
@SzabolcsGergely can we reproduce this bug on the latest release still?

SzabolcsGergely, 2024-08-16:
> @SzabolcsGergely can we reproduce this bug on the latest release still?

Needs latest develop, 2.27 still has the issue.

moratom, 2024-08-19:
@borongyuan are you able to verify that this solves the issue?

SzabolcsGergely, 2024-08-29:
Fixed in 2.28 [Release](https://github.com/luxonis/depthai-core/releases/tag/v2.28.0).

borongyuan, 2024-09-02:
Thanks, confirmed it's resolved