luxonis/depthai-core #1085: [BUG] Noisy point clouds near the bottom-right corner of OAK-D PRO camera

Luca0101, 2024-08-01:
Hello everyone, in our lab we are currently testing our OAK-D PRO camera to detect point clouds of cylindrical PVC pipes. However, we have noticed that, keeping a fixed distance from the camera (z-axis of the camera) of about 30 cm, the point clouds become increasingly noisy as we move closer to the bottom-right corner of the camera's field of view. Generally, we could say that the affected area is the triangle with vertices at the bottom-left corner, bottom-right corner, and top-right corner. I am attaching the three point clouds separately, as well as a point cloud with the three overlaid, and two photos of the current setup. The room is moderately lit without any direct light sources pointed at the camera. Any ideas on why this noise is occurring?
[Luxonis material.zip](https://github.com/user-attachments/files/16455011/Luxonis.material.zip)

Erol444, 2024-08-05:
Hi @Luca0101 ,
First, let me share docs on improving pointcloud noise:
https://docs.luxonis.com/hardware/platform/depth/configuring-stereo-depth/#Configuring%20Stereo%20Depth-6.%20Fixing%20noisy%20pointcloud
I'd say better lighting could help improve this, along with:
- Lower confidence threshold. this will result in less points, but also less noise
- Use decimation filter /w median mode: https://docs.luxonis.com/hardware/platform/depth/configuring-stereo-depth/#Configuring%20Stereo%20Depth-2.%20Fixing%20noisy%20depth-Stereo%20postprocessing%20filters-Decimation%20filter

Feel free to share pointclouds of the pipes after trying out these suggestions.
Thanks, Erik

Luca0101, 2024-08-08:
Hi @Erol444, thank you for these precious suggestions. We'll try them in the future as soon as we have some time and post the results here.
Gianluca