luxonis/depthai-core #1093: New xlink version

onthegrid007, 2024-08-10:
Hello,

I made a pull request to the xlink repository to allow clang as a working compiler on windows, Linux, etc. Now I am wondering how to get depthai and or hunter to pull the new updated repo so I don't have to use the depthai local xlink flag in cmake to get depthai-core to compile properly...

I have looked through the CMake file here in this repo and can't even find where xlink is being pulled into hunter from, otherwise I would make a pull request...

Any help would be appreciated!

-onthegrid007

themarpe, 2024-08-10:
@onthegrid007 the change can be done in `cmake/Hunter/config.cmake`

onthegrid007, 2024-08-11:
Found it thank you. Can I inquire, why not to master?

themarpe, 2024-08-11:
@onthegrid007 great!
Of course - mainly due to keeping `main` "stable", while the intermediary development, bugfixes, etc.. happen in `develop`. After everything is tested together will multiple PRs landed, then a release to `mian` is made

closed 2024-08-11 by themarpe