luxonis/depthai-core #1095: Libnop C++20 Bug

onthegrid007, 2024-08-11:
Hello,

So I have checked the current libnop version fork through the luxonis account and I see the updated changes to types/variant.h of libnop adding the template keyword to make sure it compiles with C++20 correctly but building the tests still have the former broken template macro being used, manually fixing this clears the bug and lets everything compile so it must be with the version hunter is pulling... I attempted to manually set the exact commit this was fixed by but it is not pulling down the newest version with the changes properly and I do not know as to why. With the new archive link, checksum, and version name changed it still pulls an outdated repo.

Any help would be appreciated as I can't really submit a pull request without it even working on my system first.

Thanks,
-otg007

themarpe, 2024-08-13:
@onthegrid007 try clearing hunter cache: `rm -rf ~/.hunter`
Otherwise working directly on the libnop repo might be simpler (if you mean libnop tests). DepthAI tests should function correctly, check if you are on correct branch /w all submodules updated

onthegrid007, 2024-08-14:
I will try this but even with pulling a fresh depthai-core repo libnop fails the c++20 test and I was going to put in a pull request with a newer libnop from the luxonis account but I see the c++ 20 fix commit and it's like 10 commits ahead of that so I just thought the repo being pulled by hunter was out of date... I'll try the hunter cache clearing and see if this fixes it

onthegrid007, 2024-08-15:
Found the error, the develop branch hasn't been pushed to master in over 2 years, the C++20 fix isn't in the current repo pulled by hunter... I am writing a patch...

onthegrid007, 2024-08-15:
[Done.](https://github.com/luxonis/libnop/pull/4)

onthegrid007, 2024-08-15:
There was also further template issues preventing depthai-core from compiling even with the latest develop branch...