luxonis/depthai-core #1103: [BUG] OAK-D-POE intermittent failure - INTERNAL_ERROR_CORE

laurence-diack-pk, 2024-08-19:
**Problem Description**
OAK-D-POE cameras intermittently disappear from the network and become unreachable via ping while running. The issue occurs unpredictably:

- Sometimes happens after about 5 minutes of operation
- Sometimes doesn't occur at all during a session
- Requires a power cycle to bring the cameras back online

**System Details**

- Camera Model: OAK-D-POE
- Main Application: Written in C++
- Network Configuration: 2x OAK-D-POEs sharing a subnet with a RPLIDAR-2E and Intel NUC
- Power Supply: POE
- System info attached here [log_system_information.json](https://github.com/user-attachments/files/16666925/log_system_information.json)

**Observed Behavior**

- The pipeline starts normally from a cold boot.
- At an unpredictable point, the cameras become unreachable on the network.
- Ping attempts to the camera IP addresses fail.
- Power cycling the cameras brings them back online.

**Crash Dumps**
Two crash dumps have been collected, showing the following errors:

`INTERNAL_ERROR_CORE`
`RTEMS_FATAL_SOURCE_EXCEPTION`

Crash dump files:
[crashDump_1_1844301031A3DC0E00_c7127782f2da45aac89d5b5b816d04cc45ae40be.json](https://github.com/user-attachments/files/16666893/crashDump_1_1844301031A3DC0E00_c7127782f2da45aac89d5b5b816d04cc45ae40be.json)
[crashDump_0_18443010C15E9F0F00_c7127782f2da45aac89d5b5b816d04cc45ae40be.json](https://github.com/user-attachments/files/16666894/crashDump_0_18443010C15E9F0F00_c7127782f2da45aac89d5b5b816d04cc45ae40be.json)

**Pipeline**
The main app is C++ so I can't get a pipeline grab, but here's the basic setup:
```
    // Create Nodes
    auto yolospatialdetectionnetwork = pipeline.create<dai::node::YoloSpatialDetectionNetwork>();
    auto camrgb = pipeline.create<dai::node::ColorCamera>();
    auto monoleft = pipeline.create<dai::node::MonoCamera>();
    auto monoright = pipeline.create<dai::node::MonoCamera>();
    auto stereo = pipeline.create<dai::node::StereoDepth>();
    auto objecttracker = pipeline.create<dai::node::ObjectTracker>();
    auto imu = pipeline.create<dai::node::IMU>();
    auto videoenc = pipeline.create<dai::node::VideoEncoder>();
    auto videoenc_webui = pipeline.create<dai::node::VideoEncoder>();
    auto videoenc_left = pipeline.create<dai::node::VideoEncoder>();
    auto videoenc_right = pipeline.create<dai::node::VideoEncoder>();
    auto manip = pipeline.create<dai::node::ImageManip>();
    auto xoutdepth = pipeline.create<dai::node::XLinkOut>();
    auto xouttracks = pipeline.create<dai::node::XLinkOut>();
    auto xoutdetections = pipeline.create<dai::node::XLinkOut>();
    auto xoutIMU = pipeline.create<dai::node::XLinkOut>();
    auto xoutvidenc = pipeline.create<dai::node::XLinkOut>();
    auto xoutmonoenc_left = pipeline.create<dai::node::XLinkOut>();
    auto xoutmonoenc_right = pipeline.create<dai::node::XLinkOut>();
    auto xoutvideostream = pipeline.create<dai::node::XLinkOut>();

    /// Set stream names for outputs
    xouttracks->setStreamName("tracklets");
    xoutdepth->setStreamName("depth");
    xoutdetections->setStreamName("detections");
    xoutvidenc->setStreamName("vid_enc");
    xoutmonoenc_left->setStreamName("mono_enc_left");
    xoutmonoenc_right->setStreamName("mono_enc_right");
    xoutvideostream->setStreamName("videostream");
    xoutIMU->setStreamName("imu");

    // Set properties for nodes
    camrgb->setPreviewSize(416, 416);
    camrgb->setInterleaved(false);
[... 114 lines cut ...]
    manip->out.link(videoenc_webui->input);
    videoenc_webui->bitstream.link(xoutvideostream->input);
    camrgb->video.link(videoenc->input);

    monoright->out.link(videoenc_right->input);
    monoleft->out.link(videoenc_left->input);
    videoenc->bitstream.link(xoutvidenc->input);
    videoenc_right->bitstream.link(xoutmonoenc_right->input);
    videoenc_left->bitstream.link(xoutmonoenc_left->input);

    stereo->depth.link(xoutdepth->input);

    imu->out.link(xoutIMU->input);

    camrgb->preview.link(yolospatialdetectionnetwork->input);
    stereo->depth.link(yolospatialdetectionnetwork->inputDepth);
    yolospatialdetectionnetwork->passthrough.link(objecttracker->inputTrackerFrame);
    yolospatialdetectionnetwork->passthrough.link(objecttracker->inputDetectionFrame);
    yolospatialdetectionnetwork->out.link(objecttracker->inputDetections);
    yolospatialdetectionnetwork->out.link(xoutdetections->input);

    objecttracker->out.link(xouttracks->input);
```

Any insights would be greatly appreciated, thanks

moratom, 2024-08-22:
Thanks for the bug report @laurence-diack-pk !

Just to clarify, the disconnects happen whilst you're running the app right?

moratom, 2024-08-22:
@SzabolcsGergely could you take a look at a crashdumps when you have a moment?

SzabolcsGergely, 2024-08-23:
> @SzabolcsGergely could you take a look at a crashdumps when you have a moment?

Crash occurred during a XLink read, in XLinkPlatformRead, reason unknown.

laurence-diack-pk, 2024-08-25:
> Thanks for the bug report @laurence-diack-pk !
> 
> Just to clarify, the disconnects happen whilst you're running the app right?

Yeah so it can happen while a pipeline is loaded, but also sometimes it will get into a state where it is working, then the pipeline is closed (ros node terminated / restarted) and it becomes unavailable between runs. 

It doesn't seem to be a very predictable failure and I'm having a hard time reproducing it consistently - for example I am looking at an instance right now where one of two cameras has disappeared, but I had to restart the host several times to get it into this state.

Also it may well be that the crashdumps are not a 1:1 correlation with this failure, as I have observed cases where it does this and no crashdump is retrieved.

Sorry for the vagueness, it's just sorta a black box from my end - if I can't communicate with the camera over network, it's hard to tell exactly what's going on.

I was wondering if there's any additional logging I can pull of the device itself, or perhaps some way in which I could use the M8 connector to debug over uart or usb so I can get some insight into the state of the camera when it disappears like that.

laurence-diack-pk, 2024-09-20:
I trimmed down the pipeline slightly by conditionally removing nodes that weren't necessary (imageManip and the mono encoders/outs) and that seemed to help stability greatly, though I have still had a few intermittent failures.

See: [crashDump_0_18443010C15E9F0F00_9ed7c9ae4c232ff93a3500a585a6b1c00650e22c.json](https://github.com/user-attachments/files/17068023/crashDump_0_18443010C15E9F0F00_9ed7c9ae4c232ff93a3500a585a6b1c00650e22c.json)