luxonis/depthai-core #1108: Depthai-core dependencies not built with -fPIC

cesarpgouveia, 2024-08-22:
Hi,

I'm working on supporting your camera on an Ubuntu system and integrating depthai-core into my solution. My project is built with the -fPIC (Position Independent Code) flag, so I had to modify your CMakeLists file to include this flag:

```
# Set compiler features (C++14), and disable extensions (g++14)
set_property(TARGET ${TARGET_CORE_NAME} PROPERTY CXX_STANDARD 14)
set_property(TARGET ${TARGET_CORE_NAME} PROPERTY CXX_STANDARD_REQUIRED ON)
set_property(TARGET ${TARGET_CORE_NAME} PROPERTY CXX_EXTENSIONS OFF)
**set_property(TARGET ${TARGET_CORE_NAME} PROPERTY POSITION_INDEPENDENT_CODE ON)**
```

I then built your solution with CMake and installed it on my system. However, when I try to compile my project with support for your code using the following:

```
if(VB_USE_OAKD)
    target_link_libraries(${PROJECT_NAME} depthai::opencv)
endif(VB_USE_OAKD)
```

Everything compiles fine until it reaches the depthai-core dependencies, which appear to have been built without the -fPIC flag (Am I on the right track here?):

```
/usr/lib/aarch64-linux-gnu/cmake/depthai/dependencies/lib/libXLink.a(PlatformDeviceControl.c.o): in the function XLinkPlatformBootRemote:
PlatformDeviceControl.c:(.text+0x1d0): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/lib/aarch64-linux-gnu/cmake/depthai/dependencies/lib/libXLink.a(PlatformDeviceControl.c.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol "mvLogLevel_PlatformDeviceControl" which may bind externally cannot be used when making a shared object; recompile with -fPIC
PlatformDeviceControl.c:(.text+0x218): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/lib/aarch64-linux-gnu/cmake/depthai/dependencies/lib/libXLink.a(PlatformDeviceControl.c.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol "mvLogLevel_PlatformDeviceControl" which may bind externally cannot be used when making a shared object; recompile with -fPIC
PlatformDeviceControl.c:(.text+0x254): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/lib/aarch64-linux-gnu/cmake/depthai/dependencies/lib/libXLink.a(PlatformDeviceControl.c.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol "mvLogLevel_PlatformDeviceControl" which may bind externally cannot be used when making a shared object; recompile with -fPIC
/usr/lib/aarch64-linux-gnu/cmake/depthai/dependencies/lib/libXLink.a(PlatformDeviceControl.c.o): in the function XLinkPlatformCloseRemote:
PlatformDeviceControl.c:(.text+0x748): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/lib/aarch64-linux-gnu/cmake/depthai/dependencies/lib/libXLink.a(PlatformDeviceControl.c.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol "mvLogLevel_PlatformDeviceControl" which may bind externally cannot be used when making a shared object; recompile with -fPIC
PlatformDeviceControl.c:(.text+0x780): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/lib/aarch64-linux-gnu/cmake/depthai/dependencies/lib/libXLink.a(PlatformDeviceControl.c.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol "mvLogLevel_PlatformDeviceControl" which may bind externally cannot be used when making a shared object; recompile with -fPIC
```

Have you tested this with PIC code? Could you help me identify what's going wrong?

themarpe, 2024-08-22:
@cesarpgouveia correct - have to set BUILD_SHARED for depthai to enable PIC build of its parts and the dependencies.
(and don't forget to remove Hunter cache: `rm -rf ~/.hunter` in between)

closed 2024-08-22 by themarpe

cesarpgouveia, 2024-08-22:
So I follow your suggestion, removed all libs associated with the previous depthaicore install by using:
`xargs rm < install_manifest.txt`
Then I removed the hunter cache:
`rm -rf ~/.hunter`
And finally the CMakeCache.txt from depthaicore build:
`rm build/CMakeCache.txt`

And now I'm trying to build depthaicore with these flags:

cmake -H. -Bbuild -D'BUILD_SHARED_LIBS=ON' -D'CMAKE_INSTALL_PREFIX=/usr/'

but now I'm getting the same error but on hunter:

```
/usr/bin/ld: /home/vbuser/.hunter/_Base/cb0ea1f/08f9145/69d66b5/Install/lib/liblzma.a(alone_decoder.c.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against the symbol "lzma_alone_decoder_init" which may bind externally cannot be used when creating a shared object; recompile with -fPIC
/home/vbuser/.hunter/_Base/cb0ea1f/08f9145/69d66b5/Install/lib/liblzma.a(alone_decoder.c.o): in the function lzma_alone_decoder_init:
alone_decoder.c:(.text+0x314): dangerous relocation: unsupported relocation
/usr/bin/ld: /home/vbuser/.hunter/_Base/cb0ea1f/08f9145/69d66b5/Install/lib/liblzma.a(alone_decoder.c.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against the symbol "lzma_alone_decoder_init" which may bind externally cannot be used when creating a shared object; recompile with -fPIC
/home/vbuser/.hunter/_Base/cb0ea1f/08f9145/69d66b5/Install/lib/liblzma.a(alone_decoder.c.o): in the function lzma_alone_decoder:
alone_decoder.c:(.text+0x43c): dangerous relocation: unsupported relocation
/usr/bin/ld: /home/vbuser/.hunter/_Base/cb0ea1f/08f9145/69d66b5/Install/lib/liblzma.a(stream_decoder.c.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against the symbol "lzma_stream_decoder_init" which may bind externally cannot be used when creating a shared object; recompile with -fPIC
/home/vbuser/.hunter/_Base/cb0ea1f/08f9145/69d66b5/Install/lib/liblzma.a(stream_decoder.c.o): in the function lzma_stream_decoder_init:
stream_decoder.c:(.text+0x544): dangerous relocation: unsupported relocation
/usr/bin/ld: /home/vbuser/.hunter/_Base/cb0ea1f/08f9145/69d66b5/Install/lib/liblzma.a(stream_decoder.c.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against the symbol "lzma_stream_decoder_init" which may bind externally cannot be used when creating a shared object; recompile with -fPIC
/home/vbuser/.hunter/_Base/cb0ea1f/08f9145/69d66b5/Install/lib/liblzma.a(stream_decoder.c.o): in the function lzma_stream_decoder:
stream_decoder.c:(.text+0x6d0): dangerous relocation: unsupported relocation
/usr/bin/ld: /home/vbuser/.hunter/_Base/cb0ea1f/08f9145/69d66b5/Install/lib/liblzma.a(block_decoder.c.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against the symbol "lzma_block_decoder_init" which may bind externally cannot be used when creating a shared object; recompile with -fPIC
/home/vbuser/.hunter/_Base/cb0ea1f/08f9145/69d66b5/Install/lib/liblzma.a(block_decoder.c.o): in the function lzma_block_decoder_init:
block_decoder.c:(.text+0x2ec): dangerous relocation: unsupported relocation
/usr/bin/ld: /home/vbuser/.hunter/_Base/cb0ea1f/08f9145/69d66b5/Install/lib/liblzma.a(block_decoder.c.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against the symbol "lzma_block_decoder_init" which may bind externally cannot be used when creating a shared object; recompile with -fPIC
/home/vbuser/.hunter/_Base/cb0ea1f/08f9145/69d66b5/Install/lib/liblzma.a(block_decoder.c.o): in the function lzma_block_decoder:
block_decoder.c:(.text+0x46c): dangerous relocation: unsupported relocation
```

cesarpgouveia, 2024-08-22:
@themarpe

themarpe, 2024-08-22:
@cesarpgouveia 

You'll likely need to clear the build directory as well:
```
rm -r build/
rm -rf ~/.hunter
cmake -S . -B build -D BUILD_SHARED_LIBS=ON -D CMAKE_INSTALL_PREFIX=/usr
cmake --build build --parallel
```

cesarpgouveia, 2024-08-22:
Hey @themarpe, 

The "rm -r build/" solved the thing but now when I try to build my own solution I got this error:

/usr/include/depthai-shared/utility/Serialization.hpp:9:10: fatal error: nop/serializer.h: file not exists

It should be included in depthai-core right? Why I'm getting this error?

themarpe, 2024-08-22:
@cesarpgouveia perhaps the usr/include is taking precedence - try removing the usr/include/depthai-shared or make sure include paths are prioritized correctly

themarpe, 2024-08-22:
Did you link using target_link_libraries? The nop is under 3rdparty dependencies folder, but Include paths must be set correctly for the build system to find it

cesarpgouveia, 2024-08-22:
Tried to remove the usr/include/depthai-shared and got a new error:

```
/usr/include/depthai/device/CalibrationHandler.hpp:5:10: fatal error: depthai-shared/common/CameraBoardSocket.hpp: file not exists
 #include "depthai-shared/common/CameraBoardSocket.hpp"
```

I have my main CMakeLists.txt and then I have small CMakeLists for the subprojects of the solution. On my main CMakeLists.txt I added this:

```
# OAK-D
if(VB_USE_OAKD)
	find_package(depthai CONFIG REQUIRED)
	if(${depthai_FOUND})
		message(STATUS "Compiling with support for OAK-D DepthAI.")
		add_definitions(-DUSE_OAKD)
		include_directories(${depthai_INCLUDE_DIR})
	else(${depthai_FOUND})
		message(WARNING "Compiling WITHOUT support for OAK-D DepthAI. Lib not found!")
		unset(VB_USE_OAKD)
	endif(${depthai_FOUND})
endif(VB_USE_OAKD)
```

Which "apparently" was able to find the depthai libs, and then on the CMakeLists.txt from the project that is dependent from depthai-core I added this:

```
if(VB_USE_OAKD)
	target_link_libraries(${PROJECT_NAME} depthai::opencv)
endif(VB_USE_OAKD)
```

What do you mean by the include paths must be set correctly?

themarpe, 2024-08-23:
@cesarpgouveia 

If /usr/include has priority you'll have to delete all (/usr/include/depthai, depthai-bootloader-shared, etc...)

Otherwise, check with `--build [build] --verbose` and you'll see what `-I[include/dirs]` get passed to the compiler.

```
# OAK-D
if(VB_USE_OAKD)
	find_package(depthai CONFIG REQUIRED)
	if(${depthai_FOUND})
		message(STATUS "Compiling with support for OAK-D DepthAI.")
		add_definitions(-DUSE_OAKD)
		include_directories(${depthai_INCLUDE_DIR})
	else(${depthai_FOUND})
		message(WARNING "Compiling WITHOUT support for OAK-D DepthAI. Lib not found!")
		unset(VB_USE_OAKD)
	endif(${depthai_FOUND})
endif(VB_USE_OAKD)
```

This won't fly likely, just use the `target_link_libraries(${PROJECT_NAME} depthai::opencv)` directly, which handles includes as well as the compiler definitions, etc...

cesarpgouveia, 2024-08-23:
Hey @themarpe,

Just removed the `include_directories(${depthai_INCLUDE_DIR})` and now everything builds! Thanks for the help, this include_dirs was somehow messing with the way the cmake was searching for the libs. Thank you very much for your assistance and prompt replies, really appreciate it!

-----

Btw, I'm trying to get the depthaicore version on cmake but nothing seems to work:

if(VB_USE_OAKD)
	message(STATUS "OAK-D DepthAI version found: ${depthai_VERSION}")
endif()

Is there a flag for that?

themarpe, 2024-08-23:
Great to hear!

--- 

> Btw, I'm trying to get the depthaicore version on cmake but nothing seems to work:
>
> if(VB_USE_OAKD)
> message(STATUS "OAK-D DepthAI version found: ${depthai_VERSION}")
> endif()
>
> Is there a flag for that?

There's `lib/cmake/depthai/depthaiConfigVersion.cmake` which should hold it. Not sure if it has to be explicilty applied to our CMake or its intended to be extracted by `find_package`. If the latter, you may open a PR to depthai-core to add this explicitly (not sure what is the CMake approach here, quick google search didn't yield anything standard)