luxonis/depthai-core #1126: [BUG] {Error when link C++ runtime statically}

jfulem, 2024-09-13:
I'm on windows, using MSVC (Visual Studio 2022).

Is it possible to build your **depthai-core** library with the` /MT` flag - which links the C++ runtime statically?

I tried that and it looks there are some issues with dependencies: XLink, spdlog
`XLink.lib(ProtocolManager.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MT_StaticRelease' in DatatypeEnum.obj`
`spdlog.lib(fmt.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MT_StaticRelease' in DatatypeEnum.obj`

How can I configure Hunter to build the dependencies with `/MT` as well?

Or am I missing something else?

moratom, 2024-09-22:
Hi @jfulem , sorry for the late response.

As far as I know we haven't yet compiled the library with the `/MT` flag.

I believe that the flag should be passed down to hunter dependanices in case you use `CMAKE_CXX_FLAGS` variable.

Otherwise you can try to set the flag for each dependacny separatly in this file https://github.com/luxonis/depthai-core/blob/main/cmake/Hunter/config.cmake (see how other flags are passed down).

How are you setting the `/MT` flag now?

jfulem, 2024-09-22:
@moratom, thank you.

I tried both:
 - passing CMAKE_CXX_FLAGS
 - each dependency in Hunter/config.cmake
 
 But I still get the same mismatch error. I will give it another try.