luxonis/depthai-core #1145: [BUG] {The issue of output image of OAK-1-W sensor with 400P resolution}

cjh1995-ros, 2024-10-07:
**Describe the bug**
A clear and concise description of what the bug is.
Currently, I'm working on `OAK-1-W-97` with `OV9782`. In the doc, it supports 400P, 720P, 800P. So I try to test every resolution and 720P and 800P work fine, but 400P gave me the following result. 

[image]

I tested it with following code.
```python
import depthai as dai
import cv2

pipeline = dai.Pipeline()

# Define sources and outputs
camRgb: dai.node.Camera = pipeline.create(dai.node.Camera)

#Properties
camRgb.setBoardSocket(dai.CameraBoardSocket.CAM_A)
camRgb.setSize((640, 400))

# Linking
videoOut = pipeline.create(dai.node.XLinkOut)
videoOut.setStreamName("video")
camRgb.video.link(videoOut.input)

ispOut = pipeline.create(dai.node.XLinkOut)
ispOut.setStreamName("isp")
camRgb.isp.link(ispOut.input)

with dai.Device(pipeline) as device:
    video = device.getOutputQueue(name="video", maxSize=1, blocking=False)
    isp = device.getOutputQueue(name="isp", maxSize=1, blocking=False)

    while True:
        if video.has():
            cv2.imshow("video", video.get().getCvFrame())
        if isp.has():
            cv2.imshow("isp", isp.get().getCvFrame())
        if cv2.waitKey(1) == ord('q'):
            break
```

For additional test, here's the result of `depthai-viewer`. Idk why the following log came out.
```
[18443010C1D9E4F400] [3.3.2.1] [1.634] [ColorCamera(0)] [warning] Unsupported resolution set for detected camera OV9782, needs 800_P or 720_P. Defaulting to 800_P
Couldn't start pipeline:  ColorCamera(0) - 'video' width or height (1920, 1080) bigger than maximum at current sensor resolution (1280, 800)
[2024-10-07T05:55:06Z ERROR re_viewer::depthai::depthai] Error: Couldn't start pipeline
Selecting device:  
Resetting...
Closing device...
Done
```

jakaskerl, 2024-10-07:
I think the issue is that older calibrations store ov9782 as being able to output both mono and color: `socket: CAM_A, sensorName: OV9782, width: 1280, height: 800, orientation: AUTO, supportedTypes: [COLOR, MONO]`. We can fix it by changing the calibration or to fix in FW. 
cc @SzabolcsGergely

cjh1995-ros, 2024-10-08:
@jakaskerl So do I need to calibrate the camera with chessboard?

jakaskerl, 2024-10-08:
Hi @cjh1995-ros 
Afaik the device needs to be reflashed completely (not possible through recalibration). Best to wait for FW fix. 

Workaround:
```
camRgb.setSize((1280, 800))
camRgb.setVideoSize((640, 400))
```

cjh1995-ros, 2024-10-10:
> Hi @cjh1995-ros Afaik the device needs to be reflashed completely (not possible through recalibration). Best to wait for FW fix.
> 
> Workaround:
> 
> ```

> camRgb.setSize((1280, 800))
> camRgb.setVideoSize((640, 400))
> ```

Sorry that I close the issue. I thought it solved the problem but it's not. The result gave me cropped image. But what I intended is resized image. As it cropped the image, the fov of camera is decreased so hard.  

```
camRgb.setSize((1280, 800))
camRgb.setVideoSize((640, 400))
```
[image]

```
camRgb.setSize((1280, 800))
camRgb.setVideoSize((1280, 800))
```
[image]

```
camRgb.setSize((640, 400))
camRgb.setVideoSize((640, 400))
```
[image]

cjh1995-ros, 2024-10-10:
> Hi @cjh1995-ros Afaik the device needs to be reflashed completely (not possible through recalibration). Best to wait for FW fix.
> 
> Workaround:
> 
> ```
> camRgb.setSize((1280, 800))
> camRgb.setVideoSize((640, 400))
> ```

Sorry that I closed the issue. I just checked and it works. However, fov is decreased so hard. Guess it cropped the undistorted image. Here's the results. Is there any way to resize the image in the `Camera` node?  

(setSize - serVideoSize)
800P - 400P 
[image]

800P - 800P
[image]

400P - 400P
[image]

cjh1995-ros, 2024-10-14:
> Hi @cjh1995-ros
Afaik the device needs to be reflashed completely (not possible through recalibration). Best to wait for FW fix.

@jakaskerl Then how long should I wait for the FW update?

lturing, 2024-10-28:
@jakaskerl 
I meet the same question, how long will update the FW? many thanks

cjh1995-ros, 2024-12-04:
@jakaskerl any update?

jakaskerl, 2024-12-04:
Depthai V3 Camera node has been revamped a bit so I believe it should work there when released (currently alpha). Thanks for the patience.

cjh1995-ros, 2024-12-05:
> Depthai V3 Camera node has been revamped a bit so I believe it should work there when released (currently alpha). Thanks for the patience.

@jakaskerl I'm very happy with it but sadly I can't wait for it. Could you give me insight? What if I add `ImageManip` node at the last of pipeline and resize it? Can it be work? or will it give same issue?

jakaskerl, 2024-12-05:
@cjh1995-ros 
- try camRgb.setCalibrationAlpha(1) OR
- use ColorCamera node + imagemanip to undistort it. That should work fine. https://github.com/luxonis/depthai-python/blob/main/examples/ColorCamera/rgb_undistort.py

cjh1995-ros, 2024-12-06:
> @cjh1995-ros
> 
> * try camRgb.setCalibrationAlpha(1) OR
> * use ColorCamera node + imagemanip to undistort it. That should work fine. https://github.com/luxonis/depthai-python/blob/main/examples/ColorCamera/rgb_undistort.py

I checked it work. cc @lturing 

This is modified code. I think we can close this page. Just could you please let us know when new release version is created? @jakaskerl 

```python
import depthai as dai
import cv2

pipeline = dai.Pipeline()

# Define sources and outputs
camRgb: dai.node.Camera = pipeline.create(dai.node.Camera)
manipVideo: dai.node.ImageManip = pipeline.create(dai.node.ImageManip)
manipIsp: dai.node.ImageManip = pipeline.create(dai.node.ImageManip)

#Properties
camRgb.setBoardSocket(dai.CameraBoardSocket.CAM_A)
camRgb.setSize((1280, 800))
camRgb.setVideoSize((1280, 800))

# Manip property
manipVideo.setMaxOutputFrameSize(320 * 200 * 3)
manipVideo.initialConfig.setResize(320, 200)
manipIsp.setMaxOutputFrameSize(320 * 200 * 3)
manipIsp.initialConfig.setResize(320, 200)

# Set stream names
videoOut = pipeline.create(dai.node.XLinkOut)
videoOut.setStreamName("video")

ispOut = pipeline.create(dai.node.XLinkOut)
ispOut.setStreamName("isp")

# Linking
camRgb.isp.link(manipIsp.inputImage)
camRgb.video.link(manipVideo.inputImage)

manipIsp.out.link(ispOut.input)
manipVideo.out.link(videoOut.input)

with dai.Device(pipeline) as device:
    video = device.getOutputQueue(name="video", maxSize=1, blocking=False)
    isp = device.getOutputQueue(name="isp", maxSize=1, blocking=False)

    while True:
        if video.has():
            cv2.imshow("video", video.get().getCvFrame())
        if isp.has():
            cv2.imshow("isp", isp.get().getCvFrame())
        if cv2.waitKey(1) == ord('q'):
            break

```