luxonis/depthai-core #1162: [BUG] Reconnection logic doesn't work when registering callbacks

guilhermedemouraa, 2024-10-31:
Creating a new issue here since I haven't heard back on https://github.com/luxonis/depthai-core/issues/1042.

Hello DepthAI team,

In our application, we register a callback for new frames using DepthAI's API. This callback is triggered whenever a new packet/frame is received. However, this design leaves me uncertain about how to catch exceptions or handle a dropped device event.

Specifically, I'm looking for recommendations on how to implement reconnection logic in a situation where the device is dropped, but no new frames are being received (i.e., the callback is not triggered). Since I rely heavily on the callback to process data, I am unsure where or how to catch any exceptions that might arise due to a device being dropped or encountering other issues. I found this [resource online](https://github.com/luxonis/depthai-docs-website/blob/master/source/pages/troubleshooting.rst#ping-was-missed-closing-the-device-connection), but I don't think this could work for the callback approach.

Could you provide any examples or guidance on how to detect and handle these cases within the callback system or elsewhere in the data flow? Additionally, if there are best practices for managing reconnections, I would greatly appreciate your insights.

Thank you for your continued support.

guilhermedemouraa, 2024-11-16:
Could anyone please help me?

Erol444, 2024-11-25:
Hi @guilhermedemouraa , perhaps rather use our Forum (discuss.luxonis.com) which is primary support location:)
I'd say you can use `device.isPipelineRunning()`, and if it isn't, create a new Device object and setup callbacks once again (one the new Device object). Thoughts?