luxonis/depthai-core #1215: [BUG] StereoDepth::PresetMode::HIGH_ACCURACY - v2.29.0

jfulem, 2025-01-20:
Using this setting (`StereoDepth::PresetMode::HIGH_ACCURACY`) the build produces this warning:
`"Will be removed in future releases and replaced with DEFAULT`

Based on our experiment this is not behaving the same way as using `StereoDepth::PresetMode::DEFAULT` setting.

moratom, 2025-01-21:
Thanks for the report @jfulem!

We'll likely only make the change on the major (breaking) v3 release. We're aware that the `PresetMode::DEFAULT` behaves differently compared to the existing `PresetMode::HIGH_ACCURACY` and that a silent replace would break many applications, so we'll make the change on a breaking release.

Do you mind letting us know if the HIGH_ACCURACY worked better for your usecase? We might reconsider removing it if it's still the best preset for some usecases.

jfulem, 2025-01-21:
Hi @moratom, yes, `HIGH_ACCURACY` works much much better. We also tried `HIGH_DETAIL`, but it's not the same either.

Currently, we have an issue using `v2.29.0` since it's producing warnings and we treat warnings as errors 🦖 .

moratom, 2025-01-27:
Much appreciated feedback @jfulem. 

If you don't mind expending in what ways do you find the HIGH_ACCURACY better, that'd would be great.

In either case, we'll reevaluate the presets internally and try to come up with a better alternative, before we remove the `HIGH_ACCURACY` in `v3`.

jfulem, 2025-01-28:
`HIGH_ACCURACY ` is great when we work with depth.