luxonis/depthai-core #1227: [Feature-Request] Full rust bindings?

Carter12s, 2025-01-31:
### Start with the `why`:
Rust is rapidly being adopted in the robotics community. I'm an avid proponent of Rust and believe it offers a substantially superior experience to C++. It would be a substantial benefit to Luxonis's users if well supported Rust bindings existed that wrapped the C++ api and were easy to work with.

### Move to the `what`:
The rust ecosystem offers a large set of tools to make C++ <-> Rust interop easy to achieve. As far as I can find online this blog post an associated repo are the only example of setting these up for Depth API.
https://npenkov.com/post/2022/11/19/using-cpp-from-rust-depthai-core-rgb-camera-example/

It would be ideal if rust idiomatic bindings over the C++ libraries could be automatically generated and published to crates.io.

### Move to the `how`:
Already mentioned above but between cxx and bindgen the rust community already has a lot of support for setting this up. There are many example projects out there for folks that are regularly publishing rust bindings for a C++ library.

I'd suggest https://github.com/twistedfall/opencv-rust as a solid example.

It is possible for the Rust community to achieve this, but it would be ideal if there was a single "Supported by Luxonis" version of these bindings that the community could coalesce behind.

I'm personally just dipping my toes into the DepthAI api, but am an experienced rust developer and would be wiling to contribute to setting up the bindings if this was something Luxonis was interested in?

moratom, 2025-02-05:
Thanks for the feature request, @Carter12s!

We’d love to see a community effort to provide Rust bindings for the DepthAI library and would be happy to offer some level of support. However, we don’t currently have the bandwidth to take on this project ourselves.

That said, we do have some Rust enthusiasts on our team who can provide guidance if you decide to kick off the project. The extent of our maintenance and support will largely depend on how widely the bindings are adopted by users.

Let us know if you move forward with it!

Carter12s, 2025-02-05:
Thanks for the response and totally understand!

I'll ponder this over and see if it is something I can take on. I'm currently working in a code base transitioning from C++ => Rust and we're trying to not add new C++ dependencies...

closed 2025-02-05 by Carter12s