luxonis/depthai-core #1228: [Feature-Request] Provide Development Docker Images

Carter12s, 2025-01-31:
### Start with the `why`:
Since this project is relatively complex with a substantial number of dependencies that are required, it would be very convenient to users if there were published docker images containing a full working build of the project and examples. This can significantly speed up the time to a working example for users familiar with docker, and provides a good example of how to setup the needed dependencies for the project.

Furthermore, using multi-stage docker builds these images can provide a very easy method for users to extend these images for their application.

The DepthAI repo provides similar docker images, but it would be ideal if this repo also provided development focused ones.

### Move to the `what`:

- Add a Dockerfile, probably based on an `ubuntu` image to the repo
- Expand CI to automatically generate and publish these docker images for each version release
- Update documentation to point users to an example of using the docker images

### Move to the `how`:
As above. There are many examples of this online.

moratom, 2025-02-05:
@Carter12s thanks for the feature request.

I agree, this would be super useful for getting started, so I'm adding to our backlog.

phodina, 2025-04-23:
@Carter12s Docker images are helpful for the development, however, since you can visualize the streams and connect over USB it brings additional challenges.

So to be more specific it's the GUI sharing based on the windowing protocol (X11/wayland) and sharing the access to the `/dev/bus/usb` that are little bit problematic to setup.

Therefore the question is what do you expect from the container? Just DepthAI installed and interact with OAK device over network without the GUI stack in the docker?

Carter12s, 2025-04-24:
Fair points on the GUI challenges, and USB challenges!

For my specific use case I'm fine with USB not working and with GUI elements not working. I think a very typical development flow for a robotics company adding this as a new sensor is to get a prototype stood up in a basic Docker container that can be used for evaluation and testing.

I'd like to achieve two things with these images:

- Absolutely minimize the time and number of steps from plugging the sensor in to getting C++ code I've written to save images off of the camera.
- Provide example Dockerfiles that show setting up the dependencies so that DepthAI can be easily integrated into other larger docker images.

phodina, 2025-04-25:
Okay. I use nix and I'm upstreaming the dependencies to nixpkgs so it can be deterministically built.

With that it's pretty simple to put it inside a docker container. However, there I stopped as I did not have clear expectations of the environment - USB, GUI ... 

So I'd say I can create a minimal "unofficial" docker image and push it to dockerhub to test? It will be built using the nix description but that could later be converted into proper Dockerfile.

Would this proof of concept help @moratom to define the Clickup task to have official support?

https://github.com/phodina/depthai-nix/tree/master

moratom, 2025-05-05:
@phodina we'll be adding a container with a minimal build with the merge of `v3_cleanup` which will embed all the private dependanices without exposing the symbols (and we'll get rid of public ones) so we'll be distribution/package manager agnostic.